espressif/esp-matter #285: Clean install with current tree

jonsmirl, 2023-03-27:
This command is failing.
./connectedhomeip/connectedhomeip/scripts/checkout_submodules.py --platform esp32 --shallow

I am starting over from clean tree because I have lost any ability for Google to control my devices.

jonsmirl, 2023-03-27:
git submodule update --init --depth 1
cd connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 --shallow
cd ../..
./install.sh

jonsmirl, 2023-03-27:
Next error:

Done. Made 6364 targets from 316 files in 365ms
ninja: Entering directory `/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/out/host'
ninja: error: '../../third_party/libwebsockets/repo/lib/core-net/adopt.c', needed by 'obj/third_party/libwebsockets/repo/lib/core-net/libwebsockets.adopt.c.o', missing and no known rule to make it

jonsmirl, 2023-03-27:
Fix....

cd connectedhomeip/connectedhomeip
git submodule update --init --depth 1 third_party/libwebsockets/repo

When rerunning ./install it stopped on:
Installing zap-cli

mkdir: cannot create directory ‘/home/jonsmirl/aosp/esp-matter/.zap’: File exists

rm -rf /home/jonsmirl/aosp/esp-matter/.zap
./install.sh

jonsmirl, 2023-03-27:
Next failure:
~~~
[50/749] ACTION //third_party/connectedhomeip/src/controller/data_model:data_model_codegen_codegen(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
FAILED: gen/third_party/connectedhomeip/src/controller/data_model/app/PluginApplicationCallbacks.h gen/third_party/connectedhomeip/src/controller/data_model/app/callback-stub.cpp 
python ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../../../../connectedhomeip/connectedhomeip/config/esp32/ --current-path ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model --default-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --current-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --capture-output --python-dep-list-files gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen_codegen_metadata_path_list.txt -- ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir gen/third_party/connectedhomeip/src/controller/data_model --expected-outputs gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen.expected.outputs ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.matter
Traceback (most recent call last):
  File "/home/jonsmirl/aosp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from matter_idl.matter_idl_parser import CreateParser
ModuleNotFoundError: No module named 'matter_idl'

During handling of the above exception, another exception occurred:

~~~

jonsmirl, 2023-03-27:
Fix:

Previously I built and I was missing ESP openthread. So upgraded from idf 4.3 to idf 4.4.  But I had run CHIP bootstrap while idf 4.3 environment was active and it recorded the 4.3 python environment. This run IDF 4.4 was active so the 4.3 python environment was gone.  matter_idl had been installed into the 4.3 environment.

So whack .environment and run . ./scripts/bootstrap.sh again.

jonsmirl, 2023-03-27:
That was only part of the problem...
~~~
Traceback (most recent call last):
  File "/home/jonsmirl/aosp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from matter_idl.matter_idl_parser import CreateParser
ModuleNotFoundError: No module named 'matter_idl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jonsmirl/aosp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 32, in <module>
    from matter_idl.matter_idl_parser import CreateParser
  File "/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/py_matter_idl/matter_idl/matter_idl_parser.py", line 6, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
~~~

jonsmirl, 2023-03-27:
Fix...  to do esp-matter CHIP build these modules need to be installed into the the IDF venv, not the CHIP venv.
~~~
pip install lark stringcase
~~~

jonsmirl, 2023-03-27:
~~~
cd examples/light
idf.py set-target esp32s3
idf.py erase-flash
idf.py build flash monitor
~~~
ESP32S3 booted. Now try commissioning from Google Home.
Scan QRCode, circle spins....
Popup "Something went wrong"

jonsmirl, 2023-03-27:
Go into light_switch example....
idf.py set-target esp32s3
CMake Error at CMakeLists.txt:21 (message):
  Unsupported IDF_TARGET

Switch over to my esp32c3
~~~
cd examples/light_switrch
idf.py set-target esp32c3
idf.py erase-flash
idf.py build flash monitor
~~~

ESP32C3 booted.  Now try commissioning from Google Home.
Scan QRCode, circle spins....
Popup "Something went wrong"

Reboot ESP, try again, same thing.  Check Android logs, nothing interesting.
~~~
I (1331) chip[DIS]: CHIP minimal mDNS started advertising.
I (1331) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (1341) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 71C5EE2BE18F8D62.
I (1351) chip[DIS]: mDNS service published: _matterc._udp
I (1361) chip[IN]: CASE Server enabling CASE session setups
I (1371) chip[SVR]: Joining Multicast groups
I (1371) chip[SVR]: Server Listening...
I (1381) esp_matter_core: Dynamic endpoint 0 added
I (1381) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0003's Attribute 0x0001 is 0 **********
I (1391) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0004's Attribute 0x0000 is 128 **********
I (1401) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0004's Attribute 0xFFFC is <invalid type: 0> **********
I (1411) esp_matter_core: Dynamic endpoint 1 added
E (1421) chip[DL]: Long dispatch time: 188 ms, for event type 2
I (1431) chip[DL]: WIFI_EVENT_STA_START
W (1431) wifi:Haven't to connect to a suitable AP now!
I (1431) chip[DL]: Done driving station state, nothing else to do...
I (1441) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (1451) chip[DL]: Device already advertising, stop active advertisement and restart
I (1461) NimBLE: GAP procedure initiated: stop advertising.

I (1461) NimBLE: GAP procedure initiated: advertise; 
I (1471) NimBLE: disc_mode=2
I (1471) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1481) NimBLE: 

I (1491) app_main: Commissioning window opened
> I (31331) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (31331) chip[DL]: Device already advertising, stop active advertisement and restart
~~~

jonsmirl, 2023-03-27:
Factory reset Google Home Hub. No difference.

jonsmirl, 2023-03-27:
IOS and chip-tool are able to add the device.

jonsmirl, 2023-03-28:
The Google failure is very early. It is not connecting to the BLE device correctly. I don't think it even gets into the Matter code. I get the new Matter device detected popup on Android phone. And I can use rRF Connect app from Android phone to connect to the ESP BLE devices.

jonsmirl, 2023-03-28:
~~~
cd examples/generic_switch
idf.py set-target esp32c3
idf.py build
compiler errors

cc1plus: error: unrecognized command line option '-Wno-unknown-warning-option' [-Werror]
cc1plus: all warnings being treated as errors
~~~

This build is failing because there are warnings, and this unrecognized command line option is causing warnings to count as errors.

dhrishi, 2023-03-30:
Thanks for all the details and solutions @jonsmirl . We are checking and this should be fixed soon. cc @wqx6

jadhavrohit924, 2023-03-30:
> ```
> cd examples/generic_switch
> idf.py set-target esp32c3
> idf.py build
> compiler errors
>
> cc1plus: error: unrecognized command line option '-Wno-unknown-warning-option' [-Werror]
> cc1plus: all warnings [...]

@jonsmirl I was able to build `generic_switch` example successfully with idf `release/v5.1` commit `bb9200acec7dd60e9adb4a381e5400dcd5024534`. 
esp-matter commit `bafc378e68f1c32f72329ebcc986f2cfa88d67ed`

Could you please give it a try again?

dhrishi, 2023-03-31:
@jonsmirl 

> Is it safe for me to move esp32s3 and esp32c3 onto IDF 5.1 now?

release/v1.0 branch (a tag should be created in a few days) will support Matter v1.0 with IDF v4.4.x for ESP32-C3/ESP32-S3/ESP32

IDF v5.1 for ESP32-C3/ESP32-S3/ESP32 is supported only on the main branch which will have Matter v1.1 The upstream hasn't officially released v1.1 yet. Also, IDF v5.1 release is scheduled around end of May.

dhrishi, 2023-04-12:
@jonsmirl Can you close this issue if you are able to compile on the main branch?

dhrishi, 2023-04-28:
Closing this. Feel free to re-open in case of any issues

closed 2023-04-28 by dhrishi

bingshanguxue, 2023-10-24:
> Fix... to do esp-matter CHIP build these modules need to be installed into the the IDF venv, not the CHIP venv.
> 
> ```
> pip install lark stringcase
> ```

hi, how did you fixed this?

bingshanguxue, 2023-10-24:
> That was only part of the problem...
>
> ```
> Traceback (most recent call last):
>   File [...]

hi, how did you fixed this?

jonsmirl, 2023-10-24:
You need to rerun the install.sh script in esp-matter root