espressif/esp-matter #287: Fetching pairing code, and passphrase within the code.

MartinPArmstrong, 2023-03-28:
Is there an easy way to fetch the pairing code and passphrase from within the code itself? Want to print the Wi-Fi pairing code along side passphrase by adding it to the event switch case.

```
static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) {
    case chip::DeviceLayer::DeviceEventType::kInterfaceIpAddressChanged:
        ESP_LOGI(TAG, "Interface IP Address changed");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningComplete:
        ESP_LOGI(TAG, "Commissioning complete");
        break;

    case chip::DeviceLayer::DeviceEventType::kFailSafeTimerExpired:
        ESP_LOGI(TAG, "Commissioning failed, fail safe timer expired");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStarted:
        ESP_LOGI(TAG, "Commissioning session started");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStopped:
        ESP_LOGI(TAG, "Commissioning session stopped");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowOpened:
        ESP_LOGI(TAG, "Commissioning window opened");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowClosed:
        ESP_LOGI(TAG, "Commissioning window closed");
        break;

    default:
        break;
    }
}
```

shubhamdp, 2023-03-29:
Is there an easy way to fetch the pairing code and passphrase from within the code itself? Want to print the Wi-Fi pairing code along side passphrase by adding it to the event switch case.

Is pairing code is Matter pairing code (qrcode or manual pairing code)? If yes and this is for just test purposes then you can use APIs from [QRCodeSetupPayloadGenerator.h](https://github.com/project-chip/connectedhomeip/blob/master/src/setup_payload/QRCodeSetupPayloadGenerator.h) and [ManualSetupPayloadGenerator.h](https://github.com/project-chip/connectedhomeip/blob/master/src/setup_payload/ManualSetupPayloadGenerator.h).

If you are considering this for production use cases then as per Matter spec v1.0, section 5.1.7. Generation of the Passcode, and for security purposes its better to not have passcode on the device.

I am not sure, if you will be able to fetch the wifi passcode of connected AP.

And you will have to do additional work to expose this through events, but making APIs for this is easy.

MartinPArmstrong, 2023-03-30:
Just what i was looking for. Closing the issue.