espressif/esp-matter #289: Light Example Build Issue with ESP -IDF v5.1

aggaddam, 2023-03-29:
I am unable to build ESP matter Light example code which is available under esp-matter/examples/light. I am getting the below error 

`[967/1120] Performing build step for 'chip_gn'FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build/esp-idf/chip/lib/libCHIP.a 
cd /home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build/esp-idf/chip && ninja esp32
[138/567] c++ obj/third_party/connectedhomeip/src/crypto/cryptopal_mbedtls.CHIPCryptoPALmbedTLS.cpp.oFAILED: obj/third_party/connectedhomeip/src/crypto/cryptopal_mbedtls.CHIPCryptoPALmbedTLS.cpp.o 
/home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -MMD -MF obj/third_party/connectedhomeip/src/crypto/cryptopal_mbedtls.CHIPCryptoPALmbedTLS.cpp.o.d -Og -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build/config -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/freertos/include/freertos -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/newlib/platform_include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/freertos/FreeRTOS-Kernel/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/freertos/esp_additions/include/freertos -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/freertos/esp_additions/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_hw_support/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_hw_support/include/soc -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_hw_support/include/soc/esp32c3 -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_hw_support/port/esp32c3/. -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_hw_support/port/esp32c3/private_include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/heap/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/log/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/soc/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/soc/esp32c3 -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/soc/esp32c3/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/hal/esp32c3/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/hal/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/hal/platform_port/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/include/esp32c3 -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/esp32c3 -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_common/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_system/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_system/port/soc -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_system/port/include/riscv -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_system/port/include/private -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/riscv/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/include/apps -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/include/apps/sntp -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/lwip/src/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/port/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/port/freertos/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/port/esp32xx/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/lwip/port/esp32xx/include/arch -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/include/esp32c3/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/common/osi/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/common/api/include/api -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/common/btc/profile/esp/blufi/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/common/btc/profile/esp/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -isystem/home/agaddam/Documents/test_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -isystem/home/agaddam/Documents/test_li_ac_matte [... cut ...]
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:406:14: error: 'mbedtls_pkcs5_pbkdf2_hmac' was not declared in this scope; did you mean 'mbedtls_pkcs5_pbkdf2_hmac_ext'?
  406 |     result = mbedtls_pkcs5_pbkdf2_hmac(&md_ctxt, Uint8::to_const_uchar(password), plen, Uint8::to_const_uchar(salt), slen,
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
      |              mbedtls_pkcs5_pbkdf2_hmac_ext
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
`

ESP Matter SDK: release/v1.0
ESP IDF version: release/v5.1

shubhamdp, 2023-03-29:
> ESP Matter SDK: release/v1.0
ESP IDF version: release/v5.1

@aggaddam This combination is not tested, we only support idf v4.4.3 on esp-matter release/v1.0 branch.

Please use the esp-matter main branch with idf release/v5.1 branch.

dhrishi, 2023-03-29:
release/v1.0 branch (Matter v1.0.0.2) does not support ESP IDF v5.x for ESP32, ESP32-C3, ESP32-S3.

Please try with main vrancg

closed 2023-03-29 by dhrishi