espressif/esp-matter #290: Problems commissioning, Google and chip-tool failing

jonsmirl, 2023-03-29:
I am on esp-matter af62fb0b254304e6114b91608957a421dd1ac58e  (current HEAD)

Google Home can't commission the light example.
IOS works.

When I try from chip-tool.
~~~
[1680102478.005715][863644:863645] CHIP:BLE: Device 1A:F5:27:4A:BD:E0 does not look like a CHIP device.
[1680102478.074066][863644:863645] CHIP:BLE: New device scanned: C6:3E:8E:5D:E5:07
[1680102478.074120][863644:863645] CHIP:BLE: Device discriminator match. Attempting to connect.
[1680102478.074168][863644:863645] CHIP:DL: Chip stack locking error at '../../src/system/SystemStats.cpp:70'. Code is unsafe/racy
[1680102478.074196][863644:863645] CHIP:-: chipDie chipDie chipDie
Aborted (core dumped)
~~~

jonsmirl, 2023-03-29:
I even tried a factory reset on my Nest Hub. I also checked to make sure it was enrolled in preview program. I also used Google Home to reboot it (triggers OTA).

dhrishi, 2023-03-29:
Can you please upload (not paste) the console logs?

jonsmirl, 2023-03-29:
From ESP? It does nothing. It is not making the BLE connection.

jonsmirl, 2023-03-29:
I looked and Android logs too, nothing interesting in there.

You can't get logs from Nest Hub since it is running Fuchsia.

dhrishi, 2023-03-29:
>From ESP? It does nothing. It is not making the BLE connection.

Ohh okay

jonsmirl, 2023-03-29:
Android connects to it because I get the annoying commissioning popup on Android over and over.
Nest Hub is not connecting to it.

Here is tail of the log, Android connected and I tried commissioning. The Nest Hub never even tried connecting.
~~~
I (1657) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0xFFFC is 3 **********
I (1667) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 64 **********
I (1677) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x4000 is 64 **********
I (1687) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 64 **********
I (1697) led_driver_ws2812: led set r:0, g:0, b:0
I (1707) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x4010 is null **********
I (1717) esp_matter_core: Dynamic endpoint 1 added
E (1717) chip[DL]: Long dispatch time: 340 ms, for event type 2
I (1727) chip[DL]: WIFI_EVENT_STA_START
W (1727) wifi:Haven't to connect to a suitable AP now!
I (1737) chip[DL]: Done driving station state, nothing else to do...
I (1747) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (1747) chip[DL]: Device already advertising, stop active advertisement and restart
I (1757) NimBLE: GAP procedure initiated: stop advertising.

I (1767) NimBLE: GAP procedure initiated: advertise; 
I (1767) NimBLE: disc_mode=2
I (1777) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1787) NimBLE: 

I (1787) app_main: Commissioning window opened
I (1787) led_driver_ws2812: led set r:0, g:0, b:0
I (1797) led_driver_ws2812: led set r:0, g:0, b:0
I (1797) led_driver_ws2812: led set r:63, g:51, b:40
> I (31477) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
~~~

jonsmirl, 2023-03-29:
My test IDs are still there

[image]

jonsmirl, 2023-03-29:
Have you turned on MAC Randomization?  Does it work on Google?

jonsmirl, 2023-03-31:
> @jonsmirl <https://github.com/jonsmirl> I was able to commission light
> example with Google on commit bafc378e68f1c32f72329ebcc986f2cfa88d67ed.
> and with esp-idf release/5.1.
>
> Versions
> Hub: Software Version : 47.9.4.447810048 Chromeast Version: 1.56.324896
> Home app: 2.66.1.13

I tried this using esp32s3...
bafc378e68f1c32f72329ebcc986f2cfa88d67ed
origin/release/v5.1
Home 2.66.1.13
Hub
Software Version : 47.9.4.447810048
Cast 1.56.324896

I am still failing

2023-03-30 18:25:47.048 18585-18585 MatterCommissioner
 com.google.android.gms.ui            I  Looking for device with
discriminator L[0xF00] [CONTEXT service_id=336 ]
2023-03-30 18:27:47.053 18585-18585 MatterCommissioner
 com.google.android.gms.ui            I  No matching devices discovered.
[CONTEXT service_id=336 ]

jonsmirl, 2023-03-31:
At the end of the BLE advertisement it is 0x00 0x80 0x00. Is that right, or are those supposed to be 16b fields?

jonsmirl, 2023-04-06:
I have made a breakthrough on this... I got so fed up with things I bought a Samsung tablet to compare to my Pixel phone. When my tablet arrived, I tried it and commissioning worked! Then I started comparing things and finally stumbled onto something which was different. The Google Play Services version was different between the two devices.  Now GPS is supposed to auto update so that shouldn't be possible. On the phone I checked GPS via the Google Play app: no way to force an update. Finally I found something by accident, looking at the settings entry for the GPS app, there is a section at the bottom which says "GPS installed via the Play" store. I tapped on it just to look -- and voila! It took me to GPS in the Play Store but this time there was an UPDATE button.  I hit UPDATE and GPS updated. But that still didn't work. Luckily I was aware that if you have GPS problems you probably need to remove and reinstall Google Home after fixing them. After I did that, my phone was able to commission a device again.

The GPS version I removed was from December and it was replaced with a version from March. This is probably why I have been having so many commissioning problems for the last couple of months.

dhrishi, 2023-04-06:
@jonsmirl That's great! Thanks for the update. This is really very helpful.

jonsmirl, 2023-04-06:
If Google had support people who I could actually interact with, I probably would have found this two months ago. But instead I just file bug reports into the Google blackhole and never hear anything again. I've pretty much stopped bothering to fill them out. Instead I have wasted weeks of time unable to work on my Android app because every time I update the ESP32 side I need to recommission and I haven't been able to do that for months.

jonsmirl, 2023-04-06:
Another clue that lead me to this.... I said -- if I can't commission via BLE, then why don't I open a commissioning window and use the code from IOS in the Pixel phone to commission over IP?  But that didn't work either. That not working convinced me something was really screwed up with Matter on my Pixel phone. I needed something to compare against which prompted me to buy the Samsung tablet as an alternative implementation (ie Smart Things). Once I had the Samsung and it worked, I went through every possible Android option looking for a difference and found one -- the GPS library.

dhrishi, 2023-04-07:
@jonsmirl Closing this issue as the original problem was solved

closed 2023-04-07 by dhrishi