espressif/esp-matter #291: Clusters used in the Project

SYESHASWINI, 2023-03-30:
Hello, Is there any way to find all the clusters used in the matter project and resources related to it through which we can generate a PICS file with the cluster information.
Thank You In advance.

jadhavrohit924, 2023-04-03:
@SYESHASWINI We don’t have any mechanisms to do so. We are working on it.

SYESHASWINI, 2023-04-03:
@jadhavrohit924 Where can we find the cluster information which are default to the specific device types like: on/off light, Switch etc.

jadhavrohit924, 2023-04-03:
@SYESHASWINI esp-matter adds only default/mandatory clusters of all device types. If you are using chip-tool, you can use command 
`./out/debug/chip-tool descriptor read server-list <destination-id> <endpoint-ids>`

closed 2023-04-05 by SYESHASWINI