espressif/esp-matter #292: Full Clone of ESP Matter SDK takes lot of time to pull all submodules and consumes GB's of memory in laptop

aggaddam, 2023-03-30:
Hi,

I downloaded the latest ESP Matter SDK version (using the shallow clone and without any install commands). I uploaded the matter SDK as it is into our git repository hosted in bitbucket. When I tried to download the Esp Matter SDK from our bitbucket repo and tried to install matter tools by running **./install.sh** in the esp-matter folder, I got the error saying not a valid git repository.  If I clone ESP Matter SDK manually from the git hub, it works fine. Not sure What am I missing here. Any thoughts?  

```
git clone --depth 1 https://github.com/espressif/esp-matter.git
cd esp-matter
git submodule update --init --depth 1
cd connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 --shallow

```

I tried with the full clone of ESP matter SDK which takes almost 14 GB of disk memory and almost takes 20-30 minutes
` git clone --recursive https://github.com/espressif/esp-matter.git`

Is there any way we can clone light version of ESP matter SDK by stripping off unnecessary modules/third-party libraries?

wqx6, 2023-03-30:
I didn't meet the error **[not a valid git repository](fatal: not a git repository: /esp-matter/connectedhomeip/connectedhomeip/../../.git/modules/connectedhomeip/connectedhomeip)**, Currently we have an issue for install.sh, which is that we does not checkout submodule `libwebsockets` required by `chip-tool`.

I am not sure why you have the error. Here are my steps:
```
git clone --depth 1 https://github.com/espressif/esp-matter.git
cd esp-matter
git submodule update --init --depth 1
cd connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 --shallow
git submodule update --init third_party/libwebsockets/repo
cd ../..
./install.sh
```

dhrishi, 2023-03-30:
>I downloaded the latest ESP Matter SDK version (using the shallow clone and without any install commands). 

How did you download this? If you download it as a zip, then it won't have git and none of the git commands would work. Hence the error is expected. To confirm, please check for `.git` folder inside `esp-matter`.

aggaddam, 2023-04-01:
@dhrishi Here are the steps

1. Clone the latest ESP-Matter SDK by running the below commands
`git clone --depth 1 https://github.com/espressif/esp-matter.git
cd esp-matter
git submodule update --init --depth 1
cd connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 --shallow`

2. I can see.git folder inside the esp-matter folder after the clone. Now, I am pushing this ESP-Matter SDK to the bitbucket repo for our development 
3. Now, the cloning the ESP-Matter SDK from out bitbucket repo and trying to run ./install.sh. It shows failure.  (I also noticed the .git folder is also not created when cloned from bitbucket repo)

(Note: I am able to install all the tools and run the examples if cloned directly from github i.e. as i mentioned in step 1)

shubhamdp, 2023-04-04:
Files starting with `.` are hidden files, and all the git config are inside `.git`, I guess when you ran `git add` command it failed to add those files and thats why you are seeing it as that error.

Why don't you add the esp-matter as a submodule? It would save space as well as easy to maintain.

aggaddam, 2023-04-04:
@shubhamdp can you provide steps for linking specific version of esp-matter SDK as submodule?

shubhamdp, 2023-04-04:
You can find the detailed documentation here https://git-scm.com/book/en/v2/Git-Tools-Submodules.

```
git submodule add https://github.com/espressif/esp-matter.git
# check git status, there will be two files staged to commit, `.gitmodules` and esp-matter (submodule)
# commit them 
```

aggaddam, 2023-04-04:
@shubhamdp Thanks for the details.

We are using a shallow clone for ESP32 platform so that required modules will be pulled out from github. otherwise, it takes a lot of time to clone the full repo of the ESP matter SDK

Here is the script we are using right now for shallow clone

`git clone --depth 1 https://github.com/espressif/esp-matter.git
cd esp-matter
git submodule update --init --depth 1
cd connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 --shallow
git submodule update --init third_party/libwebsockets/repo`

shubhamdp, 2023-04-07:
This is what we have been suggesting in our [programming guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories). What is the problem with above steps? And I cannot think a way to optimize this further right now.

One think that you can do is, since `.git` folder will be big in size you can download a zip for source code and then unpack it and upload that, but tracking the changes will be upto you.

aggaddam, 2023-04-07:
@shubhamdp
We are facing issues while setting target to esp32c3 with matter example. When we looked carefully, we noticed that mmb

dhrishi, 2023-04-17:
@aggaddam Were you able to solve this issue?

aggaddam, 2023-04-17:
@dhrishi It was not solved. I raised a separate issue in esd-idf repository. Here is the link
https://github.com/espressif/esp-idf/issues/11152

dhrishi, 2023-04-17:
Okay. Thanks! If it is being tracked there, would you mind closing it here?