espressif/esp-matter #293: Problem to use 2 endpoint node on apple home

XAT-Project, 2023-03-30:
Hello,

I would like create a node with 2 endpoints and use this with Apple Home ( in my case I use an iPhone on iOS16.4 ) with an ESP32C3.

I've tried with the generic switch example, and I succeeded in creating 2 endpoints and use 2 Buttons ( first on GPIO6 and second is the builtin button ) . However when I try to use this in Apple Home app, I see the 2 endpoint buttons but when I click on both of them, only one of the two endpoints activates. 

In addition, I try with 3 buttons, but I have the same issue, with only one endpoint being used when I click on each button.

Is this a known issue or am I missing a puzzle peace?

tldr; I can't get multiple endpoints to work. As far as I understood the docs this should be in spec. 

## App main code for matter

```cpp
...
int Setup_Button(struct mybut* button,node_t* node)
{
    // Initialise our button with the "real" component, like GPIO button 
    // builtin button.
    app_driver_handle_t button_handle = app_driver_button_init(button);  
    generic_switch::config_t switch_config;

    // Create a new endpoint for our switch. 
    endpoint_t *endpoint = generic_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE, button_handle);

    // Verify if endpoint and matter is available to merge our endpoint with 
    // the matter node.
    if (!node || !endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
        return 1;
    }

    // Get the endpoint ID to show them in terminal
    switch_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Generic Switch created with endpoint_id %d\n", switch_endpoint_id);

    // Create a cluster.
    cluster::fixed_label::config_t fl_config;
    cluster_t *fl_cluster = cluster::fixed_label::create(endpoint, &fl_config, CLUSTER_FLAG_SERVER);

    cluster::add_function_list()

    // Return the number for the register. Here, we will have 0x3B, 59 in decimal form. 
    cluster_t *cluster = cluster::get(endpoint, Switch::Id);

    ESP_LOGI(TAG, "Data for cluster %p - %lu - %lu\n\n", cluster,*cluster,*endpoint);

    // Add action availble on our matter device.
    cluster::switch_cluster::feature::momentary_switch::add(cluster);
    cluster::switch_cluster::feature::momentary_switch_release::add(cluster);
    cluster::switch_cluster::feature::momentary_switch_long_press::add(cluster);
    // cluster::switch_cluster::feature::momentary_switch_multi_press::add(cluster);
    
    
    printf("%d - %u \n\n",(int)ret,cluster::switch_cluster::feature::momentary_switch::get_id());
    
    return 0;
}

extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    // Matter config
    node::config_t node_config;
    // Matter core node
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    Setup_Button(NULL,node);
    

    struct mybut button;
    button.GPIO_PIN_VALUE = GPIO_NUM_6;

    Setup_Button(&button,node);

    /* Matter start */
    err = esp_matter::start(app_event_cb);

...
```

## Code to configure button 

```cpp

app_driver_handle_t app_driver_button_init(struct mybut* but)
{
    /* Initialize button */
    button_config_t config = button_driver_get_config();
    if(but != NULL)
    {
            config.type =  button_type_t::BUTTON_TYPE_GPIO;
            config.gpio_button_config.gpio_num = but->GPIO_PIN_VALUE;
            button_gpio_init(&config.gpio_button_config);
            config.long_press_time = 0;
            config.short_press_time = 0;
    }
    button_handle_t handle = iot_button_create(&config);
    iot_button_register_cb(handle, BUTTON_PRESS_DOWN, app_driver_button_initial_pressed, NULL);
    iot_button_register_cb(handle, BUTTON_PRESS_UP, app_driver_button_release, NULL);
    iot_button_register_cb(handle, BUTTON_LONG_PRESS_START, app_driver_button_long_pressed, NULL);
    iot_button_register_cb(handle, BUTTON_PRESS_REPEAT, app_driver_button_multipress_ongoing, NULL);
    iot_button_register_cb(handle, BUTTON_PRESS_REPEAT_DONE, app_driver_button_multipress_complete, NULL);

    return (app_driver_handle_t)handle;
}
```

## Struct mybut

```h
struct mybut 
{
      gpio_num_t GPIO_PIN_VALUE;
};
```

jonsmirl, 2023-03-30:
Check to make sure the buttons don't all have the same name.

XAT-Project, 2023-03-31:
Ok, I have checked the name buttons on Apple home and yes, both of them are called "button1".

However, I searched in Apple home and in the code how to change the name, but I couldn't find anything.

In addition, I have tried to change some attributes to see if the name changed or some other behavior changed, but nothing changed. 

Can you point me to where or how I change that parameter ?

XAT-Project, 2023-03-31:
My HomePod version is 16.4, and it is an HomePod mini.

shripad621git, 2023-03-31:
I tried the generic_switch example at my end with your configurations and see that it is working fine. The versions I use are:

- HomePod Version 16.4
- iPadOS version 16.2

newcomertv, 2023-03-31:
I have the same issue. I've tried the code above. 
- HomePod Mini Version 16.4

XAT-Project, 2023-03-31:
I have done a clean reinstall. I'm on the latest commit of main : bafc378 . Still using the same code I can see both buttons : [image]
However pressing any physical button always triggers the same endpoint.

shubhamdp, 2023-04-07:
@XAT-Project As I can see, your code only created one generic_switch and not two. You will have to add one more.

```
// This is your existing first switch
// Create a new endpoint for our switch. 
endpoint_t *endpoint = generic_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE, button_handle);

// create one more for second switch?
endpoint_t *endpoint_2 = generic_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE, button_handle);
```

shubhamdp, 2023-04-10:
Problem is in `Setup_Button()`, see below
``` C
    // Get the endpoint ID to show them in terminal
    switch_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Generic Switch created with endpoint_id %d\n", switch_endpoint_id);
```
`switch_endpoint_id` holds the value  for endpoint to interact. In our code, as there is only one button, this variable is declared as extern. ([`switch_endpoint_id` in app_driver.cpp](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_driver.cpp#L24) and [`switch_endpoint_id` in app_main.cpp](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp#L21))

When `Setup_Button()` is called twice, `switch_endpoint_id`  is set to `2` and onwards that any button press will interact with endpoint 2.

To fix this, every button needs to be hooked to a different endpoint. In you `struct mybut` you can add one more member for endpoint id and can use that in `app_driver.cpp` to interact with desired endpoint.

Meanwhile we will try to make this code a bit generic, so that it would be easy to add more buttons.

XAT-Project, 2023-04-12:
I have changed my program accordingly and it works now. 

I'm cleaning up my code and will post the modified snippets shortly. 

Thanks for your help.

XAT-Project, 2023-04-12:
@shripad621git 's snippet is very close to my implementation. Thanks for the help. I will close this issue now 😁.

closed 2023-04-12 by XAT-Project

Kristian8606, 2023-08-08:
@XAT-Project   help me with this https://github.com/espressif/esp-matter/issues/554#issue-1834704548