espressif/esp-matter #294: Generic switch BLE issue

jonsmirl, 2023-03-31:
[genswitch.log](https://github.com/espressif/esp-matter/files/11117394/genswitch.log)
~~~
I (825) CHIP[DL]: BLE host-controller synced
E (1335) chip[DL]: Configure Adv Data failed: Error Device Layer:0x000002
E (1335) chip[DL]: Disabling CHIPoBLE service due to error: Error Device Layer:0x000002
E (1335) chip[DL]: Long dispatch time: 576 ms, for event type 2
I (1345) chip[DL]: Starting ESP WiFi layer
~~~
this is on 5.1 with esp32c3 and esp32s3.
Something with the config file? The light example does not have this issue.

shubhamdp, 2023-03-31:
Hi Jon, you may have missed flashing the factory partition as mentioned in [README.md](https://github.com/espressif/esp-matter/tree/main/examples/generic_switch#1-additional-environment-setup).
```
esptool.py -p [port-name] write_flash 0x10000 mfg_binaries/20202020_3841.bin
```

If you remember, this is related to https://github.com/espressif/esp-matter/issues/211

jonsmirl, 2023-03-31:
Why doesn't this work without flashing mfg_binaries? Then the labels get set here:
https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp#L132

Can you add a QRCode to the readme with the changed discriminator/passcode?

shubhamdp, 2023-03-31:
`CONFIG_ENABLE_ESP32_DEVICE_INFO_PROVIDER` is required for reading fixed labels from the factory. Since this reads from the factory there is the dependency of `CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER` which reads discriminator/passcode related stuff from factory. That's why you need to flash factory binary in order to work.

Its a default qrcode for 3840/20202021: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT:Y.K9042C00KA0648G00

jonsmirl, 2023-03-31:
The readme say the discrim/passcode changed
https://github.com/espressif/esp-matter/tree/main/examples/generic_switch#1-additional-environment-setup

jonsmirl, 2023-03-31:
I was able to add with Apple using this:
~~~
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/out/host$ ./chip-tool payload generate-qrcode --discriminator 3841 --setup-pin-code 20202020 --vendor-id 65521 --product-id 32768 --version 0 --commissioning-mode 2 --rendezvous 2
[1680284908.817767][1234299:1234299] CHIP:TOO: QR Code: MT:Y.K9048N012Z.548G00
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/out/host$ 
~~~

After I got it into Apple, Apple says it is unrecognized device. So something wrong in generic switch app because Apple recognizes the generic switch devices in my app.

closed 2023-07-06 by jonsmirl