espressif/esp-matter #296: error: format '%X' expects argument of type 'unsigned int', but argument 7 has type 'uint32_t'

jonsmirl, 2023-03-31:
My app is getting hundreds of this compiler error after switching to IDF 5.1.
~~~
/home/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter_attribute_utils.cpp: In function 'void esp_matter::attribute::val_print(uint16_t, uint32_t, uint32_t, esp_matter_attr_val_t*)':
/home/jonsmirl/esp/esp-idf/components/log/include/esp_log.h:265:27: error: format '%X' expects argument of type 'unsigned int', but argument 7 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  265 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"
~~~
[esp51.log](https://github.com/espressif/esp-matter/files/11117872/esp51.log)

shubhamdp, 2023-03-31:
This is expected, I see a lot of format specifier mismatch there. Will fix this.

jonsmirl, 2023-03-31:
I need to add this until fixed:
~~~
target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")
~~~

closed 2023-03-31 by jonsmirl

brad-colbert, 2024-04-06:
> I need to add this until fixed:
> 
> ```
> idf_build_set_property(COMPILE_OPTIONS "-Wno-format-nonliteral;-Wno-format-security;-Wformat=0" APPEND)
> ```

Where did you add this?