espressif/esp-matter #297: Mac OS 13 chip-tool pairing failed

ZWXAllen, 2023-03-31:
I have successfully run the light demo.
`
I (901480) chip[DL]: CHIPoBLE advertising stopped
I (901490) app_main: Commissioning window closed
I (1675880) app_driver: Toggle button pressed
I (1675880) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (1675900) led_driver_ws2812: led set r:63, g:51, b:40
`

When I click on the button of ESP32-C3, the light above it turns on. However, when I use the command 'chip-tool pairing ble-wifi 0x7283 robotonly robotonly 20202021 3840', there is no response."

`
[1680246851959] [90764:272439] [CTL] Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 433F19F548F28708)
[1680246851969] [90764:272439] [BLE] NewConnection
[1680246851969] [90764:272439] [CTL] Setting wifi credentials from parameters
[1680246851969] [90764:272439] [CTL] Setting attestation nonce to random value
[1680246851969] [90764:272439] [CTL] Setting CSR nonce to random value
[1680246851969] [90764:272439] [CTL] Commission called for node ID 0x0000000000000011
[1]    90764 abort      chip-tool pairing ble-wifi 0x11 robotonly robotonly 20202021 3840
`

dhrishi, 2023-03-31:
@ZWXAllen Your log shows that the commissioning window is closed on the device console. Commissioning window is closed after 15 min as per the specification. Can you reboot the device and try again. If you still see a failure, then please upload (not paste) the chip-tool side logs

dhrishi, 2023-03-31:
@ZWXAllen Also, there's an abort in the chip-tool output. Please check if it is built correctly.

ZWXAllen, 2023-03-31:
**idf.py erase-flash flash monitor**

riting at 0x000d598a... (50 %)
Writing at 0x000db6b0... (51 %)
Writing at 0x000e14a5... (53 %)
Writing at 0x000e6e9c... (55 %)
Writing at 0x000ecddb... (57 %)
Writing at 0x000f2e52... (59 %)
Writing at 0x000f9873... (61 %)
Writing at 0x000fff0e... (63 %)
Writing at 0x00105f1d... (65 %)
Writing at 0x0010c257... (67 %)
Writing at 0x00111f6d... (69 %)
Writing at 0x00118147... (71 %)
Writing at 0x0011e743... (73 %)
Writing at 0x0012497b... (75 %)
Writing at 0x0012ad9b... (76 %)
Writing at 0x00131583... (78 %)
Writing at 0x00137a9c... (80 %)
Writing at 0x0013dd02... (82 %)
Writing at 0x00143374... (84 %)
Writing at 0x00148e80... (86 %)
Writing at 0x0014f0a6... (88 %)
Writing at 0x00155495... (90 %)
Writing at 0x0015c3c7... (92 %)
Writing at 0x00163361... (94 %)
Writing at 0x00169375... (96 %)
Writing at 0x0016f136... (98 %)
Writing at 0x0017590e... (100 %)
Wrote 1412896 bytes (844725 compressed) at 0x00020000 in 23.1 seconds (effective 489.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 178...
Writing at 0x0000c000... (100 %)
Wrote 3072 bytes (178 compressed) at 0x0000c000 in 0.1 seconds (effective 265.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x00017000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x00017000 in 0.2 seconds (effective 383.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/allen/esp/esp-matter/examples/light
Executing "/Users/allen/.espressif/python_env/idf5.1_py3.9_env/bin/python /Users/allen/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-14120 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace /Users/allen/esp/esp-matter/examples/light/build/light.elf -m '/Users/allen/.espressif/python_env/idf5.1_py3.9_env/bin/python' '/Users/allen/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbserial-14120 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1718
load:0x403cc710,len:0x974
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (30) boot: ESP-IDF v5.1-dev-4264-g81e1e65995 2nd stage bootloader
I (30) boot: compile time Mar 31 2023 17:34:52
I (31) boot: chip revision: v0.3
I (35) boot.esp32c3: SPI Speed      : 80MHz
I (40) boot.esp32c3: SPI Mode       : DIO
I (44) boot.esp32c3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (73) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (80) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (88) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (95) boot:  4 phy_init         RF data          01 01 00019000 00001000
[... 218 lines cut ...]
[1680255619516] [42356:135397] [DIS] Mdns: OnRegisterRecord flags: 0

When I run the command, an error occurs.

**>>> pairing ble-wifi 0x7283 robotonly robotonly  20202021 3840**
[1680255633350] [42356:135360] [TOO] Command: pairing ble-wifi 0x7283 robotonly robotonly 20202021 3840
[1680255633351] [42356:135397] [CTL] Stopping commissioning discovery over DNS-SD
[1680255633351] [42356:135397] [CTL] Setting attestation nonce to random value
[1680255633351] [42356:135397] [CTL] Setting CSR nonce to random value
[1680255633352] [42356:135397] [CTL] Generating NOC
[1680255633354] [42356:135397] [FP] Validating NOC chain
[1680255633357] [42356:135397] [FP] NOC chain validation successful
[1680255633357] [42356:135397] [FP] Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1680255633357] [42356:135397] [TS] Last Known Good Time: 2023-03-30T17:30:59
[1680255633357] [42356:135397] [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1680255633357] [42356:135397] [TS] Retaining current Last Known Good Time
[1680255633359] [42356:135397] [FP] Metadata for Fabric 0x1 persisted to storage.
[1680255633360] [42356:135397] [TS] Committing Last Known Good Time to storage: 2023-03-30T17:30:59
[1680255633362] [42356:135397] [CTL] Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 91FBB7E965E05083)
[1680255633372] [42356:135397] [BLE] NewConnection
[1680255633372] [42356:135397] [CTL] Setting wifi credentials from parameters
[1680255633372] [42356:135397] [CTL] Setting attestation nonce to random value
[1680255633372] [42356:135397] [CTL] Setting CSR nonce to random value
[1680255633372] [42356:135397] [CTL] Commission called for node ID 0x0000000000007283
[1]    42356 abort      chip-tool interactive start

ZWXAllen, 2023-03-31:
But I can configure the network and control the light in ESP32C3.

> matter esp wifi connect robotonly robotonly

> matter esp attribute set 0x1 0x6 0x0 1

I (13440) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (13440) led_driver_ws2812: led set r:63, g:51, b:40
Done
> matter esp attribute set 0x1 0x6 0x0 0

I (15260) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 0 **********
I (15260) led_driver_ws2812: led set r:0, g:0, b:0
Done
>

ZWXAllen, 2023-03-31:
When ESP32C3 is connected to the network, I usually use "chip-tool pairing onnetwork 0x11".
 
 [1680255704610] [44373:141723] [DL] KVS will be written to: file:///tmp/chip_kvs
[1680255704620] [44373:141723] [DL] _Init
[1680255704620] [44373:141723] [IN] UDP::Init bind&listen port=0
[1680255704621] [44373:141723] [IN] UDP::Init bound to port=63772
[1680255704621] [44373:141723] [IN] UDP::Init bind&listen port=0
[1680255704621] [44373:141723] [IN] UDP::Init bound to port=51202
[1680255704621] [44373:141723] [IN] BLEBase::Init - setting/overriding transport
[1680255704621] [44373:141723] [IN] TransportMgr initialized
[1680255704621] [44373:141723] [FP] Initializing FabricTable from persistent storage
[1680255704621] [44373:141723] [TS] Last Known Good Time: 2023-03-30T17:30:59
[1680255704621] [44373:141723] [FP] Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x91FBB7E965E05083, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1680255704623] [44373:141723] [ZCL] Using ZAP configuration...
[1680255704623] [44373:141723] [CTL] System State Initialized...
[1680255704623] [44373:141723] [CTL] Stopping commissioning discovery over DNS-SD
[1680255704623] [44373:141723] [CTL] Setting attestation nonce to random value
[1680255704623] [44373:141723] [CTL] Setting CSR nonce to random value
[1680255704624] [44373:141767] [DL] _OnPlatformEvent
[1680255704624] [44373:141767] [CTL] Stopping commissioning discovery over DNS-SD
[1680255704624] [44373:141767] [CTL] Setting attestation nonce to random value
[1680255704624] [44373:141767] [CTL] Setting CSR nonce to random value
[1680255704625] [44373:141767] [CTL] Generating NOC
[1680255704626] [44373:141767] [FP] Validating NOC chain
[1680255704630] [44373:141767] [FP] NOC chain validation successful
[1680255704630] [44373:141767] [FP] Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1680255704630] [44373:141767] [TS] Last Known Good Time: 2023-03-30T17:30:59
[1680255704630] [44373:141767] [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1680255704630] [44373:141767] [TS] Retaining current Last Known Good Time
[1680255704632] [44373:141767] [FP] Metadata for Fabric 0x1 persisted to storage.
[1680255704633] [44373:141767] [TS] Committing Last Known Good Time to storage: 2023-03-30T17:30:59
[1680255704634] [44373:141767] [CTL] Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 91FBB7E965E05083)
[1680255704645] [44373:141767] [DIS] Browsing for: _matterc._udp
[1680255704646] [44373:141767] [DIS] Mdns: OnBrowseAdd  name: 837071B5DF73E980, type: _matterc._udp., domain: local., interface: 7
[1680255704646] [44373:141767] [DIS] Resolve type=_matterc._udp name=837071B5DF73E980 interface=7
[1680255704646] [44373:141767] [DIS] Mdns: OnResolve flags: 0, interface: 7, fullname: 837071B5DF73E980._matterc._udp.local., hostname: 58CF790C72F0.local., port: 42005
[1680255704646] [44373:141767] [DIS] Mdns : OnNewInterface hostname:58CF790C72F0.local. fullname:837071B5DF73E980._matterc._udp.local. interface: 7 port: 5540 TXT:"VP=65521+32768,T=1,D=3840,CM=1,PH=33,PI="
[1680255704646] [44373:141767] [DIS] Mdns: OnGetAddrInfo flags: 1073741827, interface: 7, hostname: 58CF790C72F0.local.
[1680255704646] [44373:141767] [DIS] Mdns: OnNewAddress interface: 7 ip:fe80::5acf:79ff:fe0c:72f0
[1680255704646] [44373:141767] [DIS] Mdns: OnGetAddrInfo flags: 1073741826, interface: 7, hostname: 58CF790C72F0.local.
[1680255704646] [44373:141767] [DIS] Mdns: OnNewAddress interface: 7 ip:192.168.0.14
[1680255704646] [44373:141767] [DIS] Mdns: Resolve success on interface 7
[1680255704646] [44373:141767] [DIS] Discovered node:
[1680255704646] [44373:141767] [DIS]    Hostname: 58CF790C72F0
[1680255704646] [44373:141767] [DIS]    IP Address #1: fe80::5acf:79ff:fe0c:72f0
[1680255704646] [44373:141767] [DIS]    IP Address #2: 192.168.0.14
[1680255704646] [44373:141767] [DIS]    Port: 5540
[1680255704646] [44373:141767] [DIS]    Mrp Interval idle: not present
[1680255704646] [44373:141767] [DIS]    Mrp Interval active: not present
[1680255704646] [44373:141767] [DIS]    TCP Supported: 1
[1680255704646] [44373:141767] [DIS]    Vendor ID: 65521
[1680255704646] [44373:141767] [DIS]    Product ID: 32768
[1680255704646] [44373:141767] [DIS]    Long Discriminator: 3840
[1680255704646] [44373:141767] [DIS]    Pairing Hint: 33
[1680255704646] [44373:141767] [DIS]    Instance Name: 837071B5DF73E980
[1680255704646] [44373:141767] [DIS]    Commissioning Mode: 1
[1680255704646] [44373:141767] [TOO] Discovered Device: fe80::5acf:79ff:fe0c:72f0:5540
[1680255704647] [44373:141767] [DIS] Mdns: Browse failure (kDNSServiceErr_Timeout)
[1680255704649] [44373:141767] [IN] SecureSession[0x6000006c0000]: Allocated Type:1 LSID:8298
[1680255704649] [44373:141767] [SC] Assigned local session key ID 8298
[1680255704649] [44373:141767] [EM] <<< [E:35755i S:0 M:49709851] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
[1680255704649] [44373:141767] [IN] (U) Sending msg 49709851 to IP address 'UDP:[fe80::5acf:79ff:fe0c:72f0%en1]:5540'
[1680255704650] [44373:141767] [SC] Sent PBKDF param request
[1680255704650] [44373:141767] [CTL] Setting attestation nonce to random value
[1680255704650] [44373:141767] [CTL] Setting CSR nonce to random value
[1680255704650] [44373:141767] [CTL] Commission called for node ID 0x0000000000000011
[... cut ...]
[1680255710452] [44373:141723] [BLE] CancelConnection
[1680255710452] [44373:141723] [FP] Shutting down FabricTable
[1680255710452] [44373:141723] [TS] Pending Last Known Good Time: 2023-03-30T17:30:59
[1680255710452] [44373:141723] [TS] Previous Last Known Good Time: 2023-03-30T17:30:59
[1680255710452] [44373:141723] [TS] Reverted Last Known Good Time to previous value
[1680255710452] [44373:141723] [DL] Inet Layer shutdown
[1680255710452] [44373:141723] [DL] BLE shutdown
[1680255710452] [44373:141723] [DL] System Layer shutdown

ZWXAllen, 2023-03-31:
ESP32C3 LOGS

I (127575) esp_netif_handlers: sta ip: 192.168.0.14, mask: 255.255.255.0, gw: 192.168.0.1
I (127575) chip[DL]: IP_EVENT_STA_GOT_IP
I (127585) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.0.14/255.255.255.0 gateway 192.168.0.1
I (127595) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (127595) app_main: Interface IP Address changed
I (127595) chip[DIS]: Updating services using commissioning mode 1
I (127605) chip[DIS]: CHIP minimal mDNS started advertising.
I (127625) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (127625) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 837071B5DF73E980.
I (127645) chip[DIS]: mDNS service published: _matterc._udp
I (153445) chip[EM]: >>> [E:35755r S:0 M:49709851] (U) Msg RX from 0:3B63FD48419AB150 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (153465) chip[EM]: <<< [E:35755r S:0 M:201763628 (Ack:49709851)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (153475) chip[IN]: (U) Sending msg 201763628 to IP address 'UDP:[FE80::C5:F0ED:14A:183%st2]:63772'
I (153485) chip[EM]: <<< [E:35755r S:0 M:201763629 (Ack:49709851)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
I (153505) chip[IN]: (U) Sending msg 201763629 to IP address 'UDP:[FE80::C5:F0ED:14A:183%st2]:63772'
I (153845) chip[EM]: >>> [E:35755r S:0 M:49709851] (U) Msg RX from 0:3B63FD48419AB150 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (153855) chip[EM]: <<< [E:35755r S:0 M:201763630 (Ack:49709851)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (153865) chip[IN]: (U) Sending msg 201763630 to IP address 'UDP:[FE80::C5:F0ED:14A:183%st2]:63772'
I (153885) chip[EM]: >>> [E:35755r S:0 M:49709852 (Ack:201763629)] (U) Msg RX from 0:3B63FD48419AB150 [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
I (153895) chip[EM]: <<< [E:35755r S:0 M:201763631 (Ack:49709852)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (153905) chip[IN]: (U) Sending msg 201763631 to IP address 'UDP:[FE80::C5:F0ED:14A:183%st2]:63772'
I (154545) chip[EM]: <<< [E:35755r S:0 M:201763632 (Ack:49709852)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)
I (154555) chip[IN]: (U) Sending msg 201763632 to IP address 'UDP:[FE80::C5:F0ED:14A:183%st2]:63772'
E (154555) chip[DL]: Long dispatch time: 677 ms, for event type 3
I (154765) chip[EM]: >>> [E:35755r S:0 M:49709853 (Ack:201763632)] (U) Msg RX from 0:3B63FD48419AB150 [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
I (154775) chip[EM]: <<< [E:35755r S:0 M:201763633 (Ack:49709853)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (154785) chip[IN]: (U) Sending msg 201763633 to IP address 'UDP:[FE80::C5:F0ED:14A:183%st2]:63772'
I (154805) chip[EM]: <<< [E:35755r S:0 M:201763634 (Ack:49709853)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (154815) chip[IN]: (U) Sending msg 201763634 to IP address 'UDP:[FE80::C5:F0ED:14A:183%st2]:63772'
I (154815) chip[SC]: SecureSession[0x3fc9d8a0]: Moving from state 'kEstablishing' --> 'kActive'
I (154825) chip[SVR]: Commissioning completed session establishment step
I (154845) chip[DIS]: Updating services using commissioning mode 0
I (154845) chip[DIS]: CHIP minimal mDNS started advertising.
I (154865) chip[SVR]: Device completed Rendezvous process
E (154875) chip[DL]: Long dispatch time: 111 ms, for event type 3
I (154875) app_main: Commissioning session started
I (154885) NimBLE: GAP procedure initiated: stop advertising.

I (154885) chip[DL]: CHIPoBLE advertising stopped
I (154895) app_main: Commissioning window closed
I (155015) chip[EM]: >>> [E:35756r S:3466 M:265019293] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (155025) chip[EM]: >>> [E:35755r S:0 M:49709854 (Ack:201763634)] (U) Msg RX from 0:3B63FD48419AB150 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (155035) chip[DL]: NVS set: chip-config/reg-location = 0 (0x0)
I (155055) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 0 **********
I (155065) chip[EM]: <<< [E:35756r S:3466 M:28731189 (Ack:265019293)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (155075) chip[IN]: (S) Sending msg 28731189 on secure session with LSID: 3466
I (155135) chip[EM]: >>> [E:35757r S:3466 M:265019294] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (155145) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x00000030
I (155145) chip[FS]: GeneralCommissioning: Received ArmFailSafe (60s)
I (155165) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 3 **********
I (155175) chip[EM]: <<< [E:35757r S:3466 M:28731190 (Ack:265019294)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (155185) chip[IN]: (S) Sending msg 28731190 on secure session with LSID: 3466
I (155195) chip[EM]: >>> [E:35756r S:3466 M:265019295 (Ack:28731189)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[... cut ...]
I (158995) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 91FBB7E965E05083-0000000000000011.
I (159015) chip[DIS]: mDNS service published: _matter._tcp
I (159015) chip[SC]: SecureSession[0x3fc9d8a0]: Moving from state 'kActive' --> 'kPendingEviction'
I (159025) app_main: Commissioning complete
I (159025) esp_matter_core: Commissioning Complete
I (159035) app_main: Commissioning window closed
I (159035) chip[EM]: >>> [E:35766r S:3467 M:154956327 (Ack:183024268)] (S) Msg RX from 1:000000000001B669 [5083] --- Type 0000:10 (SecureChannel:StandaloneAck)

shubhamdp, 2023-03-31:
@ZWXAllen Its hard to analyse the logs in the comment, can you please put these logs into separate files and attach the files here.

ZWXAllen, 2023-03-31:
[ESP32.txt](https://raw.githubusercontent.com/ZWXAllen/imgs/master/ESP32.txt)
[ChipTool.txt](https://raw.githubusercontent.com/ZWXAllen/imgs/master/ChipTool.txt)

jadhavrohit924, 2023-04-03:
@ZWXAllen Looks like same issue has been discussed here https://github.com/project-chip/connectedhomeip/issues/23435. 

Can you please try the solution mentioned [here](https://github.com/project-chip/connectedhomeip/issues/23435#issuecomment-1332636700) ?

andycarle, 2023-04-04:
I'm seeing this same failure on macOS 12.6.3. 

```console
CHIP:CTL: Commission called for node ID 0x0000000000001236
[1]    13896 abort      ./chip-tool interactive start
```

dhrishi, 2023-04-27:
@ZWXAllen @andycarle IMO, for better support, the aborts with Darwin chip-tool should be reported in the [connectedhomeip repo](https://github.com/project-chip/connectedhomeip/issues) so that folks from Apple can take a look.

dhrishi, 2023-05-06:
Closing this assuming this is or will be reported in connectedhomeip repo

closed 2023-05-06 by dhrishi