espressif/esp-matter #299: Tool to make string for QRCode

jonsmirl, 2023-03-31:
Where is a tool to take a discriminator and passcode and compute the string needed to generate the qrcode?
A string like this: MT:Y.K9042C00KA0648G00

dhrishi, 2023-03-31:
@jonsmirl Please check point 4. Command Usage [here](https://github.com/espressif/esp-matter/issues/159#issuecomment-1377236626)

jonsmirl, 2023-03-31:
This is because the code in the generic switch example has been changed to 20202020/3841

jonsmirl, 2023-03-31:
~~~
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/out/host$ ./chip-tool payload generate-qrcode --discriminator 3841 --setup-pin-code 20202020 --vendor-id 65521 --product-id 32768 --version 0 --commissioning-mode 2 --rendezvous 5
[1680284908.817767][1234299:1234299] CHIP:TOO: QR Code: MT:Y.K90W82112Z.548G00
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/out/host$ 
~~~

closed 2023-03-31 by jonsmirl