espressif/esp-matter #301: Callback for Commands / atomic sets of Attribute updates?

andycarle, 2023-03-31:
Is there any mechanism in the Espressif SDK for Matter to directly receive Matter Commands or to receive atomic sets of Attribute updates generated from a Command? 

For example, imagine that a light switch issues a `MoveToLevelWithOnOff` (Cluster 0x0008, Command 0x04). As best I can tell, this is only propagated through the `esp-matter` SDK with a series of callbacks to `app_attribute_update_cb`. The `CurrentLevel` Attribute (0x0008, 0x0000) may come through first, followed by the `RemainingTime` Attribute (0x0008, 0x0001) which describes how long the level transition should take. In this scenario, if the application reacted to the `CurrentLevel` change first before knowing that a transition time was coming later, it will have executed the `MoveToLevelWithOnOff` Command incorrectly.

I see that the Matter SDK has `MatterPreCommandReceivedCallback` and `MatterPostCommandReceivedCallback` callbacks that could partially be used to reconstruct the incoming command. But these do not seem to be incorporated into the `esp-matter` SDK. Nor do they contain all of the data fields that come in as part of the Command.

So, what is the `esp-matter` SDK approved way to capture these command transaction semantics?

jonsmirl, 2023-04-01:
esp-matter is not doing that, that is simply how the CHIP SDK works. The state machines are here: 
https://github.com/project-chip/connectedhomeip/tree/master/src/app/clusters/level-control
https://github.com/project-chip/connectedhomeip/tree/master/src/app/clusters/on-off-server

shripad621git, 2023-04-06:
The Espressif’s SDK for Matter handles the commands being executed by calling the corresponding 
callbacks in the Matter repository.
For e.g.: As mentioned, for the command ```MoveToLeveLWithOnOff```, it is handled by calling
```emberAfLevelControlClusterMoveToLevelWithOnOffCallback``` in the CHIP repository.
The state machine for level control is implemented in CHIP and in the case of the example command specified above, as per implementation, it takes into account the transition time when updating the current level attribute. 
In case of attributes being updated, the attribute update callback is triggered for each attribute that is being updated .We receive the callbacks with types ```PRE_UPDATE``` and ```POST_UPDATE``` before and after the attribute is updated  
respectively.
In case you want to implement a custom cluster which has some attributes being updated on the receipt of a custom command ,you can use the callback with type```PRE_UPDATE``` to perform some action before updation and update the attributes and based on the attributes changed in ```PRE_UPDATE```, perform required actions in ```POST_UPDATE```.

shripad621git, 2023-04-17:
Can we close this issue if your issue is resolved?

andycarle, 2023-04-17:
Yes. Thank you both for your assistance.

closed 2023-04-17 by andycarle