espressif/esp-matter #302: Getting error while setting target esp32c3 with ESP matter light Example

aggaddam, 2023-04-01:
I got the below error when i tried to set target as esp32 c3 with light example in ESP-Matter SDK

`t     managed_components  README.md          sdkconfig.defaults.esp32c2  sdkconfig.defaults.esp32s2
dependencies.lock  partitions.csv      sdkconfig          sdkconfig.defaults.esp32c6  sdkconfig.defaults.ext_plat_ci
agaddam@agaddam-Latitude-5310:~/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light$ idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig will be created.
Running cmake in directory /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/agaddam/.espressif/python_env/idf5.1_py3.10_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=1 /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light"...
-- Existing sdkconfig '/home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/sdkconfig' renamed to '/home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/sdkconfig.old'.
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Component directory /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
Processing 6 dependencies:
[1/6] espressif/button (2.5.0)
[2/6] espressif/esp_secure_cert_mgr (2.2.1)
[3/6] espressif/json_parser (1.0.0)
[4/6] espressif/led_strip (1.0.0)
[5/6] espressif/mdns (1.0.9)
[6/6] idf (5.1.0)
-- Found Python3: /home/agaddam/.espressif/python_env/idf5.1_py3.10_env/bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- Project sdkconfig file /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/sdkconfig
Loading defaults file /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/sdkconfig.defaults...
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/agaddam/.espressif/python_env/idf5.1_py3.10_env/bin/python (found version "3.10.6") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "light" version: v1.0
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Components: app_bridge app_reset app_trace app_update bootloader bootloader_support bt chip cmock console cxx device driver efuse esp-tls esp32_mbedtls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_matter esp_matter_bridge esp_matter_console esp_matter_controller esp_matter_rainmaker esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__button espressif__esp_secure_cert_mgr espressif__json_parser espressif__led_strip espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 iot_button json led_driver log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/common/app_bridge /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/common/app_reset /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/app_trace /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/app_update /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bootloader /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bootloader_support /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/bt /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/cmock /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/console /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/cxx /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/device_hal/device /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/driver /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/efuse /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp-tls /home/agaddam/Documents/ws_ac_li_ac_ [... cut ...]
CMake Error at /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/mbedtls/mbedtls/library/CMakeLists.txt:277 (add_library):
  No SOURCES given to target: mbedtls

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1, output of the command is in the /home/agaddam/Documents/ws_ac_li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-matter/examples/light/build/log/idf_py_stderr_output_41682 and /`

jonsmirl, 2023-04-01:
when you installed the 5.1IDF did you do 'git submodules update --init'?

I gave up on IDF5.1 and went back to IDF4.4. I was getting some memory corruption that I could not identify the source of.

aggaddam, 2023-04-03:
@jonsmirl 
I cloned using this command 

`git clone --recurse-submodules -b release/v5.1 https://github.com/espressif/esp-idf.git`

jadhavrohit924, 2023-04-03:
@aggaddam Please use `. ./install.sh` command again after cloning into release/v5.1.

aggaddam, 2023-04-03:
@jadhavrohit924 Tried again running . ./install.sh but still the same error exists

shubhamdp, 2023-04-03:
@aggaddam Can you please share the version of `esptool`?
```
source $IDF_PATH/export.sh
python3 -m pip show esptool
```

aggaddam, 2023-04-03:
@shubhamdp Here it is

shubhamdp, 2023-04-04:
@aggaddam `esptool` version is pretty new, so no problem with that. Also, have to execute the `install.sh` script and not source it, sorry I missed it earlier.

For IDF, as per documentation, we have tested esp-matter on IDF commit [bb9200ace](https://github.com/espressif/esp-idf/tree/bb9200ace) from release/v5.1.

Can you please checkout that specific commit, update the submodules recursively, run install.sh and try again?
```
cd esp-idf
git checkout bb9200ace
git submodule --update --init --recursive
./install.sh

cd esp-matter
source esp-idf/export.sh
source export.sh
cd examples/light
idf.py ser-target esp32c3 build
```

dhrishi, 2023-04-05:
@aggaddam It should ideally work with the above steps. Please close this issue if it works

dhrishi, 2023-04-12:
@aggaddam I am closing this now. Please re-open by uploading (not pasting) appropriate logs if you are still unable to compile by following all the steps in the documentation.

closed 2023-04-12 by dhrishi