espressif/esp-matter #304: Build error after modifying compile level

maenkai, 2023-04-03:
IDF-VERSION = V4.4.2
Modify compile level ： Og -> O2

cmake error:
```shell
-- Adding linker script /home/mek/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/mek/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/mek/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
CMake Error at /home/mek/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/CMakeLists.txt:45 (message):
  CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE shall be set

maenkai, 2023-04-03:
This configuration is OK for idf-V4.3.1 
There was a problem in idf-V4.4.2
```shell
CONFIG_COMPILER_OPTIMIZATION=2
CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_ENABLE

maenkai, 2023-04-07:
1.Setting this configuration is fine, but it seems like you won't be able to fire assert after configuring this item
2.I want to keep assert
3.This problem does not occur in V4.3.1

shubhamdp, 2023-04-07:
As I see this, its not at all related to IDF, see here: https://github.com/project-chip/connectedhomeip/blob/115eb6d384b587b84bd90f51b6ecae06dd1c9b4f/config/esp32/components/chip/CMakeLists.txt#L40-L49

This code checks if there is any sort of optimizations, it explicitly checks if assertions are disabled.

shubhamdp, 2023-04-07:
@maenkai PR: https://github.com/project-chip/connectedhomeip/pull/11775 mentions more details why the above changes were made.

closed 2023-04-10 by maenkai