espressif/esp-matter #305: Why does the memory increase after paired

sidhoo2, 2023-04-03:
W (16270) main_logic: current free heap:21.75K inter free heap:15.50K min free heap:13.90K
W (18280) main_logic: current free heap:80.34K inter free heap:74.08K min free heap:12.41K

log:
I (13460) chip[DL]: Done driving station state, nothing else to do...
E (13460) chip[DL]: Long dispatch time: 128 ms, for event type 49152
I (13470) chip[DL]: WIFI_EVENT_AP_STOP
I (13470) chip[DL]: WiFi AP state change: Deactivating -> NotActive
I (13480) phy: chan:1,max_power:80
I (13480) phy: chan:2,max_power:80
I (13490) phy: chan:3,max_power:80
I (13490) phy: chan:4,max_power:80
I (13500) phy: chan:5,max_power:80
I (13500) phy: chan:6,max_power:80
I (13500) phy: chan:7,max_power:80
I (13510) phy: chan:8,max_power:80
I (13510) phy: chan:9,max_power:80
I (13520) phy: chan:10,max_power:80
I (13520) phy: chan:11,max_power:80
I (13530) phy: chan:12,max_power:80
I (13530) phy: chan:13,max_power:80
I (13530) phy: chan:14,max_power:80
I (13540) chip[DIS]: Advertise operational node BDB661861EBF7F51-00000000D5154FCB
I (13550) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: BDB661861EBF7F51-00000000D5154FCB.
I (13560) chip[DIS]: mDNS service published: _matter._tcp
I (13560) chip[SVR]: Operational advertising enabled
I (13570) chip[DL]: Updating advertising data
I (13600) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
W (13670) main_logic: current free heap:28.57K inter free heap:22.31K min free heap:14.02K
W (13780) wifi:<ba-add>idx:0 (ifx:0, c8:bf:4c:3e:a3:41), tid:6, ssn:2, winSize:64
I (14470) chip[DL]: IP_EVENT_GOT_IP6
I (14470) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:7edf:a1ff:fec1:d880
I (14470) app_main: Interface IP Address changed
I (14480) chip[DIS]: Updating services using commissioning mode 0
I (14480) chip[DIS]: CHIP minimal mDNS started advertising.
I (14490) chip[DIS]: Advertise operational node BDB661861EBF7F51-00000000D5154FCB
I (14500) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: BDB661861EBF7F51-00000000D5154FCB.
I (14510) chip[DIS]: mDNS service published: _matter._tcp
I (14620) esp_netif_handlers: sta ip: 192.168.31.58, mask: 255.255.255.0, gw: 192.168.31.1
I (14620) chip[DL]: IP_EVENT_STA_GOT_IP
I (14620) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.31.58/255.255.255.0 gateway 192.168.31.1
I (14630) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (14640) app_main: Interface IP Address changed
I (14640) chip[DIS]: Updating services using commissioning mode 0
I (14650) chip[DIS]: CHIP minimal mDNS started advertising.
I (14660) chip[DIS]: Advertise operational node BDB661861EBF7F51-00000000D5154FCB
I (14660) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: BDB661861EBF7F51-00000000D5154FCB.
I (14680) chip[DIS]: mDNS service published: _matter._tcp
I (14680) ROUTE_HOOK: Hook already installed on netif, skip...
W (14700) wifi:<ba-add>idx:1 (ifx:0, c8:bf:4c:3e:a3:41), tid:0, ssn:0, winSize:64
I (14700) chip[EM]: >>> [E:43789r M:42237911] (U) Msg RX from 0:453B2DB24F89FD9B [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (14710) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3fc9c560
I (14720) chip[IN]: CASE Server disabling CASE session setups
I (14720) chip[EM]: <<< [E:43789r M:138685889 (Ack:42237911)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (14740) chip[IN]: (U) Sending msg 138685889 to IP address 'UDP:[FE80::1CD0:3056:6AA4:235B%st1]:54069'
I (14750) chip[SC]: Received Sigma1 msg
I (14750) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x00000000D5154FCB
I (15240) chip[EM]: <<< [E:43789r M:138685890 (Ack:42237911)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (15240) chip[IN]: (U) Sending msg 138685890 to IP address 'UDP:[FE80::1CD0:3056:6AA4:235B%st1]:54069'
I (15250) chip[SC]: Sent Sigma2 msg
E (15250) chip[DL]: Long dispatch time: 556 ms, for event type 3
I (15520) chip[EM]: >>> [E:43789r M:42237912 (Ack:138685890)] (U) Msg RX from 0:453B2DB24F89FD9B [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (15520) chip[EM]: <<< [E:43789r M:138685891 (Ack:42237912)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (15530) chip[IN]: (U) Sending msg 138685891 to IP address 'UDP:[FE80::1CD0:3056:6AA4:235B%st1]:54069'
I (15540) chip[SC]: Received Sigma3 msg
E (15550) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (16220) chip[EM]: <<< [E:43789r M:138685892 (Ack:42237912)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (16230) chip[IN]: (U) Sending msg 138685892 to IP address 'UDP:[FE80::1CD0:3056:6AA4:235B%st1]:54069'
I (16240) chip[SC]: SecureSession[0x3fc9adf0]: Moving from state 'kEstablishing' --> 'kActive'
I (16240) chip[IN]: CASE Session established to peer: <1E2BC527C2DB3FA5, 1>
[... 58 lines cut ...]
I (17040) chip[IN]: (S) Sending msg 69914674 on secure session with LSID: 10482
I (17070) chip[EM]: >>> [E:43796r M:133822349 (Ack:69914674)] (S) Msg RX from 1:1E2BC527C2DB3FA5 [7F51] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (17260) chip[EM]: >>> [E:43797r M:133822350] (S) Msg RX from 1:1E2BC527C2DB3FA5 [7F51] --- Type 0001:0a (IM:TimedRequest)
I (17260) chip[EM]: <<< [E:43797r M:69914675 (Ack:133822350)] (S) Msg TX to 1:1E2BC527C2DB3FA5 [7F51] --- Type 0001:01 (IM:StatusResponse)
I (17270) chip[IN]: (S) Sending msg 69914675 on secure session with LSID: 10482
I (17300) chip[EM]: >>> [E:43797r M:133822351 (Ack:69914675)] (S) Msg RX from 1:1E2BC527C2DB3FA5 [7F51] --- Type 0001:08 (IM:InvokeCommandRequest)
I (17300) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x0000003C
I (17310) chip[ZCL]: Received command to open commissioning window
I (17320) chip[DIS]: Updating services using commissioning mode 2
I (17330) chip[DIS]: CHIP minimal mDNS started advertising.
I (17340) chip[DIS]: Advertise operational node BDB661861EBF7F51-00000000D5154FCB
I (17340) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: BDB661861EBF7F51-00000000D5154FCB.
I (17360) chip[DIS]: mDNS service published: _matter._tcp
I (17360) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=2740/10 cm=2
I (17370) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 6C042797FAD65741.
I (17390) chip[DIS]: mDNS service published: _matterc._udp
I (17390) chip[ZCL]: Commissioning window is now open
I (17390) chip[EM]: <<< [E:43797r M:69914676 (Ack:133822351)] (S) Msg TX to 1:1E2BC527C2DB3FA5 [7F51] --- Type 0001:09 (IM:InvokeCommandResponse)
I (17410) chip[IN]: (S) Sending msg 69914676 on secure session with LSID: 10482
E (17420) chip[DL]: Long dispatch time: 118 ms, for event type 3
I (17420) app_main: Commissioning window opened
I (17440) chip[EM]: >>> [E:43797r M:133822352 (Ack:69914676)] (S) Msg RX from 1:1E2BC527C2DB3FA5 [7F51] --- Type 0000:10 (SecureChannel:StandaloneAck)
W (18280) main_logic: current free heap:80.34K inter free heap:74.08K min free heap:12.41K
W (20280) main_logic: current free heap:80.34K inter free heap:74.08K min free heap:12.41K
W (22280) main_logic: current free heap:80.34K inter free heap:74.08K min free heap:12.41K

shubhamdp, 2023-04-03:
@sidhoo2 By default, BLE is only used for commissioning, therefore we deactivate the BLE functionality and recover all the memory allocated to it, as shown in the following code. https://github.com/espressif/esp-matter/blob/3a65dfc75398bb3e4ce58418011499cd794a0586/components/esp_matter/esp_matter_core.cpp#L859-L891

If you want to use the BLE even after commissioning then you can disable `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` and free memory won't go up.

sidhoo2, 2023-04-03:
> @sidhoo2 By default, BLE is only used for commissioning, therefore we deactivate the BLE functionality and recover all the memory allocated to it, as shown in the following code. [...]

@shubhamdp you are right , but the func not implemented,  and  who spend the ram before paired ?
`esp_err_t esp_bt_controller_mem_release(esp_bt_mode_t mode)
{
    ESP_LOGW(BTDM_LOG_TAG, "%s not implemented, return OK", __func__);
    return ESP_OK;
}`

shubhamdp, 2023-04-04:
> you are right , but the func not implemented, and who spend the ram before paired ?

For ESP32, `esp_bt_mem_release()` API releases memory, but for other SoCs, call to deinit does that work. On BLE initialization it allocates the memory it require.

> In addition，just run the matter plug app, there is only 68K free heap left , did anybody meet this issue

If you see, the observations are pretty old, almost 9 months back, and since we are continuously making changes, its hard to update it regularly.

sidhoo2, 2023-04-04:
@shubhamdp  i see, and is there any method to optimize the RAM usage?

shubhamdp, 2023-04-04:
You can perform the configuration optimizations, few are listed [here](https://github.com/project-chip/connectedhomeip/blob/master/examples/temperature-measurement-app/esp32/sdkconfig.optimize.defaults).

You can also refer the IDF [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html).

dhrishi, 2023-04-04:
Closing as the original question is answered.

closed 2023-04-04 by dhrishi