espressif/esp-matter #306: Timeout when calling thermostat commands

markusp00, 2023-04-04:
## Issue
I have implemented the thermostat cluster on an endpoint. The Setpoint Raise/Lower command works as expected. When I'm trying to run the SetWeeklySchedule, GetWeeklySchedule, and ClearWeeklySchedule commands from chip-tool, I receive timeout errors. Any ideas for why the commands are not executed successfully? (It might be caused by deficient configuration from my side, if so I would greatly appreciate some guidance).

## Log from chip-tool
```
>>> thermostat clear-weekly-schedule 0x7283 0x2
[1680634018.459424][10926:10928] CHIP:TOO: Sending command to node 0x7283
[1680634018.459491][10926:10928] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000007283]
[1680634018.459498][10926:10928] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1680634018.459507][10926:10928] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 1 --> 2
[1680634018.570283][10926:10928] CHIP:DIS: UDP:[fe80::7204:1dff:fea9:1114%enp0s3]:5540: new best score: 3
[1680634018.570367][10926:10928] CHIP:DIS: Lookup clearing interface for non LL address
[1680634018.570382][10926:10928] CHIP:DIS: UDP:[fd43:6727:9396:9685:7204:1dff:fea9:1114%enp0s3]:5540: new best score: 6
[1680634018.570424][10926:10928] CHIP:DIS: UDP:192.168.50.56%enp0s3:5540: score has not improved: 2
[1680634018.570431][10926:10928] CHIP:DIS: Checking node lookup status after 111 ms
[1680634018.570434][10926:10928] CHIP:DIS: Keeping DNSSD lookup active
[1680634018.570502][10926:10928] CHIP:DIS: UDP:[fe80::7204:1dff:fea9:1114%enp0s3]:5540: score has not improved: 3
[1680634018.570574][10926:10928] CHIP:DIS: UDP:[fd43:6727:9396:9685:7204:1dff:fea9:1114%enp0s3]:5540: score has not improved: 6
[1680634018.570616][10926:10928] CHIP:DIS: UDP:192.168.50.56%enp0s3:5540: score has not improved: 2
[1680634018.570622][10926:10928] CHIP:DIS: Checking node lookup status after 111 ms
[1680634018.570625][10926:10928] CHIP:DIS: Keeping DNSSD lookup active
[1680634018.660954][10926:10928] CHIP:DIS: Checking node lookup status after 202 ms
[1680634018.661015][10926:10928] CHIP:DIS: OperationalSessionSetup[1:0000000000007283]: Updating device address to UDP:[fd43:6727:9396:9685:7204:1dff:fea9:1114]:5540 while in state 2
[1680634018.661025][10926:10928] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 2 --> 3
[1680634018.661099][10926:10928] CHIP:IN: SecureSession[0x7f478c038000]: Allocated Type:2 LSID:24658
[1680634018.661138][10926:10928] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000007283
[1680634018.661800][10926:10928] CHIP:EM: <<< [E:64752i M:251970741] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[1680634018.662079][10926:10928] CHIP:IN: (U) Sending msg 251970741 to IP address 'UDP:[fd43:6727:9396:9685:7204:1dff:fea9:1114]:5540'
[1680634018.662268][10926:10928] CHIP:SC: Sent Sigma1 msg
[1680634018.662290][10926:10928] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 3 --> 4
[1680634018.931396][10926:10928] CHIP:EM: >>> [E:64752i M:241454959 (Ack:251970741)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1680634018.931441][10926:10928] CHIP:EM: Found matching exchange: 64752i, Delegate: 0x7f478c039598
[1680634018.931461][10926:10928] CHIP:EM: Rxd Ack; Removing MessageCounter:251970741 from Retrans Table on exchange 64752i
[1680634018.985312][10926:10928] CHIP:EM: >>> [E:64752i M:241454960 (Ack:251970741)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:33 (SecureChannel:CASE_Sigma2Resume)
[1680634018.985325][10926:10928] CHIP:EM: Found matching exchange: 64752i, Delegate: 0x7f478c039598
[1680634018.985329][10926:10928] CHIP:EM: CHIP MessageCounter:251970741 not in RetransTable on exchange 64752i
[1680634018.985337][10926:10928] CHIP:SC: Received Sigma2Resume msg
[1680634018.985392][10926:10928] CHIP:SC: Peer assigned session session ID 44232
[1680634018.985836][10926:10928] CHIP:SC: Sending status report. Protocol code 0, exchange 64752
[1680634018.985862][10926:10928] CHIP:EM: <<< [E:64752i M:251970742 (Ack:241454960)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
[1680634018.985880][10926:10928] CHIP:IN: (U) Sending msg 251970742 to IP address 'UDP:[fd43:6727:9396:9685:7204:1dff:fea9:1114%enp0s3]:5540'
[1680634018.985947][10926:10928] CHIP:SC: SecureSession[0x7f478c038000]: Moving from state 'kEstablishing' --> 'kActive'
[1680634018.985955][10926:10928] CHIP:IN: SecureSession[0x7f478c038000]: Activated - Type:2 LSID:24658
[1680634018.985956][10926:10928] CHIP:IN: New secure session activated for device <0000000000007283, 1>, LSID:24658 PSID:44232!
[1680634018.985960][10926:10928] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 4 --> 5
[1680634018.985971][10926:10928] CHIP:TOO: Sending cluster (0x00000201) command (0x00000003) on endpoint 2
[1680634018.985981][10926:10928] CHIP:DMG: ICR moving to [AddingComm]
[1680634018.985984][10926:10928] CHIP:DMG: ICR moving to [AddedComma]
[1680634018.985999][10926:10928] CHIP:EM: <<< [E:64753i M:241053170] (S) Msg TX to 1:0000000000007283 [D27C] --- Type 0001:08 (IM:InvokeCommandRequest)
[1680634018.986004][10926:10928] CHIP:IN: (S) Sending msg 241053170 on secure session with LSID: 24658
[1680634018.986019][10926:10928] CHIP:DMG: ICR moving to [CommandSen]
[1680634019.126623][10926:10928] CHIP:EM: >>> [E:64752i M:241454961 (Ack:251970742)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1680634019.126642][10926:10928] CHIP:EM: Found matching exchange: 64752i, Delegate: (nil)
[1680634019.126655][10926:10928] CHIP:EM: Rxd Ack; Removing MessageCounter:251970742 from Retrans Table on exchange 64752i
[1680634019.209464][10926:10928] CHIP:EM: >>> [E:64753i M:162080939 (Ack:241053170)] (S) Msg RX from 1:0000000000007283 [D27C] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1680634019.209479][10926:10928] CHIP:EM: Found matching exchange: 64753i, Delegate: 0x7f4794050658
[1680634019.209486][10926:10928] CHIP:EM: Rxd Ack; Removing MessageCounter:241053170 from Retrans Table on exchange 64753i
[1680634025.216224][10926:10928] CHIP:IN: SecureSession[0x7f478c038000]: MarkAsDefunct Type:2 LSID:24658
[1680634025.216365][10926:10928] CHIP:SC: SecureSession[0x7f478c038000]: Moving from state 'kActive' --> 'kDefunct'
[1680634025.216427][10926:10928] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 64753i
[1680634025.216453][10926:10928] CHIP:TOO: Error: ../../src/app/CommandSender.cpp:248: CHIP Error 0x00000032: Timeout
[1680634025.216460][10926:10928] CHIP:DMG: ICR moving to [AwaitingDe]
[1680634025.216606][10926:10926] CHIP:TOO: Run command failure: ../../src/app/CommandSender.cpp:248: CHIP Error 0x00000032: Timeout
```
## Log from device console
```
I (1959870) chip[EM]: >>> [E:64758r M:251970747] (U) Msg RX from 0:6537F3AD26D87AE6 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1959870) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3fca2fc0
I (1959890) chip[IN]: CASE Server disabling CASE session setups
I (1959900) chip[EM]: <<< [E:64758r M:241454970 (Ack:251970747)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1959910) chip[IN]: (U) Sending msg 241454970 to IP address 'UDP:[FD43:6727:9396:9685:5586:5CD2:C834:C518%st1]:44809'
I (1959920) chip[SC]: Received Sigma1 msg
I (1959950) chip[EM]: <<< [E:64758r M:241454971 (Ack:251970747)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:33 (SecureChannel:CASE_Sigma2Resume)
I (1959960) chip[IN]: (U) Sending msg 241454971 to IP address 'UDP:[FD43:6727:9396:9685:5586:5CD2:C834:C518%st1]:44809'
I (1959980) chip[EM]: >>> [E:64758r M:251970748 (Ack:241454971)] (U) Msg RX from 0:6537F3AD26D87AE6 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (1960000) chip[SC]: Success status report received. Session was established
I (1960060) chip[SC]: SecureSession[0x3fca1a78]: Moving from state 'kEstablishing' --> 'kActive'
I (1960060) chip[IN]: CASE Session established to peer: <000000000001B669, 1>
I (1960070) chip[IN]: CASE Server enabling CASE session setups
I (1960070) chip[EM]: <<< [E:64758r M:241454972 (Ack:251970748)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[... cut ...]
I (1960110) chip[EM]: >>> [E:64759r M:143102504] (S) Msg RX from 1:000000000001B669 [D27C] --- Type 0001:08 (IM:InvokeCommandRequest)
I (1960120) esp_matter_command: Received command 0x0003 for enpoint 0x0002's cluster 0x00000201
E (1960160) chip[DMG]: Failed to send command response: 3
I (1960170) chip[EM]: <<< [E:64759r M:62697130 (Ack:143102504)] (S) Msg TX to 1:000000000001B669 [D27C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1960180) chip[IN]: (S) Sending msg 62697130 on secure session with LSID: 44235
```

VaishaliAvhale, 2023-04-06:
@markusp00 The commands SetWeeklySchedule, GetWeeklySchedule, and ClearWeeklySchedule are enabled for the schedule configuration (SCH) feature, which is an optional feature for the default Thermostat device type as per the Matter device library specification.
Also, the upstream SDK has no implementation of these commands yet: https://github.com/projectchip/connectedhomeip/blob/master/src/app/clusters/thermostat-server/thermostat-server.cpp#L356-#L378. As a result, there's a timeout error.

markusp00, 2023-04-10:
@VaishaliAvhale Thanks for the answer, I was not aware of this.

closed 2023-04-10 by markusp00