espressif/esp-matter #308: Devices are showing offline in google home

SYESHASWINI, 2023-04-05:
I have created matter project in the google console and the devices are adding successfully, but the devices are showing offline.

Google Home app version: 2.66.1.13
Nest Hub version(GUIK2 2nd gen): 1.56.324896

dhrishi, 2023-04-06:
@SYESHASWINI After commissioning, the device cannot be controlled ever? Or it is accessible later? If you are able to control it later on, it can be an issue on the Google side. Please upload your device side logs

jonsmirl, 2023-04-06:
Now that I can commission, I am seeing same thing
[fabric.log](https://github.com/espressif/esp-matter/files/11173461/fabric.log)

Something wrong with fabric ids.
~~~
ureChannel:StandaloneAck)
I (127010) chip[IN]: (U) Sending msg 160102715 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'
I (127010) chip[SC]: Received Sigma1 msg
E (127020) chip[SC]: CASE failed to match destination ID with local fabrics
I (127020) chip[EM]: <<< [E:65192r S:0 M:160102716 (Ack:23425378)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)I (127020) chip[IN]: (U) Sending msg 160102716 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'E (127030) chip[IN]: CASE Session establishment failed: 10
I (127030) chip[IN]: CASE Server enabling CASE session setups
E (127030) chip[EM]: OnMessageReceived failed, err = 10
I (127030) chip[EM]: >>> [E:65192r S:0 M:23425379 (Ack:160102716)] (U) Msg RX from 0:B2E1FC3469C833FD [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
~~~

jonsmirl, 2023-04-06:
The fix for this is to factory reset your Nest Hub.

jonsmirl, 2023-04-07:
I did notice that after I commission my device it shows as offline and no amount of poking on Google Home will bring it on-line. If you reboot the device it will come on-line.

dhrishi, 2023-04-12:
@SYESHASWINI @jonsmirl As of today's date, I am able to commission and control the devices using Google Home just fine (without reboot). Can you please confirm and close the issue if it works

SYESHASWINI, 2023-04-12:
@dhrishi devices are successfully adding and operating, Thank You.

closed 2023-04-12 by SYESHASWINI