espressif/esp-matter #309: Issue while enabling OTA for devices

SYESHASWINI, 2023-04-05:
After enabling OTA in menuconfig, getting errors while building the file. 
Attaching the error log.

[ota_error.odt](https://github.com/espressif/esp-matter/files/11158544/ota_error.odt)

jadhavrohit924, 2023-04-06:
@SYESHASWINI Could you provide us more details?

1. What example did you tried?
2. What steps you have followed?
3. On which branch you are trying out v1.0 or main?

shubhamdp, 2023-04-07:
@SYESHASWINI Building firmware image was successful. But building an OTA image which can be used is broken right now. Can you please disable `CONFIG_CHIP_OTA_IMAGE_BUILD` and rebuilt it?

If you want to build an ota image, please use the [ota_image_tool.py](https://github.com/project-chip/connectedhomeip/blob/master/src/app/ota_image_tool.py) to generate ota firmware image.

SYESHASWINI, 2023-04-10:
@tried with sample light program

SYESHASWINI, 2023-04-10:
@jadhavrohit924 tried with sample light program

SYESHASWINI, 2023-04-10:
@shubhamdp Can I get any resources to generate the ota image.

shubhamdp, 2023-04-17:
@SYESHASWINI here's the help of tool:  https://github.com/project-chip/connectedhomeip/blob/v1.0.0.2/src/app/ota_image_tool.py#L328

But a simple OTA image can be generated with following command:
- Upgrade the version in `examples/<app>/CMakeLists.txt` by changing below lines https://github.com/espressif/esp-matter/blob/b40bf8e398161bcac515fd57eb13d14e031e3a91/examples/light/CMakeLists.txt#L27-L28
- Build the project: `idf.py build`
- Build an OTA image with following command
    ```
    ➜ ./connectedhomeip/connectedhomeip/src/app/ota_image_tool.py create \
             --vendor-id 0xFFF1 --product-id 0x8001 \
             --version 2 --version-str "v2.0" \
             --digest-algorithm sha256 \
             examples/light/build/light.bin light-matter-ota.bin
    ```

Now, you can use `light-matter-ota.bin` for performing OTA.

SYESHASWINI, 2023-04-18:
@shubhamdp  I have taken two esp32c3's where one is OTA requestor enabled and other is loaded with OTA provider app.  Commissioned the device with the QR of OTA provider, but the query response is getting failed. 

[ota_provider.txt](https://github.com/espressif/esp-matter/files/11261059/ota_provider.txt)
[ota_requestor.txt](https://github.com/espressif/esp-matter/files/11261060/ota_requestor.txt)

jadhavrohit924, 2023-04-19:
@SYESHASWINI Have you followed the steps mentioned [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/ota.md) ?

Could you please confirm that `Software version` of the image being ota `is greater` than currently version?

Also I think you have attached wrong `ota_provider` logs.

jadhavrohit924, 2023-04-19:
Edit:
@SYESHASWINI Please attach `ota_requestor`, `ota_provider` and `chip-tool` logs along with all the commands you have executed to analyse the issue in better way.

SYESHASWINI, 2023-04-20:
@jadhavrohit924 how to find the provider node id. 

Ota provider examples is taken from [here](https://github.com/project-chip/connectedhomeip/tree/master/examples/ota-provider-app/esp32l).
I have commissioned the device through QR code from ota-provider example, in which the announce command is received.

SYESHASWINI, 2023-04-25:
@jadhavrohit924 I queried the matter ota through "matter ota query" in the console and the query response is failed.
attaching the error log.
[ota_query_requestory.txt](https://github.com/espressif/esp-matter/files/11322207/ota_query_requestory.txt)

jadhavrohit924, 2023-04-25:
@SYESHASWINI 

> I have commissioned the device through QR code from ota-provider example, in which the announce command is received.

If you are using home app, I would suggest to use chip-tool. So that you will know what is the `node-id` of the ota provider and ota requestor. 

> I queried the matter ota through "matter ota query" in the console and the query response is failed.
attaching the error log.

From the logs,  `Failed to connect to node 0x0000000056791ABE: 3000004` which means ota-requestor is not able to connect to ota-provider.

Here are the steps I want you to try out using `chip-tool`.

- Build the image you wanted to OTA. ( Make sure the software version of image should be greater than the software version of image on which you are trying OTA)
- Commission the ota-requestor. ( This software version should be less than the image software version which you build for OTA)
- Follow [this](https://github.com/project-chip/connectedhomeip/tree/master/examples/ota-provider-app/esp32) to copy OTA image to ota-provider and Commission the ota-provider. And `Write the ACL attribute`
- Run `announce-ota-provider` command mentioned [here](https://github.com/project-chip/connectedhomeip/tree/master/examples/ota-requestor-app/esp32#query-for-an-ota-image)

If you fail to perform OTA please attach all the logs i.e. `ota-requesto`, `ota-provider` and `chip-tool`.

SYESHASWINI, 2023-04-28:
@jadhavrohit924 I have performed Ota Announce command and the status seems success. Where can I find the Version, to confirm that OTA is successful.
Attaching the chip-tool log.
[ota-announce.txt](https://github.com/espressif/esp-matter/files/11354204/ota-announce.txt)

dhrishi, 2023-04-29:
I am closing this as the OTA status is seen as success

closed 2023-04-29 by dhrishi