espressif/esp-matter #310: Issues while building lighting application on esp32

Jayesh19101999, 2023-04-06:
Hello currently I am following this guide https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html
and i have clone both the repositories esp-idf and esp-matter  but while building i am facing these errors 
can someone help me out through this? --
 

einfochips@PUNECPU0517:~/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build
Executing "ninja all"...
[2/616] Generating ../../gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h, ../../gen/app-codegen/cpp-app/app/callback-stub.cpp
FAILED: gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h gen/app-codegen/cpp-app/app/callback-stub.cpp /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/callback-stub.cpp
cd /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/main && /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app --expected-outputs /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/expected.outputs /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/lighting-common/lighting-app.matter
Traceback (most recent call last):
  File "/home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from matter_idl.matter_idl_parser import CreateParser
ModuleNotFoundError: No module named 'matter_idl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/scripts/codegen.py", line 32, in <module>
    from matter_idl.matter_idl_parser import CreateParser
  File "/home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/scripts/py_matter_idl/matter_idl/matter_idl_parser.py", line 6, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
[4/616] Performing build step for 'bootloader'
[1/1] cd /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/bootloader/esp-idf/esptool_py && /home/einfochips/Desktop/matter/connectedhomeip/.environment/pigweed-venv/bin/python /home/einfochips/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/bootloader/bootloader.bin
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
[6/614] Generating ../../gen/app-zapgen/zapgen/app-templates/zap-generated/access.h, ../../ge...config.h, ../../gen/app-zapgen/zapgen/app-templates/zap-generated/IMClusterCommandHandler.cpp
🔧 Using temporary state directory: /tmp/zappzame0zu/zap.CaTdVX
🤖 ZAP generation started:
    🔍 input files: /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/lighting-common/lighting-app.zap
    🔍 output pattern: /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated
    🔍 using templates: /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/src/app/zap-templates/app-templates.json
    🔍 using zcl data: /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/src/app/zap-templates/zcl/zcl.json
    🔍 zap version: ver. 2023.3.27, featureLevel 94, commit: 15f96c4f997301da1174ef29156c1c7d442f5d44 from 2023-03-27T17:20:50.000Z, mode: binary, exe: zap-cli
🕐 Setup time: 80s 937ms
👉 using input file: /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/lighting-common/lighting-app.zap
👉 using output destination: /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated
🕐 File loading time: 1s 74ms
🤖 Generating files:
    ✍  /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h
    ✍  /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated/IMClusterCommandHandler.cpp
    ✍  /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated/CHIPClusters.h
    ✍  /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated/CHIPClientCallbacks.h
    ✍  /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated/endpoint_config.h
    ✍  /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated/access.h
🕐 Generation time: 2s 193ms
Searching for zcl file from /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/lighting-common/lighting-app.zap
Files generated in: /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates/zap-generated
[7/614] Performing build step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/chip/lib/libCHIP.a
cd /home/einfochips/Desktop/32ESP/esp-matter/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/chip && ninja esp32
[0/1] Regenerating ninja files
[1/759] stamp obj/mbedtls/mbedtls.stamp
[2/759] stamp obj/third_party/connectedhomeip/src/controller/data_model/data_model_codegen_codegen._metadata_path_list.txt.stamp
[3/759] stamp obj/third_party/connectedhomeip/src/controller/data_model/data_model_zapgen_files._metadata_path_list.txt.stamp
[4/759] stamp obj/third_party/connectedhomeip/src/controller/data_model/data_model_zapgen_zap_pregen._metadata_path_list.txt.stamp
[5/759] ACTION //third_party/connectedhomeip/src/crypto:gen_crypto_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[6/759] ACTION //third_party/connectedhomeip/src/inet:gen_inet_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[7/759] stamp obj/third_party/connectedhomeip/src/crypto/gen_crypto_buildconfig.stamp
[8/759] stamp obj/third_party/connectedhomeip/src/inet/gen_inet_buildconfig.stamp
[9/759] stamp obj/third_party/connectedhomeip/src/crypto/crypto_buildconfig.stamp
[10/759] stamp obj/third_party/connectedhomeip/src/inet/inet_buildconfig.stamp
[11/759] ACTION //third_party/connectedhomeip/src/lib/asn1:gen_asn1oid(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[12/759] ACTION //third_party/connectedhomeip/src/app:gen_app_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[13/759] ACTION //third_party/connectedhomeip/src/lib/core:gen_chip_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[14/759] stamp obj/third_party/connectedhomeip/src/lib/asn1/gen_asn1oid.stamp
[15/759] ACTION //third_party/connectedhomeip/src/ble:gen_ble_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[16/759] stamp obj/third_party/connectedhomeip/src/lib/core/gen_chip_buildconfig.stamp
[... 19 lines cut ...]
[... cut ...]
[39/759] ACTION //third_party/connectedhomeip/src/setup_payload:gen_additional_data_payload_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[40/759] ACTION //third_party/connectedhomeip/src/system:gen_system_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[41/759] c++ obj/third_party/connectedhomeip/src/trace/trace.trace.cpp.o
[42/759] ACTION //third_party/connectedhomeip/src/controller/data_model:data_model_zapgen_zap_pregen(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

dhrishi, 2023-04-06:
@Jayesh19101999 If you are using the `esp-matter` repository, I assume that you want to use the applications present there. In that case, you would have to use `esp-matter/examples/light` example instead of the `lighting-app` that is present in the connectedhomeip submodule.

shubhamdp, 2023-04-07:
If you are building an example in connectedhomeip repository, please follow guide from here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/setup_idf_chip.md

Jayesh19101999, 2023-04-07:
@dhrishi Hi ,Actually I wanted ON/OFF cluster which not present in light application.  esp-matter/examples/light This example creates a Color Temperature Light device using the ESP Matter data model. Hence thats the reason why i used lighting application

dhrishi, 2023-04-07:
@Jayesh19101999 OnOff cluster is there in the light example too, but along with LevelControl and ColorControl cluster. Do you want to build just an OnOff light? If yes, you can replace https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L139-#L147 this to create an on_off_light

Jayesh19101999, 2023-04-07:
@dhrishi As i have already flashed Light application and successfully commissioned through chip-tool app, i am trying to toggle using this command "_onoff on_ 0x7283 0x1_" without any error but not reflecting anything on ESP32. This light should reflect on board LED or should i have to connect the external LED to any GPIO?

dhrishi, 2023-04-07:
@Jayesh19101999 If you are seeing the prints on the console that the attribute value has changed then you should be good. In case of ESP32-C3, there's an onboard LED that we use and change. You can connect an external LED and use it.

As you have already compiled and flashed the application, I am closing this issue.

closed 2023-04-07 by dhrishi

Jayesh19101999, 2023-04-10:
Thank you
But my ESP32 is not C3 its DEVKIT V1 on which GPIO i should connect the LED