espressif/esp-matter #312: Issues while building light application on esp32c6

qimi-dev, 2023-04-06:
I configure the environment and build the Light application(Enable Thread) according to this document (https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html).

But i get the following error message in the console.

```
I (22) boot: ESP-IDF v5.1-dev-4124-gbb9200acec 2nd stage bootloader
I (23) boot: compile time Apr  6 2023 23:09:47
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 40MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
W (46) bootloader_random: bootloader_random_enable() has not been implemented yet
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (73) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (80) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (87) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (95) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (102) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (110) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (117) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (125) boot: End of partition table
I (130) esp_image: segment 0: paddr=00020020 vaddr=42120020 size=3f0a8h (258216) map
I (264) esp_image: segment 1: paddr=0005f0d0 vaddr=40800000 size=00f48h (  3912) load
I (268) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=11f648h (1177160) map
I (844) esp_image: segment 3: paddr=0017f670 vaddr=40800f48 size=0edf0h ( 60912) load
I (878) esp_image: segment 4: paddr=0018e468 vaddr=4080fd40 size=02308h (  8968) load
I (892) boot: Loaded app from partition at offset 0x20000
I (893) boot: Disabling RNG early entropy source...
W (893) bootloader_random: bootloader_random_enable() has not been implemented yet
I (913) cpu_start: Pro cpu up.
W (922) clk: esp_perip_clk_init() has not been implemented yet
I (929) cpu_start: Pro cpu start user code
I (929) cpu_start: cpu freq: 160000000 Hz
I (929) cpu_start: Application information:
I (932) cpu_start: Project name:     light
I (937) cpu_start: App version:      v1.0
I (941) cpu_start: Compile time:     Apr  6 2023 23:13:19
I (948) cpu_start: ELF file SHA256:  38c984d9e4ba48df...
I (954) cpu_start: ESP-IDF:          v5.1-dev-4124-gbb9200acec
I (960) cpu_start: Min chip rev:     v0.0
I (965) cpu_start: Max chip rev:     v0.99 
I (970) cpu_start: Chip rev:         v0.0
I (974) heap_init: Initializing. RAM available for dynamic allocation:
I (982) heap_init: At 40827770 len 00054EA0 (339 KiB): D/IRAM
I (988) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (995) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (1002) spi_flash: detected chip: generic
I (1006) spi_flash: flash io: dio
W (1010) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (1023) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (1035) sleep: Configure to isolate all GPIO pins in sleep state
I (1041) sleep: Enable automatic switching of GPIO sleep configuration
I (1048) coexist: coex firmware version: 5315623
I (1053) coexist: coexist rom version 5b8dcfa
I (1058) app_start: Starting scheduler on CPU0
I (1063) main_task: Started on CPU0
I (1063) main_task: Calling app_main()
I (1083) led_driver_ws2812: Initializing light driver
I (1083) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
W (1093) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
I (1093) app_main: Light created with endpoint_id 1
I (1113) chip[DL]: NVS set: chip-counters/reboot-count = 6 (0x6)
I (1113) chip[DL]: Real time clock set to 946684800 (0100/00/01 00:00:00 UTC)
I (1113) phy_init: phy_version 102,e0e553c,Feb 16 2023,16:20:06
W (1123) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
E (1123) OPENTHREAD: esp_openthread_platform_init(104): OpenThread storage partition not found
E (1143) OPENTHREAD: esp_openthread_init(23): Failed to initialize OpenThread platform driver
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42068164
0x42068164: openthread_init_stack at /Users/qimi/Application/Espressif/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/OpenthreadLauncher.c:69 (discriminator 1)

file: "../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/OpenthreadLauncher.c" line 76
func: openthread_init_stack
expression: esp_openthread_init(&config)

abort() was called at PC 0x4080ae93 on core 0
0x4080ae93: _esp_error_check_failed at /Users/qimi/Application/Espressif/esp-idf/components/esp_system/esp_err.c:50

Stack dump detected
Core  0 register dump:
MEPC    : 0x408005dc  RA      : 0x4080ae9e  SP      : 0x4082a5e0  GP      : 0x40810540  
0x408005dc: panic_abort at /Users/qimi/Application/Espressif/esp-idf/components/esp_system/panic.c:451

0x4080ae9e: __ubsan_include at /Users/qimi/Application/Espressif/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x407ec348  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4082a61c  S1      : 0x4082a600  A0      : 0x4082a61c  A1      : 0x4082a5fe  
A2      : 0x00000000  A3      : 0x4082a649  A4      : 0x00000001  A5      : 0x40826000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x40826000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000
```
My light application is not working properly, what should I do to fix it.

wqx6, 2023-04-07:
ESP32C6 is set a WiFi device as default. To enable Thread, you need do the following things.
* change the menuconfig
```
# Enable OpenThread
CONFIG_OPENTHREAD_ENABLED=y
# Disable Wi-Fi Station
CONFIG_ENABLE_WIFI_STATION=n
# Disable minimal mDNS
CONFIG_USE_MINIMAL_MDNS=n
```
* modify the partition table csv file partitions.csv, and add a partition named ot_storage like we did in partitions_h2.csv

qimi-dev, 2023-04-07:
> ESP32C6 is set a WiFi device as default. To enable Thread, you need do the following things.
>
> * change the menuconfig
>
> ```
> # Enable OpenThread
> CONFIG_OPENTHREAD_ENABLED=y
> # Disable Wi-Fi Station [...]

Thanks a lot, configuring the partitions.csv file solved my problem

closed 2023-04-07 by qimi-dev