espressif/esp-matter #313: Failed to post event to CHIP Platform event queue

jonsmirl, 2023-04-06:
I keep getting these errors, is there some way to make this queue longer?
~~~
E (92049) chip[DL]: Failed to post event to CHIP Platform event queue
E (92049) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
I (92049) esp_matter_attribute: ********** Endpoint 0x0003's Cluster 0x001E's Attribute 0xFFFD is 1 **********
E (92049) chip[DMG]: Error retrieving data from clusterId: 0x0000_001E, err = b
E (92059) chip[DL]: Failed to post event to CHIP Platform event queue
E (92059) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (92069) chip[DL]: Failed to post event to CHIP Platform event queue
E (92069) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
I (92069) chip[EM]: <<< [E:36347r S:43590 M:117479787 (Ack:121440306)] (S) Msg TX 
~~~

jonsmirl, 2023-04-06:
So does CONFIG_MAX_EVENT_QUEUE_SIZE end up in CHIP_DEVICE_CONFIG_MAX_EVENT_QUEUE_SIZE?

jonsmirl, 2023-04-06:
Setting CONFIG_MAX_EVENT_QUEUE_SIZE to 50 instead of 25 seems to have made it go away. I also noticed that my commissioning was significantly faster. I guess it was dropping items from that queue and then having to trigger slow network retries
[fabric.log](https://github.com/espressif/esp-matter/files/11174382/fabric.log)

shubhamdp, 2023-04-07:
> So does CONFIG_MAX_EVENT_QUEUE_SIZE end up in CHIP_DEVICE_CONFIG_MAX_EVENT_QUEUE_SIZE?

Yes, https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/CHIPDevicePlatformConfig.h#L39

Increasing it to 50 would have increased the static DRAM usage by (25 * sizeof([ChipDeviceEvent](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CHIPDeviceEvent.h#L373)))

jonsmirl, 2023-04-07:
I have 2MB of free PSRAM. Can we push it into PSRAM?

shubhamdp, 2023-04-17:
Right now I don't see any direct option for this. But, we can do a workaround.

Set `CONFIG_MAX_EVENT_QUEUE_SIZE` to higher value (say 50 as per your suggestion).

Set [`CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal) to some value (eg: 1K), please make sure to set it to less than `50 * sizeof(ChipDeviceEvent)` bytes.

By this method, all the allocations more than configured value will be from PSRAM.

stefano055415, 2023-05-02:
@jonsmirl @shubhamdp 

I too encounter the same problem

even on the light example

Has anyone solved it?

jonsmirl, 2023-05-02:
Set CONFIG_MAX_EVENT_QUEUE_SIZE higher. I ended up at 32 instead of 50. So it just needed a little more,

Something inside of chip probably should be blocking when the queue is full and then work to empty it before generating more entries.

stefano055415, 2023-05-02:
@jonsmirl 

Thanks, I tried and it actually works, I hope they fix it

jonsmirl, 2023-10-31:
More recent version of this issue: https://github.com/espressif/esp-matter/issues/656

closed 2023-10-31 by jonsmirl