espressif/esp-matter #316: Failed to commit pending.....

jonsmirl, 2023-04-07:
Something is broken with this. After I reboot the device I can't commit anything. I suspect this has to do with switching to encrypted partitions.  After reboot, maybe the device can't write to the encypted partitions any more?
~~~
I (86310) esp_matter_command: Received command 0x00000004 for endpoint 0x0000's cluster 0x00000030
I (86310) chip[FS]: GeneralCommissioning: Received CommissioningComplete
I (86450) chip[FP]: Metadata for Fabric 0x4 persisted to storage.
E (86600) chip[FP]: Failed to commit pending operational keypair 5001105
E (87160) chip[FP]: Failed to commit pending operational certificates 5001105I (87160) chip[TS]: Committing Last Known Good Time to storage: 2023-04-07T00:56:51
I (87460) chip[FP]: Fabric (0x4) deleted.
I (87460) chip[ZCL]: OpCreds: Fabric index 0x4 was removed
I (87460) chip[DIS]: Updating services using commissioning mode 0
I (87460) chip[DIS]: CHIP minimal mDNS started advertising.
I (87480) chip[DIS]: Advertise operational node 950545243CFE2D15-00000000F2DEF84C
I (87490) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 950545243CFE2D15-00000000F2DEF84C.
I (87500) chip[DIS]: mDNS service published: _matter._tcp
I (88060) chip[TS]: Pending Last Known Good Time: 2023-04-07T00:56:51
I (88060) chip[TS]: Previous Last Known Good Time: 2023-04-07T00:56:51
I (88070) chip[TS]: Reverted Last Known Good Time to previous value
E (88090) chip[FS]: GeneralCommissioning: Failed to commit pending fabric data: 5001105
I (88090) chip[EM]: <<< [E:3535r S:40338 M:207031141 (Ack:33103704)] (S) Msg TX to 4:B8DEDAADD0CAAAD1 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (88090) chip[IN]: (S) Sending msg 207031141 on secure session with LSID: 40338
~~~

jonsmirl, 2023-04-07:
All of this is without rebooting....
In the attached log I take a device with erased flash, flash it, and then commission it. That all works. 
Then at 114713, I share the device from Google Home to Google Sample App for Matter.
In that sharing process I get these errors:
~~~
I (127653) chip[FP]: Metadata for Fabric 0x2 persisted to storage.
E (128433) chip[FP]: Failed to commit pending operational certificates 5001105
I (128433) chip[TS]: Committing Last Known Good Time to storage: 2023-04-07T16:55:29
I (128693) chip[FP]: Fabric (0x2) deleted.
I (128693) chip[ZCL]: OpCreds: Fabric index 0x2 was removed
I (128693) chip[DIS]: Updating services using commissioning mode 0
I (128693) chip[DIS]: CHIP minimal mDNS started advertising.
I (128723) chip[DIS]: Advertise operational node 950545243CFE2D15-0000000047D19BA9
I (128723) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 950545243CFE2D15-0000000047D19BA9.
I (128733) chip[DIS]: mDNS service published: _matter._tcp
I (129113) chip[TS]: Pending Last Known Good Time: 2023-04-07T16:55:29
I (129113) chip[TS]: Previous Last Known Good Time: 2023-04-07T16:55:29
I (129113) chip[TS]: Reverted Last Known Good Time to previous value
E (129163) chip[FS]: GeneralCommissioning: Failed to commit pending fabric data: 5001105
I (129163) chip[EM]: <<< [E:3925r S:62934 M:65781139 (Ack:255691429)] (S) Msg TX to 2:B8DEDAADD0CAAAD1 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
~~~
Now go back and look at 50053, at that previous point in time the same code ran without error when committing fabric 1.
~~~
I (56043) esp_matter_command: Received command 0x00000004 for endpoint 0x0000's cluster 0x00000030
I (56043) chip[FS]: GeneralCommissioning: Received CommissioningComplete
I (56053) chip[FP]: Metadata for Fabric 0x1 persisted to storage.
I (56203) chip[TS]: Committing Last Known Good Time to storage: 2023-04-07T16:55:29
I (56203) chip[ZCL]: OpCreds: Fabric index 0x1 was committed to storage. Compressed Fabric Id 0x950545243CFE2D15, FabricId F6CEBBE0CBBC9E0F, NodeId 0000000047D19BA9, VendorId 0x6006
I (56283) chip[FS]: GeneralCommissioning: Successfully commited pending fabric data
I (56283) chip[FS]: Fail-safe cleanly disarmed
~~~

[commit.log](https://github.com/espressif/esp-matter/files/11180264/commit.log)

jonsmirl, 2023-04-07:
How long do these commits take? Could the previous one not be finished when then makes the next one fail?

jonsmirl, 2023-04-07:
5001105 maps to ESP_ERR_NVS_NOT_ENOUGH_SPACE

I am using the partition table from the light example. Can this really be too small?
~~~
esp_secure_cert,  0x3F, ,0xd000,    0x2000, encrypted
nvs,      data, nvs,     0x10000,   0x6000,
nvs_keys, data, nvs_keys,,          0x1000, encrypted
otadata,  data, ota,     ,          0x2000
phy_init, data, phy,     ,          0x1000,
ota_0,    app,  ota_0,   0x20000,   0x1E0000,
ota_1,    app,  ota_1,   0x200000,  0x1E0000,
fctry,    data, nvs,     0x3E0000,  0x6000
~~~
So let's make the partition larger.
~~~
nvs,      data, nvs,     0x10000,   0xB000,
~~~
And that works.  So, what is taking up more than 24,576 bytes in the nvs partition?

jonsmirl, 2023-04-07:
Next I added a print on read write. Why is is reading these keys over and over, maybe 100 times?
~~~
JDS Read Key g/su/0
JDS Read Key g/su/1
JDS Read Key g/su/2
JDS Read Key g/su/3
JDS Read Key g/su/4
JDS Read Key g/su/5
JDS Read Key g/su/6
JDS Read Key g/su/7
JDS Read Key g/su/8
JDS Read Key g/su/9
JDS Read Key g/su/a
JDS Read Key g/su/b
JDS Read Key g/su/c
JDS Read Key g/su/d
JDS Read Key g/su/e
~~~

Is there some tool which can extract this partition and decode it?

jonsmirl, 2023-04-07:
[flash.txt](https://github.com/espressif/esp-matter/files/11181990/flash.txt)
So there is way more in this partition than I expected...

Lot's of questions here, first one why are all these 8 and 18 bit value being persisted out in 24 bytes?
While are 300:4002, 300:4003, 300:4004 being persisted when I have that feature turned off?
Why is g/su/.... being read over and over and it isn't in the flash?
I think f/.. is fabrics. Why are there three entries when I have only made two fabrics?

and now I see why I ran out of space. lots of junk in here.

~~~
Namespace endpoint_0
  28:5            : BLOB  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  31:4            : BLOB  01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  33:1            : BLOB  0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  3E:3            : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Namespace endpoint_1
  6:0             : BLOB  01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  6:4003          : BLOB  8F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  8:4000          : BLOB  88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:7           : BLOB  0A 00 00 00 00 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:3           : BLOB  0A 00 00 00 00 00 00 00 6B 61 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4           : BLOB  0A 00 00 00 00 00 00 00 7D 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4002        : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4003        : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4004        : BLOB  0A 00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  8:0             : BLOB  88 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:8           : BLOB  0F 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4001        : BLOB  0F 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Namespace endpoint_2
  3B:1            : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Namespace endpoint_3
  6:0             : BLOB  01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  6:4003          : BLOB  8F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  8:4000          : BLOB  88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:7           : BLOB  0A 00 00 00 00 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:3           : BLOB  0A 00 00 00 00 00 00 00 6B 61 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4           : BLOB  0A 00 00 00 00 00 00 00 7D 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4002        : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4003        : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4004        : BLOB  0A 00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  8:0             : BLOB  88 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:8           : BLOB  0F 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4001        : BLOB  0F 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Namespace endpoint_4
  3B:1            : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Namespace endpoint_5
  6:0             : BLOB  01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  6:4003          : BLOB  8F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  8:4000          : BLOB  88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:7           : BLOB  0A 00 00 00 00 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:3           : BLOB  0A 00 00 00 00 00 00 00 6B 61 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4           : BLOB  0A 00 00 00 00 00 00 00 7D 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4002        : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4003        : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4004        : BLOB  0A 00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  8:0             : BLOB  88 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:8           : BLOB  0F 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:4001        : BLOB  0F 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Namespace endpoint_6
  3B:1            : BLOB  08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Namespace endpoint_7
  6:0             : BLOB  01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  6:4003          : BLOB  8F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  8:4000          : BLOB  88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  300:7           : BLOB  0A 00 00 00 00 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
  f/3/o           : BLOB  15 24 00 01 30 01 61 04 A8 4E F3 D7 13 8A E1 6F F8 4A A1 58 52 49 62 3D CA 1B 8D E8 50 99 72 FF ...
  f/3/n           : BLOB  15 30 01 01 01 24 02 01 37 03 24 14 00 18 26 04 80 12 54 28 26 05 80 15 20 3B 37 06 24 15 01 24 ...
  f/3/r           : BLOB  15 30 01 01 00 24 02 01 37 03 24 14 00 18 26 04 80 12 54 28 26 05 80 15 20 3B 37 06 24 14 00 18 ...
  g/fidx          : BLOB  15 24 00 04 36 01 04 01 04 02 04 03 18 18 
Namespace node
  min_uu_ep_id    : U16   18
Namespace bridge
  ep_id_array     : BLOB  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
~~~

jonsmirl, 2023-04-07:
When I look at the device again in Google Home it says it has three fabrics, but F2 is F1 duplicated.

jonsmirl, 2023-04-10:
Wasting space with the 20 byes is because this code in esp-matter core is insufficient:
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L1316

It needs to be a switch() with more cases to account for one and two byte attribute values. A similar switch is needed when the values are read.

jonsmirl, 2023-04-10:
Also, is it esp-matter which is not respecting the feature map and simply persisting all attribute values?

jonsmirl, 2023-04-11:
I believe your permissions are set to view this:
https://github.com/Lowpan-com/lowpan/blob/lovyan/main/ep.cpp#L332
I have 18 endpoints.  

Boris fixed the g/su/... bug.

  300:3 Mandatory
  300:4 Mandatory
  300:7 Mandatory
  300:8 Mandatory
  300:4001 Mandatory
  300:4002 Optional
  300:4003 Optional
  300:4004 Optional
 I had copied pasted coded which enabled 4002/4003/4004. I just removed that code, so that part was my bug.

When you get shrinking the blobs down working I can combine everything and recheck. Then I will hunt for who is duplicating that fabric.

jonsmirl, 2023-04-11:
This is the fix for the g/su/.. bug.
https://github.com/project-chip/connectedhomeip/pull/26039

shubhamdp, 2023-04-12:
@jonsmirl I have working patch for this, but thinking about handling upgrades with this as this will break the nvs data structure. Right now handling this using config option. Also, there's a way to handle it runtime, working on that.

jonsmirl, 2023-04-12:
Always write the values using the new format. When you read the values you need to allow for two different sizes on the read - old and new.

shubhamdp, 2023-04-20:
@jonsmirl we have pushed the changes in https://github.com/espressif/esp-matter/commit/05016db89b498811c3083e85f9381f129706d3c2, PTAL.

Keeping the issue open for a while.

dhrishi, 2023-04-27:
Closing this now

closed 2023-04-27 by dhrishi