espressif/esp-matter #317: run install.sh error

beyor98, 2023-04-08:
When I run install.sh, always report an error as follows:

Bootstrap: SSL error in Python when downloading CIPD client.
If using system Python try

    sudo pip install certifi

And if on the system Python on a Mac try

    /Applications/Python 3.6/Install Certificates.command

If using Homebrew Python try

    brew install openssl
    brew uninstall python
    brew install python

If those don't work, address all the potential issues shown 
by the following command.

    brew doctor

Otherwise, check that your machine's Python can use SSL, testing with the httplib module on Python 2 or http.client on Python 3.
Traceback (most recent call last):

ssl.SSLZeroReturnError: TLS/SSL connection has been closed (EOF) (_ssl.c:997)

Following the prompts to modify is not effective

shubhamdp, 2023-04-10:
@beyor98 One of us had faced the similar issue and had performed the steps prompted on the terminal.

Below is the steps that were followed:
```
# From the command line
$ python -m certifi

# Need link cacert.pem as cert.pem
$ ln -s /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/certifi/cacert.pem cert.pem
$ pwd
/Library/Frameworks/Python.framework/Versions/3.7/etc/openssl
```

Can you please try the above, please make sure to use the appropriate paths and python version directories on your system.

beyor98, 2023-04-11:
> @beyor98我们中的一个人遇到过类似的问题，并执行了终端上提示的步骤。
> 
> 以下是所遵循的步骤：
> 
> ```
> # From the command line
> $ python -m certifi
> 
> # Need link cacert.pem as cert.pem
> $ ln -s /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/certifi/cacert.pem cert.pem
> $ pwd
> /Library/Frameworks/Python.framework/Versions/3.7/etc/openssl
> ```
> 
> 你能试试上面的方法吗，请确保在你的系统上使用适当的路径和 python 版本目录。

--------------------------------------------------------
(myenv) hs@hs-vir-mach:~/esp-matter$ python -m certifi
/home/hs/esp-matter/myenv/lib/python3.8/site-packages/certifi/cacert.pem
(myenv) hs@hs-vir-mach:~/esp-matter$ cd /home/hs/esp-matter/myenv/lib/python3.8/site-packages/OpenSSL
(myenv) hs@hs-vir-mach:~/esp-matter/myenv/lib/python3.8/site-packages/OpenSSL$ ln -s /home/hs/esp-matter/myenv/lib/python3.8/site-packages/certifi/cacert.pem cert.pem

I'm not sure if the operation is correct, but it doesn't work and still reports the same error.
Could you please help me take a look again

dhrishi, 2023-04-14:
@beyor98 Please close the issue if this helps

beyor98, 2023-04-25:
Finally, the network source was replaced and the problem was solved.

closed 2023-04-25 by beyor98