espressif/esp-matter #318: switch subscription not working

Luca8991, 2023-04-08:
I've flashed the `light_switch` example on the **ESP32-C3** board.
I've successfully commissioned the device using the `chip-tool` command line (built from the official [connectedhomeip](https://github.com/project-chip/connectedhomeip) repo) from a Raspberry Pi, assigning it the `0x2` node id.

### Error
However, when I try to subscribe to the switch's `current-position` attribute using the command:
```shell
chip-tool switch subscribe current-position 0 2 0x2 0x0
```
between all the logs I get this relevant **error** message:
```
[1680971730.486122][27531:27533] CHIP:DMG: StatusResponseMessage =
[1680971730.486199][27531:27533] CHIP:DMG: {
[1680971730.486264][27531:27533] CHIP:DMG: 	Status = 0x80 (INVALID_ACTION),
[1680971730.486337][27531:27533] CHIP:DMG: 	InteractionModelRevision = 1
[1680971730.486404][27531:27533] CHIP:DMG: }
[1680971730.486468][27531:27533] CHIP:IM: Received status response, status is 0x80 (INVALID_ACTION)
```

Also, looking at the device shell I see this weird message:
```
E (153940) chip[IM]: Subscription from [1:000000000001B669] has no access at all. Rejecting result.
```

### Solutions I tried
I tried setting the access for the commissioner to the switch:
```shell
chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null }]' 0x2 0
```
Even if this command returns a successful response, I keep getting the access **error** when trying to subscribe.

Luca8991, 2023-04-08:
Also, when I try to read the current-position attribute value **without** a subscription:
```
chip-tool switch read current-position 0x2 0x0
```
I get the following message back:
```
[1680975381.814994][32273:32275] CHIP:DMG: ReportDataMessage =
[1680975381.815065][32273:32275] CHIP:DMG: {
[1680975381.815124][32273:32275] CHIP:DMG: 	AttributeReportIBs =
[1680975381.815208][32273:32275] CHIP:DMG: 	[
[1680975381.815274][32273:32275] CHIP:DMG: 		AttributeReportIB =
[1680975381.815360][32273:32275] CHIP:DMG: 		{
[1680975381.815430][32273:32275] CHIP:DMG: 			AttributeStatusIB =
[1680975381.815570][32273:32275] CHIP:DMG: 			{
[1680975381.815645][32273:32275] CHIP:DMG: 				AttributePathIB =
[1680975381.815731][32273:32275] CHIP:DMG: 				{
[1680975381.815817][32273:32275] CHIP:DMG: 					Endpoint = 0x0,
[1680975381.815907][32273:32275] CHIP:DMG: 					Cluster = 0x3b,
[1680975381.815998][32273:32275] CHIP:DMG: 					Attribute = 0x0000_0001,
[1680975381.816085][32273:32275] CHIP:DMG: 				}
[1680975381.816177][32273:32275] CHIP:DMG: 					
[1680975381.816257][32273:32275] CHIP:DMG: 				StatusIB =
[1680975381.816341][32273:32275] CHIP:DMG: 				{
[1680975381.816423][32273:32275] CHIP:DMG: 					status = 0xc3 (UNSUPPORTED_CLUSTER),
[1680975381.816510][32273:32275] CHIP:DMG: 				},
[1680975381.816592][32273:32275] CHIP:DMG: 				
[1680975381.816666][32273:32275] CHIP:DMG: 			},
[1680975381.816749][32273:32275] CHIP:DMG: 			
[1680975381.816816][32273:32275] CHIP:DMG: 		},
[1680975381.816895][32273:32275] CHIP:DMG: 		
[1680975381.816958][32273:32275] CHIP:DMG: 	],
[1680975381.817035][32273:32275] CHIP:DMG: 	
[1680975381.817098][32273:32275] CHIP:DMG: 	SuppressResponse = true, 
[1680975381.817163][32273:32275] CHIP:DMG: 	InteractionModelRevision = 1
[1680975381.817231][32273:32275] CHIP:DMG: }
[1680975381.817626][32273:32275] CHIP:TOO: Response Failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
[1680975381.817894][32273:32273] CHIP:TOO: Run command failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
```
So, it seems the `light_switch` example is not configured properly.

shubhamdp, 2023-04-10:
@Luca8991  The light_switch example does not feature an on-off server cluster, instead it includes an on-off client cluster. Therefore, it does not have any attributes from on-off cluster.

If you have a use case which requires an on/off server cluster, please refer to the generic_switch example.

closed 2023-04-10 by dhrishi

Luca8991, 2023-04-10:
@shubhamdp I also tried the `generic_switch` example without success... I keep getting the same `UNSUPPORTED_CLUSTER` error as in the `light_switch` one.
Btw, I also tried flashing the [all_clusters_app](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32) from the official **connectedhomeip** repo, but I got the same error.
Maybe I'm missing something?

shubhamdp, 2023-04-11:
@Luca8991 can you please paste the chip-tool commands that you are trying?

Luca8991, 2023-04-13:
@shubhamdp sure, this is the command I'm sending to the ESP32 with the `generic_switch` firmware:
```shell
chip-tool switch read current-position 0x2 0x0
```
and this is the log I get from `chip-tool`:
```shell
[1681379564.566729][4172:4174] CHIP:DMG: ReportDataMessage =
[1681379564.566785][4172:4174] CHIP:DMG: {
[1681379564.566829][4172:4174] CHIP:DMG: 	AttributeReportIBs =
[1681379564.566893][4172:4174] CHIP:DMG: 	[
[1681379564.566942][4172:4174] CHIP:DMG: 		AttributeReportIB =
[1681379564.567008][4172:4174] CHIP:DMG: 		{
[1681379564.567061][4172:4174] CHIP:DMG: 			AttributeStatusIB =
[1681379564.567128][4172:4174] CHIP:DMG: 			{
[1681379564.567184][4172:4174] CHIP:DMG: 				AttributePathIB =
[1681379564.567248][4172:4174] CHIP:DMG: 				{
[1681379564.567314][4172:4174] CHIP:DMG: 					Endpoint = 0x0,
[1681379564.567382][4172:4174] CHIP:DMG: 					Cluster = 0x3b,
[1681379564.567452][4172:4174] CHIP:DMG: 					Attribute = 0x0000_0001,
[1681379564.567517][4172:4174] CHIP:DMG: 				}
[1681379564.567590][4172:4174] CHIP:DMG: 					
[1681379564.567652][4172:4174] CHIP:DMG: 				StatusIB =
[1681379564.567717][4172:4174] CHIP:DMG: 				{
[1681379564.567782][4172:4174] CHIP:DMG: 					status = 0xc3 (UNSUPPORTED_CLUSTER),
[1681379564.567847][4172:4174] CHIP:DMG: 				},
[1681379564.567912][4172:4174] CHIP:DMG: 				
[1681379564.567970][4172:4174] CHIP:DMG: 			},
[1681379564.568035][4172:4174] CHIP:DMG: 			
[1681379564.568087][4172:4174] CHIP:DMG: 		},
[1681379564.568148][4172:4174] CHIP:DMG: 		
[1681379564.568196][4172:4174] CHIP:DMG: 	],
[1681379564.568256][4172:4174] CHIP:DMG: 	
[1681379564.568304][4172:4174] CHIP:DMG: 	SuppressResponse = true, 
[1681379564.568356][4172:4174] CHIP:DMG: 	InteractionModelRevision = 1
[1681379564.568403][4172:4174] CHIP:DMG: }
[1681379564.568610][4172:4174] CHIP:TOO: Response Failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
```