espressif/esp-matter #319: Issue while  Commissioning the device using chip-tool

Jayesh19101999, 2023-04-10:
I am trying to control the device using chip-tool even the device is been connected and  Commissioning is been completed still nothing gets reflected on ESP32

I (102700) chip[SVR]: Commissioning completed successfully
I (102710) chip[DIS]: Updating services using commissioning mode 0
I (102710) chip[DIS]: CHIP minimal mDNS started advertising.
I (102740) chip[DIS]: Advertise operational node 1759F6ABA397E88D-0000000000000462
I (102740) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 1759F6ABA397E88D-0000000000000462.
I (102760) chip[DIS]: mDNS service published: _matter._tcp
I (102760) chip[SC]: SecureSession[0x3ffc8870]: Moving from state 'kActive' --> 'kPendingEviction'
I (102770) chip[BLE]: Releasing end point's BLE connection back to application.
I (102780) app_main: Commissioning complete
I (102780) esp_matter_core: Commissioning Complete
I (102780) app_main: Commissioning window closed
I (102810) chip[EM]: >>> [E:23681r S:10691 M:5756975 (Ack:238363229)] (S) Msg RX from 1:000000000001B669 [E88D] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (102850) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 0
I (102850) chip[DL]: CHIPoBLE unsubscribe received
E (102850) chip[BLE]: no endpoint for unsub recvd
I (105140) chip[DL]: BLE GAP connection terminated (con 0 reason 0x213)
I (105140) NimBLE: GAP procedure initiated: stop advertising.

I (105160) esp_matter_core: BLE deinit successful and memory reclaimed

> matter esp wifi connect D-Link_DIR-600M comcast123
I (430700) wifi_init: rx ba win: 6
I (430700) wifi_init: tcpip mbox: 32
I (430700) wifi_init: udp mbox: 6
I (430700) wifi_init: tcp mbox: 6
I (430710) wifi_init: tcp tx win: 5744
I (430710) wifi_init: tcp rx win: 5744
I (430720) wifi_init: tcp mss: 1440
I (430730) wifi_init: WiFi IRAM OP enabled
I (430730) wifi_init: WiFi RX IRAM OP enabled
I (430730) wifi:state: run -> init (0)
I (430740) wifi:pm stop, total sleep time: 301798779 us / 332758274 us

W (430740) wifi:<ba-del>idx
I (430740) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
W (430760) wifi:hmac tx: ifx0 stop, discard
I (430830) wifi:flush txq
I (430830) wifi:stop sw txq
I (430830) wifi:lmac stop hw txq
I (430840) wifi:mode : sta (30:c6:f7:29:37:f0)
I (430840) wifi:enable tsf
Done
E (430850) chip[DL]: Long dispatch time: 148 ms, for event type 2
I (430850) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (430850) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (430860) wifi:state: init -> auth (b0)
I (430870) chip[DL]: Done driving station state, nothing else to do...
I (430870) chip[DL]: WIFI_EVENT_STA_STOP
I (430870) wifi:state: auth -> assoc (0)
I (430880) chip[DL]: Done driving station state, nothing else to do...
I (430880) chip[DL]: WIFI_EVENT_STA_START
I (430890) chip[DL]: Done driving station state, nothing else to do...
I (430900) wifi:state: assoc -> run (10)
                                                                                                                                                                                            I (430980) wifi:connected with D-Link_DIR-600M, aid = 4, channel 1, 40U, bssid = 0c:b6:d2:1c:66:74
I (430980) wifi:security: WPA2-PSK, phy: bgn, rssi: -51
I (430990) wifi:pm start, type: 1

I (430990) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (431010) chip[DL]: WIFI_EVENT_STA_CONNECTED
I (431010) chip[DL]: Done driving station state, nothing else to do...
W (431030) wifi:<ba-add>idx:0 (ifx:0, 0c:b6:d2:1c:66:74), tid:0, ssn:2, winSize:64
I (432380) chip[DL]: IP_EVENT_STA_GOT_IP 192.168.0.5, mask: 255.255.255.0, gw: 192.168.0.1
I (432380) chip[DL]: IPv4 address ready on WiFi station interface: 192.168.0.5/255.255.255.0 gateway 192.168.0.1
I (432390) app_main: Interface IP Address changed
I (432410) chip[DIS]: Updating services using commissioning mode 0
I (432410) chip[DIS]: CHIP minimal mDNS started advertising.
I (432420) chip[DIS]: Advertise operational node 1759F6ABA397E88D-0000000000000462
I (432420) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 1759F6ABA397E88D-0000000000000462.
I (432440) chip[DIS]: mDNS service published: _matter._tcp

dhrishi, 2023-04-10:
@Jayesh19101999 

> matter esp wifi connect D-Link_DIR-600M comcast123

Why is this done after the commissioning is complete? Commissioning also helps the device connect to the Wi-Fi network, so this shouldn't be required.

Jayesh19101999, 2023-04-11:
@dhrishi yes Understood..Its ok I am able to control it through the app thanks :)

closed 2023-04-11 by dhrishi