espressif/esp-matter #322: WindowCovering showing 'no response' in Apple Home App

jzwyssig, 2023-04-11:
This still seems to be an issue:
I added a bridged window covering with the latest esp-matter changes and idf 4.4.4 according to your code @stefano055415. It works perfectly with Google Home, but still not with Apple's Home app (HomePod & iPad are up to date with version 16.4). Strangely, the iPad shows the correct percentage of the roller shade and it is even possible to adjust the level of the shade and it is correctly synced with Google. Therefore, I guess it is still an Apple Home problem. Does it work for you?

_Originally posted by @jzwyssig in https://github.com/espressif/esp-matter/issues/168#issuecomment-1488878622_

jadhavrohit924, 2023-04-26:
Hi @jzwyssig , I was able to control window covering with homekit apple. 
With Home pod version 16.4.1 and  iPadOS 16.4.1.
[image]

Could you please try again by updating versions if not updated?

svenehlers, 2023-05-04:
This might be related to #343

dhrishi, 2023-05-06:
That's right. It should help. @jzwyssig Closing this. Please re-open if you face issues after following #343

closed 2023-05-06 by dhrishi