espressif/esp-matter #329: Can you update some examples？

Ko-Hon, 2023-04-12:
- may update some example like :

sensors: temperature / humanity / light

conrtrol :Thermostat/ fan 

thanks。

dhrishi, 2023-04-14:
@Ko-Hon The esp-matter data model APIs should be straightforward to add any device types. Is there a specific thing you are looking into or facing issues?

Ko-Hon, 2023-04-16:
yes but i don't know how, So I want to develop, or learn, based on a few examples. This makes it faster to get started

see：https://github.com/project-chip/connectedhomeip/tree/master/examples

lboue, 2023-04-16:
> * may update some example like :
> 
> sensors: temperature / humanity / light

You mean humidity?

dhrishi, 2023-05-07:
@Ko-Hon The examples in this repo should be simpler to add as you can dynamically create the data model using the APIs. Please see this as a reference https://github.com/espressif/esp-matter/issues/361#issuecomment-1530915395