espressif/esp-matter #332: What does esp-endpoint Extended Color Light look like in apple home?

moculll, 2023-04-13:
HELP!!
I want to use attribute "color loop",but When I add esp_matter lib 's extended color light in to my code , my apple home shows a light with level control and temperature control and also hue control,but I can't find where to change the color loop,help!!!!

jadhavrohit924, 2023-04-13:
@moculll `color loop` is feature which is not enabled by default in extended color light. You will have to explicitly add it to your example.

jadhavrohit924, 2023-04-13:
This is how you can add.

```
/* Add additional features to the node */
    cluster_t *device_cluster = cluster::get(device_endpoint, ColorControl::Id);
    cluster::color_control::feature::color_loop::config_t cl;
    cluster::color_control::feature::color_loop::add(device_cluster, &cl);
```

moculll, 2023-04-13:
@jadhavrohit924 So glad to receive your apply!I'm sorry that I didn't mention that I've add my code to esp-matter-lib 
`if (!endpoint) {
        ESP_LOGE(TAG, "Endpoint cannot be NULL");
        return NULL;
    }
    add_device_type(endpoint, get_device_type_id(), get_device_type_version());

    descriptor::create(endpoint, CLUSTER_FLAG_SERVER);
    cluster_t *identify_cluster = identify::create(endpoint, &(config->identify), CLUSTER_FLAG_SERVER);
    identify::command::create_trigger_effect(identify_cluster);
    groups::create(endpoint, &(config->groups), CLUSTER_FLAG_SERVER);
    scenes::create(endpoint, &(config->scenes), CLUSTER_FLAG_SERVER);
    on_off::create(endpoint, &(config->on_off), CLUSTER_FLAG_SERVER, on_off::feature::lighting::get_id());
    level_control::create(endpoint, &(config->level_control), CLUSTER_FLAG_SERVER,
                          level_control::feature::on_off::get_id() | level_control::feature::lighting::get_id());
    //color_control::create(endpoint, &(config->color_control), CLUSTER_FLAG_SERVER,
                          //color_control::feature::color_temperature::get_id() | color_control::feature::xy::get_id());
    color_control::create(endpoint, &(config->color_control), CLUSTER_FLAG_SERVER,
                          0x1f);

    return endpoint;`
` if (features & feature::color_loop::get_id()) {
        feature::color_loop::add(cluster, &(config->color_loop));
    }`
please look at " color_control::create" line, the 0x1f is bin:11111,which means opened the hs / temp / color loop , and in the "if" line it does add the color loop,but I just can't see the change in apple home (T T)

moculll, 2023-04-13:
@jadhavrohit924 thanks for your code!I will try it to see if it works T T

moculll, 2023-04-13:
@jadhavrohit924 I just tried to add the color_loop feature to the color_control cluster, but everything worked as the same, I don't know what to do now, do you know how color_loop works like in the normal situation?

VaishaliAvhale, 2023-04-14:
The difference between Default esp-matter light and light with color_loop feature is in Default esp-matter light you can select a color with different temperatures by clicking on the option given below the button and you increase its brightness through the button and about light with color_loop feature you can double tap on any one color option, you can see a color loop screen and through that color loop you can change the temperature of light and other thing is same as default light.
[image]
[image]

dhrishi, 2023-04-29:
@moculll It was an error on our side. We've added the fix https://github.com/espressif/esp-matter/commit/0870fbae89d5bc384bf8c85d08af531f1fc9ee34. Please try it out and let us know if it helps

moculll, 2023-05-04:
> @moculll It was an error on our side. We've added the fix [0870fba](https://github.com/espressif/esp-matter/commit/0870fbae89d5bc384bf8c85d08af531f1fc9ee34). Please try it out and let us know if it helps

Thank you！I will try it later and let you know if it works!!

moculll, 2023-05-04:
> @moculll It was an error on our side. We've added the fix [0870fba](https://github.com/espressif/esp-matter/commit/0870fbae89d5bc384bf8c85d08af531f1fc9ee34). Please try it out and let us know if it helps

Not working as I expected XD, from "/* Attributes not managed internally */" I know that the color loop function shall be coded by myself, but I just wonder that when I open the colorloop feature, will apple home show a botton that I can open/close colorloop ?

jadhavrohit924, 2023-05-05:
@moculll We have tested multiple combinations of features with apple but we are not seeing any type of button or any UI when we add `color_loop` feature.

dhrishi, 2023-05-06:
@moculll If you feel that the required attributes are missing after the color_loop feature has been added please let us know. Otherwise, I would suggest to report this in connectedhomeip for Apple folks to take a look.

moculll, 2023-05-09:
> @dhrishi 

> @moculll We have tested multiple combinations of features with apple but we are not seeing any type of button or any UI when we add `color_loop` feature.

Thank you very much, I've tested it too, and the weird thing is that I used the 'setcolorLoop' command in esp-matter there seem nothing to be changed, but when I used the chip-tool to set colorloop active attribute, it changed so rapidly

moculll, 2023-05-09:
> 
@dhrishi @jadhavrohit924 
I just wonder that if I can't use ui to control colorloop or other function, what's the meaning of these func exist, normal user can't use them by simple click, and developers have to develop an app to support users

moculll, 2023-06-09:
Thankful to all of u, the question have been resolved, as far as I can see, apple home app don't have color loop except color temp/brightness/onoff/hs control function ui, and in the matter server in which created color_control cluster can not set colorloop directly (maybe I haven't found the correct usage...?) ,and when I used chip-tool to set like blow:
`sudo chip-tool colorcontrol color-loop-set 7 1 0 3 0 1 1 destination-id endpoint-id`, the light  ran colorloop successfully, so this is my conclusion, thanks again

closed 2023-06-09 by moculll