espressif/esp-matter #334: why did I set the colorloopactivate to 1 and the color no change

moculll, 2023-04-14:
`if (cluster_id == OnOff::Id) {
            if (attribute_id == OnOff::Attributes::OnOff::Id) {
                //err = app_driver_light_set_power(handle, val);
                node_t *node = node::get();
                endpoint_t *endpoint = endpoint::get(node, light_endpoint_id);

                uint32_t cluster_id = ColorControl::Id;
                cluster_t *cluster = cluster::get(endpoint, cluster_id);
                chip::app::Clusters::ColorControl::Attributes::ColorLoopActive::Set(light_endpoint_id, 1);
                chip::app::Clusters::ColorControl::Attributes::ColorLoopDirection::Set(light_endpoint_id, 0);
                chip::app::Clusters::ColorControl::Attributes::ColorLoopTime::Set(light_endpoint_id, 5);
                chip::app::Clusters::ColorControl::Attributes::ColorLoopStartEnhancedHue::Set(light_endpoint_id, 0xfff0);`
this is my code, which means when I open/close the light, it will set to color_loop mode, but the reality is there is no changing

jadhavrohit924, 2023-04-17:
Hi @moculll, could you use esp-matter's sdk to add these attributes. Since these attributes are comes under `color_loop` feature, you can just add `color_loop` feature. And test again? This is how you can add.

```
/* Add additional features to the node */
    cluster_t *device_cluster = cluster::get(device_endpoint, ColorControl::Id);
    cluster::color_control::feature::color_loop::config_t cl;
    cluster::color_control::feature::color_loop::add(device_cluster, &cl);
```

jadhavrohit924, 2023-04-17:
@moculll Please note. 
```
Support for Enhanced Hue SHALL require support for Hue/Saturation.

Support for Color Loop SHALL require support for Enhanced Hue.
```

dhrishi, 2023-04-29:
Fixed with https://github.com/espressif/esp-matter/commit/0870fbae89d5bc384bf8c85d08af531f1fc9ee34

closed 2023-04-29 by dhrishi