espressif/esp-matter #336: Fails the matter test case

maenkai, 2023-04-19:
esp-idf: v4.4.2
esp-matter: commit-id fde2c7ea84fee46d4dfa8a874d63a054229ceee6

The following tests cannot be passed using the Matter Test Harness(V2.6)
```shell
TC-O-2.3
12b: Reads OffWaitTime attribute from DUT
21b: Reads OffWaitTime attribute from DUT

maenkai, 2023-04-19:
FAIL:
[image]

TH-V2.6 LOG
[UI_Test_Run_2023_4_19_17_36_28.txt](https://github.com/espressif/esp-matter/files/11271535/UI_Test_Run_2023_4_19_17_36_28.txt)

wqx6, 2023-04-19:
Can you please disable the log and test again? The log output might affect the attribute value.

maenkai, 2023-04-20:
> Can you please disable the log and test again? The log output might affect the attribute value.

I have modified the configuration item, but still failed
CONFIG_LOG_DEFAULT_LEVEL_NONE=y

stefano055415, 2023-05-09:
@maenkai @wqx6 

I also have the same problem but I use esp-idf: v5.0.1 and esp-matter_ commit 1b95fcf66793c2ead7df933d8d80eefd7f8f6b95 

did you find a solution?

Thank you

maenkai, 2023-05-10:
> @maenkai @wqx6
> 
> I also have the same problem but I use esp-idf: v5.0.1 and esp-matter_ commit [1b95fcf](https://github.com/espressif/esp-matter/commit/1b95fcf66793c2ead7df933d8d80eefd7f8f6b95)
> 
> did you find a solution?
> 
> Thank you

Using menuconfig to turn off the log reduces the chance of failure, but this is not a complete solution

stefano055415, 2023-05-10:
@maenkai 

I tried disabling logging, but out of ten tests only one passed without errors, error reduction is minimal

hope it gets fixed soon

Thank you

stefano055415, 2023-05-10:
@maenkai 

I have noticed that the problem occurs systematically on tests 14c 20b 23c and 24c

always: reads OnWaitTime attribute from DUT

maybe it can help

maenkai, 2023-05-16:
https://github.com/espressif/esp-matter/issues/131
It seems to be similar to the problem. When will it be solved

maenkai, 2023-06-07:
@wqx6 I am going to do the test certification, please help solve the problem

VaishaliAvhale, 2023-06-19:
[On-Off Cluster Test Plan.zip](https://github.com/espressif/esp-matter/files/11786999/On-Off.Cluster.Test.Plan.zip)
please update your PICS file and retest the test case.

dhrishi, 2023-06-30:
@maenkai Can you please confirm if the above PICS helped you?

maenkai, 2023-07-05:
> @maenkai Can you please confirm if the above PICS helped you?

Still failing, what does the failure of the project have to do with it, maybe I can optimize from the application layer

VaishaliAvhale, 2023-07-06:
I have tested above test case using  the same PICS file which I shared with you and It is passed
About the steps which are failing in your test case are not applicable . Check it in attached Test Report!
If your facing diffrent issue please share it's logs and Test Report
also which TH version you are using? Please use updated TH version (Version: v2.9-beta1)
[Test Report.pdf](https://github.com/espressif/esp-matter/files/11965388/Test.Report.pdf)

maenkai, 2023-07-06:
> I have tested the above test case using the same PICS file which I shared with you and It is passed The steps which are failing in your test case are not applicable. Check it in the attached Test Report! If you're facing a different issue please share its logs and Test Report also which TH version you are using? Please use the updated TH version (Version: v2.9-beta1) [Test Report.pdf](https://github.com/espressif/esp-matter/files/11965419/Test.Report.pdf)

My version is 2.6, only 2.7 can be found on the official website, please tell me how to get the next version 2.9, thank you!

VaishaliAvhale, 2023-07-07:
Follow the below TH upgrade instructions:

cd chip-certification-tool
git pull
git checkout release/2.9-TE1
./scripts/update.sh
./scripts/ubuntu/update-sample-apps.sh
./scripts/ubuntu/update-paa-certs.sh
./scripts/stop.sh
./scripts/build-no-cache.sh
./scripts/start.sh

closed 2023-07-10 by maenkai