espressif/esp-matter #337: ESP32 unable to Add Accessory to Apple Home

Tim4497, 2023-04-19:
I am not able to add my device to Apple Home, the WiFi connection seems to get established.

Am I missing something? I tried on 2 ESP32 with two different router/networks and several of the examples.

Heres my full output:
```I (27) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (27) boot: compile time 17:17:55
I (27) boot: chip revision: v3.0
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (75) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (82) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (90) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (97) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (105) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (112) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (120) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (127) boot: End of partition table
I (132) boot: No factory image, trying OTA 0
I (137) boot_comm: chip revision: 3, min. application chip revision: 0
I (144) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=2fd44h (195908) map
I (223) esp_image: segment 1: paddr=0004fd6c vaddr=3ffbdb60 size=002ach (   684) load
I (224) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=f0c78h (986232) map
I (586) esp_image: segment 3: paddr=00140ca0 vaddr=3ffbde0c size=0551ch ( 21788) load
I (595) esp_image: segment 4: paddr=001461c4 vaddr=40080000 size=1d52ch (120108) load
I (659) boot: Loaded app from partition at offset 0x20000
I (703) boot: Set actual ota_seq=1 in otadata[0]
I (703) boot: Disabling RNG early entropy source...
I (714) cpu_start: Pro cpu up.
I (715) cpu_start: Starting app cpu, entry point is 0x4008141c
0x4008141c: call_start_cpu1 at /Users/t.winkler/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (731) cpu_start: Pro cpu start user code
I (731) cpu_start: cpu freq: 160000000
I (731) cpu_start: Application information:
I (735) cpu_start: Project name:     light
I (740) cpu_start: App version:      v1.0
I (745) cpu_start: Compile time:     Apr 18 2023 17:17:43
I (751) cpu_start: ELF file SHA256:  8b16f2d470db2749...
I (757) cpu_start: ESP-IDF:          v4.4.4
I (762) heap_init: Initializing. RAM available for dynamic allocation:
I (769) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (775) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (781) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (787) heap_init: At 3FFD49D0 len 0000B630 (45 KiB): DRAM
I (793) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (800) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (806) heap_init: At 4009D52C len 00002AD4 (10 KiB): IRAM
I (814) spi_flash: detected chip: generic
I (817) spi_flash: flash io: dio
W (821) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (837) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (868) led_driver_gpio: Initializing light driver
I (868) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
W (928) esp_matter_core: Command 0x00000047 on cluster 0x00000300 already exists. Not creating again.
I (928) app_main: Light created with endpoint_id 1
I (958) wifi:wifi driver task: 3ffdbc34, prio:23, stack:6656, core=0
I (958) system_api: Base MAC address is not set
I (958) system_api: read default base MAC address from EFUSE
I (968) wifi:wifi firmware version: 6567a16
I (968) wifi:wifi certification version: v7.0
I (968) wifi:config NVS flash: enabled
[... 474 lines cut ...]
I (35958) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000021C713E1
I (36968) chip[EM]: <<< [E:6330r S:0 M:185503352 (Ack:44652592)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (36978) chip[IN]: (U) Sending msg 185503352 to IP address 'UDP:[FE80::C2:D18A:68BF:B28B%st1]:56036'
I (36988) chip[SC]: Sent Sigma2 msg
E (36988) chip[DL]: Long dispatch time: 1089 ms, for event type 3
I (37068) chip[EM]: >>> [E:6330r S:0 M:44652593 (Ack:185503352)] (U) Msg RX from 0:A3EBC84558C3FD0D [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (37068) chip[EM]: <<< [E:6330r S:0 M:185503353 (Ack:44652593)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (37088) chip[IN]: (U) Sending msg 185503353 to IP address 'UDP:[FE80::C2:D18A:68BF:B28B%st1]:56036'
I (37098) chip[SC]: Received Sigma3 msg
E (37098) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (38058) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
E (38488) chip[CR]: mbedTLS error: ECP - The signature is not valid
E (38488) chip[DL]: Long dispatch time: 1372 ms, for event type 2
I (38488) chip[EM]: <<< [E:6330r S:0 M:185503354 (Ack:44652593)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (38508) chip[IN]: (U) Sending msg 185503354 to IP address 'UDP:[FE80::C2:D18A:68BF:B28B%st1]:56036'
E (38518) chip[IN]: CASE Session establishment failed: 14
I (38518) chip[IN]: CASE Server enabling CASE session setups
I (38608) chip[EM]: >>> [E:6330r S:0 M:44652594 (Ack:185503354)] (U) Msg RX from 0:A3EBC84558C3FD0D [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (40988) NimBLE: GATT procedure initiated: indicate; 
I (40988) NimBLE: att_handle=14

I (41028) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (43548) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (46048) NimBLE: GATT procedure initiated: indicate; 
I (46048) NimBLE: att_handle=14```

Tim4497, 2023-04-20:
I saw that you updated the requirements for esp-idf with the newest esp-matter version, so I downgraded to branch `matter_1_1/sve` but I know face the issue:
```
Installing zap-cli
mv: zap/ is not a directory
```

Tim4497, 2023-04-20:
Switching to `release/v1.0` gives me:

```
 BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (3.8s)
  Setting up Python environment.....[|]
ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/
Command: python3 /Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/build/config/linux/pkg-config.py openssl
Returned 1.
stderr:

Package openssl was not found in the pkg-config search path.
Perhaps you should add the directory containing `openssl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openssl' found
Could not run pkg-config.

See //src/crypto/BUILD.gn:84:3: whence it was called.
  pkg_config("openssl_config") {
  ^-----------------------------
See //BUILD.gn:152:7: which caused the file to be included.
      "${chip_root}/src/crypto",
      ^------------------------
['gn', 'gen', '/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 335, in install_packages
    subprocess.check_call(
  File "/Users/t.winkler/.pyenv/versions/3.8.16/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
    sys.exit(main())
  File "/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
    result = step(spin)
  File "/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 661, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 370, in install
    install_packages(gn_target)
  File "/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 343, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/t.winkler/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
```

Tim4497, 2023-04-20:
I was able to comission the device via the ESP Matter iPhone App, after comissioning I opened the commissioning window again and tried to add the device in Apple Home via the setup code given.

On connecting the device in Apple Home I get the following logs:

```
I (379302) app_main: Commissioning window opened
I (379372) chip[EM]: >>> [E:45360r S:6710 M:242052366 (Ack:77976992)] (S) Msg RX from 1:308F68197EA962E2 [0FE5] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (413832) chip[EM]: >>> [E:25129r S:0 M:65139185] (U) Msg RX from 0:D0D70702835A20A6 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (413832) chip[EM]: <<< [E:25129r S:0 M:128088303 (Ack:65139185)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (413852) chip[IN]: (U) Sending msg 128088303 to IP address 'UDP:192.168.2.105%st2:61568'
I (413862) chip[EM]: <<< [E:25129r S:0 M:128088304 (Ack:65139185)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
I (413882) chip[IN]: (U) Sending msg 128088304 to IP address 'UDP:192.168.2.105%st2:61568'
I (414242) chip[IN]: (U) Sending msg 128088304 to IP address 'UDP:192.168.2.105%st2:61568'
I (414322) chip[EM]: >>> [E:25129r S:0 M:65139186 (Ack:128088304)] (U) Msg RX from 0:D0D70702835A20A6 [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
I (414342) chip[EM]: <<< [E:25129r S:0 M:128088305 (Ack:65139186)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (414352) chip[IN]: (U) Sending msg 128088305 to IP address 'UDP:192.168.2.105%st2:61568'
I (415242) chip[EM]: <<< [E:25129r S:0 M:128088306 (Ack:65139186)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)
I (415262) chip[IN]: (U) Sending msg 128088306 to IP address 'UDP:192.168.2.105%st2:61568'
E (415262) chip[DL]: Long dispatch time: 936 ms, for event type 3
I (415262) chip[EM]: >>> [E:25129r S:0 M:65139187 (Ack:128088304)] (U) Msg RX from 0:D0D70702835A20A6 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (415562) chip[EM]: >>> [E:25129r S:0 M:65139188 (Ack:128088306)] (U) Msg RX from 0:D0D70702835A20A6 [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
I (415572) chip[EM]: <<< [E:25129r S:0 M:128088307 (Ack:65139188)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (415592) chip[IN]: (U) Sending msg 128088307 to IP address 'UDP:192.168.2.105%st2:61568'
I (415592) chip[EM]: <<< [E:25129r S:0 M:128088308 (Ack:65139188)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (415612) chip[IN]: (U) Sending msg 128088308 to IP address 'UDP:192.168.2.105%st2:61568'
I (415622) chip[SC]: SecureSession[0x3ffc87f0]: Moving from state 'kEstablishing' --> 'kActive'
I (415622) chip[SVR]: Commissioning completed session establishment step
I (415642) chip[DIS]: Updating services using commissioning mode 0
I (415642) chip[DIS]: CHIP minimal mDNS started advertising.
I (415672) chip[DIS]: Advertise operational node 8FE07CC066120FE5-0000000000000001
I (415672) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 8FE07CC066120FE5-0000000000000001.
I (415682) chip[DIS]: mDNS service published: _matter._tcp
I (415692) chip[SVR]: Device completed Rendezvous process
E (415692) chip[DL]: Long dispatch time: 130 ms, for event type 3
I (415702) app_main: Commissioning session started
I (415702) chip[EM]: >>> [E:25129r S:0 M:65139189 (Ack:128088308)] (U) Msg RX from 0:D0D70702835A20A6 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (415722) chip[EM]: >>> [E:25130r S:6712 M:219321523] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (415732) app_main: Commissioning window closed
I (415742) chip[EM]: <<< [E:25130r S:6712 M:235834080 (Ack:219321523)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (415752) chip[IN]: (S) Sending msg 235834080 on secure session with LSID: 6712
I (415832) chip[EM]: >>> [E:25130r S:6712 M:219321524 (Ack:235834080)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (415842) chip[EM]: >>> [E:25131r S:6712 M:219321525] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (415852) chip[EM]: <<< [E:25131r S:6712 M:235834081 (Ack:219321525)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (415872) chip[IN]: (S) Sending msg 235834081 on secure session with LSID: 6712
I (415922) chip[EM]: >>> [E:25131r S:6712 M:219321526 (Ack:235834081)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (415952) chip[EM]: >>> [E:25132r S:6712 M:219321527] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (415962) chip[EM]: <<< [E:25132r S:6712 M:235834082 (Ack:219321527)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (415982) chip[IN]: (S) Sending msg 235834082 on secure session with LSID: 6712
I (416012) chip[EM]: >>> [E:25132r S:6712 M:219321528 (Ack:235834082)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (416782) chip[EM]: >>> [E:25133r S:6712 M:219321529] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (416792) chip[EM]: <<< [E:25133r S:6712 M:235834083 (Ack:219321529)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[... cut ...]
I (533082) chip[EM]: <<< [E:38766r S:6714 M:267040212 (Ack:205258176)] (S) Msg TX to 1:308F68197EA962E2 [0FE5] --- Type 0001:09 (IM:InvokeCommandResponse)
I (533092) chip[IN]: (S) Sending msg 267040212 on secure session with LSID: 6714
I (533112) chip[EM]: >>> [E:38767r S:6714 M:205258177 (Ack:267040210)] (S) Msg RX from 1:308F68197EA962E2 [0FE5] --- Type 0001:08 (IM:InvokeCommandRequest)
```

As I can commission the device somehow with my phone I'm grand now, but If you have time please have a look into it.

jadhavrohit924, 2023-04-26:
@Tim4497 Just to clarify.

1. Did you able to commission with `Apple Home` and `chip-tool`?
2. You are facing problem to open commissioning window with  `ESP Matter iPhone App`?

Please mention any other issue you are facing if I missed it.

Please put the logs in a file and attach the file here. Analysing the copy-pasted is very hard.

Tim4497, 2023-04-26:
> 1. Did you able to commission with `Apple Home` and `chip-tool`?

No, the only way I can commission is to do while idf is monitoring, sending the command `matter esp wifi connect <ssid> <password>`

Only after that I can commission via chip tool or with the ESP Matter Comissionair App: [Espressif-Matter](https://apps.apple.com/de/app/espressif-matter/id1604739172)

dhrishi, 2023-04-29:
@Tim4497 I assume that you are using a hub (like a HomePod or AppleTv) that is mandatory in the setup. Please check if the hub and the iPhone/iPad are connected to the same AP that has internet connectivity

Tim4497, 2023-05-02:
> @Tim4497 I assume that you are using a hub (like a HomePod or AppleTv) that is mandatory in the setup. Please check if the hub and the iPhone/iPad are connected to the same AP that has internet connectivity

I use a AppleTV newest Generation its on the same network and the latest version

jadhavrohit924, 2023-05-05:
@Tim4497 
The error comes up when `the public key in NOC is not the one it singed`. 
We have not seen this issue you are facing till now. We will try to reproduce it. Can you please share all the steps you have followed while commissioning with apple right from HomePod or AppleTv setup?

Tim4497, 2023-05-11:
I have a Apple TV, it seems to work now, maybe it was an Update on one of my devices..

closed 2023-05-11 by Tim4497