espressif/esp-matter #338: Alexa pair failed: Fail-safe timer expired

sidhoo2, 2023-04-20:
esp-matter commitid: commit 9bb410d995b6ae81e2025467e5d5bbf494c1e6b1 (HEAD -> release/v1.0, origin/release/v1.0)
andriod alexa app version:2023.8 build version:2.2.5.6769.0
esp32c3

I (50316) chip[EM]: >>> [E:64068r M:74845101] (U) Msg RX from 0:4087D8524D4CA55B [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (50330) chip[EM]: <<< [E:64068r M:117983163] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
I (50341) chip[IN]: (U) Sending msg 117983163 to IP address 'BLE'
W (50349) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (50356) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (50364) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
W (50375) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
E (50385) EF-E: [bl0942_read_register:159]recv bl0942 timeout
I (50391) EF-I: [bl0942_data_reset:207]uart1 init
E (50435) EF-E: [bl0942_data_update:330]Addr_I_RMS Error
E (50495) EF-E: [bl0942_read_register:159]recv bl0942 timeout
I (50495) EF-I: [bl0942_data_reset:207]uart1 init
E (50535) EF-E: [bl0942_data_update:343]Addr_V_RMS Error
E (50596) EF-E: [bl0942_read_register:159]recv bl0942 timeout
I (50596) EF-I: [bl0942_data_reset:207]uart1 init
E (50636) EF-E: [bl0942_data_update:363]Addr_WATT Error
E (50696) EF-E: [bl0942_read_register:159]recv bl0942 timeout
I (50696) EF-I: [bl0942_data_reset:207]uart1 init
I (50716) EF-I: [post_timetask_info:708][post_timetask_info] index[2], need_send_task[1], iot_ack[1], ble_ack[1]
I (50716) EF-I: [post_one_timetask_info:631]read timetask info id[2], cfg[0], en[0], time[0:0:0]
I (50723) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (50734) EF-I: [post_timetask_info_by_ble:613]post timetask info to ble
I (50741) chip[EM]: >>> [E:64068r M:74845102] (U) Msg RX from 0:4087D8524D4CA55B [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
E (50748) EF-E: [bl0942_data_update:392]Addr_FREQ Error
W (50924) EF-W: [memoryMonitoringTasks:43]current free heap:46.09K inter free heap:39.84K min free heap:45.86K

I (51224) EF-I: [post_timetask_info:708][post_timetask_info] index[3], need_send_task[1], iot_ack[1], ble_ack[1]
I (51228) EF-I: [post_one_timetask_info:631]read timetask info id[3], cfg[0], en[0], time[0:0:0]
I (51237) EF-I: [post_timetask_info_by_ble:613]post timetask info to ble
E (51286) EF-E: [bl0942_send_byte:102]send byte failed
E (51351) EF-E: [bl0942_read_register:159]recv bl0942 timeout
I (51353) EF-I: [bl0942_data_reset:207]uart1 init
E (51408) EF-E: [bl0942_data_update:343]Addr_V_RMS Error
E (51473) EF-E: [bl0942_read_register:159]recv bl0942 timeout
I (51477) EF-I: [bl0942_data_reset:207]uart1 init
E (51532) EF-E: [bl0942_data_update:363]Addr_WATT Error
I (51596) chip[EM]: <<< [E:64068r M:117983164] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)
E (51597) EF-E: [bl0942_read_register:159]recv bl0942 timeout
I (51598) chip[IN]: (U) Sending msg 117983164 to IP address 'BLE'
I (51605) EF-I: [bl0942_data_reset:207]uart1 init
E (51616) chip[DL]: Long dispatch time: 875 ms, for event type 7
W (51623) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (51631) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (51629) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (51650) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0
W (51651) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (51664) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (51673) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
E (51679) EF-E: [bl0942_data_update:392]Addr_FREQ Error
I (51712) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (51714) chip[EM]: >>> [E:64068r M:74845103] (U) Msg RX from 0:4087D8524D4CA55B [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
I (51724) chip[EM]: <<< [E:64068r M:117983165] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (51735) chip[IN]: (U) Sending msg 117983165 to IP address 'BLE'
I (51730) EF-I: [post_timetask_info:708][post_timetask_info] index[4], need_send_task[1], iot_ack[1], ble_ack[1]
I (51752) EF-I: [post_one_timetask_info:631]read timetask info id[4], cfg[0], en[0], time[0:0:0]
I (51762) EF-I: [post_timetask_info_by_ble:613]post timetask info to ble
I (51766) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (51752) chip[SC]: SecureSession[0x3fc9d820]: Moving from state 'kEstablishing' --> 'kActive'
I (51788) chip[SVR]: Commissioning completed session establishment step
I (51796) chip[DIS]: Updating services using commissioning mode 0
I (51802) chip[DIS]: CHIP minimal mDNS started advertising.
I (51809) chip[SVR]: Device completed Rendezvous process
E (51815) chip[DL]: Long dispatch time: 102 ms, for event type 7
W (51821) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
W (51830) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
W (51838) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
[... 202 lines cut ...]

I (55525) NimBLE: 

I (55528) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
W (55535) NimBLE_BLE_PRPH: ble_send_task is stopping
W (55541) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
I (55857) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0
W (55857) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
W (55860) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
I (56857) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0
W (56857) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
W (56860) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
W (57125) EF-W: [memoryMonitoringTasks:43]current free heap:53.05K inter free heap:46.80K min free heap:44.39K

I (57857) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0
W (57857) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
W (57860) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
I (57945) EF-I: [post_ems_history_data_task:462][post_ems_history_data_task] energy file index[1], energy seq[2]
I (57945) EF-I: [post_ems_history_data_task:487][post_ems_history_data_task] power file index[2], offset[48], seq[2]
I (58857) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0
W (58857) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
W (58860) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:0, kAdvertisingEnabled:1, no connets:1
W (59126) EF-W: [memoryMonitoringTasks:43]current free heap:53.05K inter free heap:46.80K min free heap:44.39K

I (59857) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0

wqx6, 2023-04-21:
What DAC&PAI certificates are you using? Looks like the BLE connection terminated after the Alexa verified the Device Attestation Certificates Chain? There might be some issues on your DAC&PAI certificates.

sidhoo2, 2023-04-21:
@wqx6 TEST DAC and CD certificates; different pid would reprt this error ?

wqx6, 2023-04-21:
Could you tell me the PID that you're using? The TEST DACs are hardcoded in `connectedhomeip/src/credentials/examples/ExampleDACs.cpp` and only support VID 0xFFF1-0xFFF3 and PID 0x8000-0x801F. If your PID falls outside this range, it may cause issues. 

Also if your want to use other VID&PID you can use the [mfg_tool](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool) to generate the DAC chain with the [Non-VID root certificate](https://github.com/espressif/connectedhomeip/blob/bd17b9f52f41eeac5aaa74c1ecf3a76e06dd6958/credentials/test/attestation/Chip-Test-PAA-NoVID-Cert.pem) and its private key. and generate the CD from the [signing cert](https://github.com/espressif/connectedhomeip/blob/bd17b9f52f41eeac5aaa74c1ecf3a76e06dd6958/credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem) and it key.

sidhoo2, 2023-04-23:
@wqx6  yes, i use the public vid and pid , fff1 \ 8000   in the SDK; but i cannot find the CD in this dir :connectedhomeip/connectedhomeip/credentials/test/certification-declaration/

sidhoo2, 2023-04-23:
I (69887) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (69903) chip[EM]: >>> [E:15203r M:244812453] (U) Msg RX from 0:52E21B18C2FA45DC [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (69917) chip[EM]: <<< [E:15203r M:31424385] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
I (69928) chip[IN]: (U) Sending msg 31424385 to IP address 'BLE'
I (69931) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0
W (69938) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (69949) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (69958) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (69967) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (70002) EF-I: [post_timetask_info:708][post_timetask_info] index[2], need_send_task[1], iot_ack[1], ble_ack[1]
I (70002) EF-I: [post_one_timetask_info:631]read timetask info id[2], cfg[0], en[0], time[0:0:0]
I (70011) EF-I: [post_timetask_info_by_ble:613]post timetask info to ble
I (70021) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
W (70028) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (70157) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (70159) chip[EM]: >>> [E:15203r M:244812454] (U) Msg RX from 0:52E21B18C2FA45DC [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
W (70188) EF-W: [memoryMonitoringTasks:43]current free heap:43.61K inter free heap:37.35K min free heap:43.39K

I (70508) EF-I: [post_timetask_info:708][post_timetask_info] index[3], need_send_task[1], iot_ack[1], ble_ack[1]
I (70508) EF-I: [post_one_timetask_info:631]read timetask info id[3], cfg[0], en[0], time[0:0:0]
I (70518) EF-I: [post_timetask_info_by_ble:613]post timetask info to ble
W (70825) ledc: LEDC FADE TOO FAST
I (70931) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0
I (71008) chip[EM]: <<< [E:15203r M:31424386] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)
I (71009) chip[IN]: (U) Sending msg 31424386 to IP address 'BLE'
I (71015) EF-I: [post_timetask_info:708][post_timetask_info] index[4], need_send_task[1], iot_ack[1], ble_ack[1]
E (71018) chip[DL]: Long dispatch time: 861 ms, for event type 7
W (71033) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (71042) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (71050) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
W (71059) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (71027) EF-I: [post_one_timetask_info:631]read timetask info id[4], cfg[0], en[0], time[0:0:0]
I (71078) EF-I: [post_timetask_info_by_ble:613]post timetask info to ble
I (71101) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
W (71101) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (71102) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (71118) chip[EM]: >>> [E:15203r M:244812455] (U) Msg RX from 0:52E21B18C2FA45DC [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
I (71131) chip[EM]: <<< [E:15203r M:31424387] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (71142) chip[IN]: (U) Sending msg 31424387 to IP address 'BLE'
I (71150) chip[SC]: SecureSession[0x3fc9ddb0]: Moving from state 'kEstablishing' --> 'kActive'
I (71158) chip[SVR]: Commissioning completed session establishment step
I (71165) chip[DIS]: Updating services using commissioning mode 0
I (71172) chip[DIS]: CHIP minimal mDNS started advertising.
I (71178) chip[SVR]: Device completed Rendezvous process
W (71184) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
W (71193) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (71201) app_main: Commissioning session started
W (71207) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (71215) app_main: Commissioning window closed
I (71236) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
W (71236) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (71267) EF-I: [main_task_wdt_feed_task:64]feed watch dog for main task
I (71282) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (71284) chip[EM]: >>> [E:15204r M:71401006] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
W (71293) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (71303) chip[DL]: NVS set: chip-config/reg-location = 0 (0x0)
I (71307) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is ld **********
I (71319) chip[EM]: <<< [E:15204r M:156111895] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (71329) chip[IN]: (S) Sending msg 156111895 on secure session with LSID: 62343
W (71339) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (71461) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
W (71462) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
[... cut ...]
I (71619) chip[IN]: (S) Sending msg 156111896 on secure session with LSID: 62343
W (71629) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
W (71636) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (71731) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
W (71731) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
W (71837) ledc: LEDC FADE TOO FAST
I (71931) CHIP[DL]: BLEManagerImpl::_SetAdvertisingMode:0

wqx6, 2023-04-24:
> @wqx6 yes, i use the public vid and pid , fff1 \ 8000 in the SDK; but i cannot find the CD in this dir :connectedhomeip/connectedhomeip/credentials/test/certification-declaration/

If you use the ExampleDACProvider the certificates in DAC chain and the CD are hardcoded in `connectedhomeip/src/credentials/examples/DeviceAttestationCredsExample.cpp`. You can see an array named `kCdForAllExamples` which is the CD for all the PIDs(0x8000-0x8063) and VIDs(0xFFF1-0xFFF3).

sidhoo2, 2023-04-26:
i will retesting alexa and new device with pravite DAC and pair code again, continue...

dhrishi, 2023-05-06:
Closing this. Please re-open if you still face issues with your DAC provider

closed 2023-05-06 by dhrishi

sidhoo2, 2023-05-21:
The reason is that the firmware for generating network configuration is missing essential device information.