espressif/esp-matter #339: esp_matter_attribute_utils::val_print() - missing data type

YOGYUI, 2023-04-20:
Source code: esp_matter_attrubite_utils.cpp
Function name: val_print
Issue: missing conditional statement for ESP_MATTER_VAL_TYPE_INT8
example:
```
I (301759) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000202's Attribute 0x00000000 is <invalid type: 7> **********
```

shripad621git, 2023-04-25:
The example log shared is the `FanMode `attribute (0x0000) of` Fan Control Cluster`(0x0202)and it is of the type `enum8.` The check for  `ESP_MATTER_VAL_TYPE_ENUM8 `is present in the `val_print()` function.
The missing conditional check for `ESP_MATTER_VAL_TYPE_INT8` has been addressed in the PR [342](https://github.com/espressif/esp-matter/pull/342).

YOGYUI, 2023-04-25:
@shripad621git OK, I got it. I'll wait for next commit. Thanks.

closed 2023-04-25 by YOGYUI