espressif/esp-matter #341: install.sh fail

maenkai, 2023-04-23:
esp-idf: v4.4.3
esp-matter: fde2c

```shell
  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (2.1s)
  Setting up Python environment.....done (6.5s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

Usage: pw [OPTIONS] [USER@][KEY] [USER]
Try 'pw --help' for help.

Error: no such option: --no-banner

maenkai, 2023-04-23:
Previous versions did not have this problem. Why do these environmental problems always occur every time I pull the latest code of esp-matter?

wqx6, 2023-04-24:
Could you please remove the pigweed environment and try again?
```
rm -rf $ESP_MATTER/connectedhomeip/connectedhomeip/.environment/
```

closed 2023-04-25 by maenkai