espressif/esp-matter #347: [BUG] Error flashing Matter light example into esp32c3

LinJiaDeng, 2023-04-26:
**Reproduction steps**
I'm following the following espressif documentation for Matter implementation:
[Espressif documentation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/developing.html)

Installed the prerequisites for ESP IDF:

```
mkdir -p ~/esp cd ~/esp git clone -b v4.4.3 --recursive https://github.com/espressif/esp-idf.git

cd ~/esp/esp-idf ./install.sh esp32
```

Getting the Repositories:

```
cd esp-idf
source ./export.sh
cd ..

git clone --depth 1 https://github.com/espressif/esp-matter.git
cd esp-matter
git submodule update --init --depth 1
cd ./connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 linux --shallow
cd ../..
./install.sh
cd ..
```

After getting the Repositories:

```
cd esp-idf; source ./export.sh; cd ..
cd esp-matter; source ./export.sh; cd..
cd esp-matter/examples/light
idf.py set-target esp32c3
idf.py -p /dev/ttyUSB0 flash
```

**Bug prevalence**
Everytime I try to flash an example

**GitHub hash of the SDK that was being used**
Espressif

**Platform**
esp32

**Platform Version(s)**
No response

**Anything else?**
Screenshot of error message

[image]

(Already tried to reinstall ESP-IDF)

shripad621git, 2023-04-26:
Please refer to the note [here](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/get-started/index.html) mentioned in step 3. Please give a try by doing the steps below:
```
cd ~/esp/esp-idf
./install.sh esp32c3  (for esp32c3 as target)
. ./export.sh
cd path/to/esp-matter
./install.sh
. ./export.sh
```

LinJiaDeng, 2023-04-27:
Already tried these steps but it gave me the same error so I decided to start from the beginning.

Following the [document](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/developing.html) mentioned before, I started with the **2.1.1 Host Setup** (I tried to use a virtual enviroment with Ubuntu 22.04 LTS and also a Raspberry Pi 3 also with Ubuntu Server 22.04 LTS), I was able to follow the [Prerequisites for ESP IDF](https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32c3/get-started/index.html) (version 4.4.4 for esp32c3) without a problem but I can't say the same for the [Prerequisites for Matter](https://github.com/espressif/connectedhomeip/tree/bb9200ace/docs/guides/BUILDING.md#prerequisites). 

First, the link does not work (it redirects you to a non-existing github site) I was able to find [the webpage](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#prerequisites) in another site. 
The first command `git clone --recurse-submodules git@github.com:project-chip/connectedhomeip.git` pops up an error related with the permission (publickey, I've done a research and is something that I have to configure) so I used a similar command that I believe it does the same `git clone --recurse-submodules https://github.com/project-chip/connectedhomeip.git`. After that I continued with the **Prepare for building** section (the **Synchronizing submodules** and **Prerequisites**  sections didn't gave me problems) where it appeared a new error related with the ./enviroment/env_root.txt when I run the command: `source scripts/activate.sh`.

dhrishi, 2023-04-27:
@LinJiaDeng 

>First, the link does not work (it redirects you to a [non-existing github website](https://github.com/espressif/connectedhomeip/tree/bb9200ace/docs/guides/BUILDING.md#prerequisites)) but I was able to find [the webpage](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#prerequisites) in another site.

Thanks for reporting. We will fix the link.

> The first command git clone --recurse-submodules git@github.com:project-chip/connectedhomeip.git pops up an error related with the permission (publickey, I've done a research and is something that I [...]

You should just follow the pre-requisites from the connectedhomeip repo as mentioned. The other steps should be from the esp-matter documentation [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/developing.html#getting-the-repositories)

You can retry by deleting the `esp-matter/connectedhomeip/connectedhomeip/.environment` folder. If you still face issues, then please upload (and not paste) the set of entire commands you execute.

LinJiaDeng, 2023-05-01:
Hi @dhrishi, thanks for your response.
I was able to follow the pre-requisites from the connectedhomeip repository, but now I'm facing a new issue with the esp-matter documentation, specifically, when trying to flash the firmware.
Below it is attached a screenshot of the error and all the commands used, if you need more information, please let me know.
[image]

[Commands used.txt](https://github.com/espressif/esp-matter/files/11366468/Commands.used.txt)

LinJiaDeng, 2023-05-01:
I've tried again entering the commands

`idf.py set-target esp32c3`
`idf.py build`
`idf.py -p /dev/ttyUSB0 flash`

And it worked, closing this post.

closed 2023-05-01 by LinJiaDeng