espressif/esp-matter #349: Update periodically attribute

glennmullerar, 2023-04-26:
Hello,

I am using BLE for my master's degree and I want to create a temperature sensor endpoint that updates every minute. I managed to build the endpoint, but I have trouble changing the value of the Measured Value attribute. 

I tried to do it from a freertos task, but it makes my whole program restart. 

However, I can do it from a callback of the example light_switch.

dhrishi, 2023-04-26:
@glennmullerar You can refer [this](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L62-#L70) to update the attribute locally. Set the value appropriately (in the link above, it just toggles the OnOff attribute)

glennmullerar, 2023-04-27:
thank you for your answer. 

However, I want to be able to update it every minute. For now I try to do it this way (see code below). But this way makes the application reboot (see message before reboot below). 

```cpp
void vReadSensor(void * pvParameters ){

    temperature_sensor_handle_t temp_sensor = (temperature_sensor_handle_t) (pvParameters);

    ESP_LOGI(TAG, "Read temperature");
    for( ;; )
    {
        ESP_ERROR_CHECK(temperature_sensor_get_celsius(temp_sensor, &tsens_value));
        ESP_LOGI(TAG, "Temperature value %.02f °C", tsens_value);

        ESP_LOGI(TAG, "Trying to update attribute");

        uint32_t cluster_id = 0x0402;
        uint32_t attribute_id = 0x0000;

        node_t *node = node::get();
        endpoint_t *endpoint = endpoint::get(node, temp_sensor_endpoint_id);
        cluster_t *cluster = cluster::get(endpoint, cluster_id);
        attribute_t *attribute = attribute::get(cluster, attribute_id);

        esp_matter_attr_val_t val = esp_matter_invalid(NULL);
        attribute::get_val(attribute, &val);
        val.val.i16 = (int16_t) tsens_value;

        attribute::update(temp_sensor_endpoint_id, cluster_id, attribute_id, &val);        

        vTaskDelay(pdMS_TO_TICKS(TASK_SLEEP_TIME_MS));
    }
}
```

Here is the error before reboot. 

```bash
***ERROR*** A stack overflow in task Read Sensor has been detected.
Core  0 register dump:
MEPC    : 0x403808a8  RA      : 0x4038b1f8  SP      : 0x3fc97360  GP      : 0x3fc94400  
0x403808a8: panic_abort at /Users/glennmullerardaya/esp/esp-idf/components/esp_system/panic.c:423

0x4038b1f8: __ubsan_include at /Users/glennmullerardaya/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x3fc740b4  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x00000001  
S0/FP   : 0x3fc97384  S1      : 0x3fc97384  A0      : 0x3fc97384  A1      : 0x3c102378  
A2      : 0x0000000f  A3      : 0x3fc973b5  A4      : 0x00000001  A5      : 0x3fca5000  
A6      : 0xa0000000  A7      : 0x0000000a  S2      : 0x00001881  S3      : 0x3fca5614  
S4      : 0x3fcdfbf8  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000001  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000001  S11     : 0x00000002  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000001  T6      : 0x9ffffffe  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
3fc97360: 0x00000000 0x00000000 0x00000000 0x4038e04c 0x00000000 0x00000000 0x3c10233c 0x3fcac3dc
0x4038e04c: esp_startup_start_app_common at /Users/glennmullerardaya/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:59

3fc97380: 0x3c102364 0x452a2a2a 0x524f5252 0x202a2a2a 0x74732041 0x206b6361 0x7265766f 0x776f6c66
3fc973a0: 0x206e6920 0x6b736174 0x61655220 0x65532064 0x726f736e 0x73616820 0x65656220 0x6564206e
3fc973c0: 0x74636574 0x002e6465 0x00000000 0x00000000 0x00000000 0x3fca55e0 0x3fcdfbd4 0x4038cac8
0x4038cac8: vTaskSwitchContext at /Users/glennmullerardaya/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3648 (discriminator 3)

3fc973e0: 0x00000000 0x3fca55dc 0x3fc97430 0x3fcac3ac 0x00000000 0x00000001 0x0000001d 0x00000001
3fc97400: 0x00000000 0x00000000 0x3fcdfbf8 0x00000001 0x00001881 0x80000004 0x3fcdfbd4 0x4038dac0
0x4038dac0: isr_skip_decrement at /Users/glennmullerardaya/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/portasm.S:88
[... 15 lines cut ...]
3fc974e0: 0x3fc974e4 0xffffffff 0x3fc974e4 0x3fc974e4 0x00000000 0x3fc974f8 0xffffffff 0x3fc974f8
3fc97500: 0x3fc974f8 0x00000000 0x3fc9750c 0xffffffff 0x3fc9750c 0x3fc9750c 0x00000000 0x3fc97520
3fc97520: 0xffffffff 0x3fc97520 0x3fc97520 0x00000000 0x3fc97534 0xffffffff 0x3fc97534 0x3fc97534
3fc97540: 0x00000000 0x3fc97548 0xffffffff 0x3fc97548 0x3fc97548 0x00000000 0x3fc9755c 0xffffffff
3fc97560: 0x3fc9755c 0x3fc9755c 0x00000000 0x3fc97570 0xffffffff 0x3fc97570 0x3fc97570 0x00000000
3fc97580: 0x3fc97584 0xffffffff 0x3fc97584 0x3fc97584 0x00000000 0x3fc97598 0xffffffff 0x3fc97598
3fc975a0: 0x3fc97598 0x00000000 0x3fc975ac 0xffffffff 0x3fc975ac 0x3fc975ac 0x00000000 0x3fc975c0
3fc975c0: 0xffffffff 0x3fc975c0 0x3fc975c0 0x00000000 0x3fc975d4 0xffffffff 0x3fc975d4 0x3fc975d4
3fc975e0: 0x00000000 0x3fc975e8 0xffffffff 0x3fc975e8 0x3fc975e8 0x00000000 0x3fc975fc 0xffffffff
3fc97600: 0x3fc975fc 0x3fc975fc 0x00000000 0x3fc97610 0xffffffff 0x3fc97610 0x3fc97610 0x00000000
3fc97620: 0x3fc97624 0xffffffff 0x3fc97624 0x3fc97624 0x00000001 0x3fc97638 0xffffffff 0x3fcb0b50
3fc97640: 0x3fcb0b50 0x00000000 0x3fc9764c 0xffffffff 0x3fc9764c 0x3fc9764c 0x00000000 0x3fc97660
3fc97660: 0xffffffff 0x3fc97660 0x3fc97660 0x00000006 0x3fc97674 0xffffffff 0x3fcdfde4 0x3fcac3ac
3fc97680: 0x00000000 0x3fc97688 0xffffffff 0x3fc97688 0x3fc97688 0x00000000 0x3fc97694 0x00000000
3fc976a0: 0x00000000 0x00000000 0x3fc976ac 0xffffffff 0x3fc976ac 0x3fc976ac 0x00000000 0x3fc976c0
3fc976c0: 0xffffffff 0x3fc976c0 0x3fc976c0 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000
3fc976e0: 0xb33fffff 0x00000000 0x00000000 0x3fc976e8 0x00000000 0x00000000 0x00000000 0x3fc97700
3fc97700: 0xffffffff 0x3fc97700 0x3fc97700 0x00000000 0x3fc97714 0xffffffff 0x3fc97714 0x3fc97714
3fc97720: 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000 0x00000009
3fc97740: 0x3fca7e70 0x3fca7ed8 0x3fca7f40 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000

ELF file SHA256: 956f16ad2972b0bc

Rebooting...
```

dhrishi, 2023-04-27:
@glennmullerar 

>***ERROR*** A stack overflow in task Read Sensor has been detected.

The stack of your Read Sensor task seems to be low to handle this operation. You need to increase the stack.

glennmullerar, 2023-04-28:
It works thanks for your help.

closed 2023-04-28 by glennmullerar