espressif/esp-matter #350: Custom/non-standard device type

Gonta01, 2023-04-26:
Is it possible to create custom/non-standard device type? 
In  Espressif's SDK for Matter guide (https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#adding-custom-data-model-fields) is described how to create custom endpoints, clusters, attributes, and commands that are not defined in the Matter specification but nothing said about custom device types.

dhrishi, 2023-04-27:
@Gonta01 There is no separate API for creation of a custom device type. Device type, as per the Matter specification, is the highest semantic element of the Matter data model that conforms to a set of one or more endpoints.

So, basically, you can create custom endpoint(s) and cluster(s) to create a sort of custom device type. Note that, any custom entity requires the other end (commissioner) to understandthe same.

dhrishi, 2023-04-29:
Closing this as the question is answered

closed 2023-04-29 by dhrishi