espressif/esp-matter #352: apple home successfully created fabric, but pair failed

sidhoo2, 2023-04-26:
IDF:4.4.3  
esp-matter:
commit 9bb410d995b6ae81e2025467e5d5bbf494c1e6b1 (HEAD -> release/v1.0)
Merge: 80a22dc 83c25e9
Author: Hrishikesh Dhayagude <[hrishi@espressif.com](mailto:hrishi@espressif.com)>
Date:   Fri Mar 31 12:25:30 2023 +0800

I (6780) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (6781) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (6787) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (6840) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (6840) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (6847) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (6857) chip[EM]: >>> [E:50500r M:159526131] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (6868) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (6878) chip[EM]: <<< [E:50500r M:237483382] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (6887) chip[IN]: (S) Sending msg 237483382 on secure session with LSID: 43616
I (6896) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (6930) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (6930) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (6931) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (6947) chip[EM]: >>> [E:50501r M:159526132] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (6959) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:0, no connets:0
I (6969) chip[DL]: NVS set: chip-config/reg-location = 0 (0x0)
I (6973) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is ld **********
I (6985) chip[EM]: <<< [E:50501r M:237483383] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (6994) chip[IN]: (S) Sending msg 237483383 on secure session with LSID: 43616
I (6996) EF-I: [post_timetask_info:708][post_timetask_info] index[7], need_send_task[1], iot_ack[1], ble_ack[1]
I (7013) EF-I: [post_one_timetask_info:631]read timetask info id[7], cfg[0], en[0], time[0:0:0]
I (7014) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7031) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7039) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7023) EF-I: [post_timetask_info_by_ble:613]post timetask info to ble
I (7050) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (7066) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7073) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7140) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (7140) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (7147) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7157) chip[EM]: >>> [E:50502r M:159526133] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (7169) esp_matter_command: Received command 0x00000000 for endpoint 0x0000's cluster 0x00000030
I (7177) chip[FS]: GeneralCommissioning: Received ArmFailSafe (60s)
I (7184) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is ld **********
I (7197) chip[EM]: <<< [E:50502r M:237483384] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (7207) chip[IN]: (S) Sending msg 237483384 on secure session with LSID: 43616
I (7217) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7224) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7260) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (7261) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (7267) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7277) chip[EM]: >>> [E:50503r M:159526134] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (7289) esp_matter_command: Received command 0x00000002 for endpoint 0x0000's cluster 0x00000030
I (7298) chip[DL]: NVS set: chip-config/reg-location = 0 (0x0)
I (7304) chip[DL]: NVS set: chip-config/country-code = "XX"
I (7310) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is ld **********
I (7323) chip[EM]: <<< [E:50503r M:237483385] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (7333) chip[IN]: (S) Sending msg 237483385 on secure session with LSID: 43616
I (7343) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7350) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7380) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (7381) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (7387) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7397) chip[EM]: >>> [E:50504r M:159526135] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (7409) esp_matter_command: Received command 0x00000002 for endpoint 0x0000's cluster 0x0000003E
I (7417) chip[ZCL]: OpCreds: Certificate Chain request received for PAI
I (7428) chip[EM]: <<< [E:50504r M:237483386] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (7437) chip[IN]: (S) Sending msg 237483386 on secure session with LSID: 43616
I (7447) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7453) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (7497) EF-I: [post_timetask_info:708][post_timetask_info] index[8], need_send_task[1], iot_ack[1], ble_ack[1]
[... 170 lines cut ...]
I (19590) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (19620) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (19621) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (19627) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (19637) chip[EM]: >>> [E:50511r M:159526142] (S) Msg RX from 1:FFFFFFFB00000000 [06B6] --- Type 0001:08 (IM:InvokeCommandRequest)
I (19650) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is ld **********
I (19661) chip[EM]: <<< [E:50511r M:237483393] (S) Msg TX to 1:FFFFFFFB00000000 [06B6] --- Type 0001:09 (IM:InvokeCommandResponse)
I (19671) chip[IN]: (S) Sending msg 237483393 on secure session with LSID: 43616
I (19681) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (19688) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (19740) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (19741) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (19748) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (19757) chip[EM]: >>> [E:50512r M:159526143] (S) Msg RX from 1:FFFFFFFB00000000 [06B6] --- Type 0001:08 (IM:InvokeCommandRequest)
I (19769) chip[NP]: ESP NetworkCommissioningDelegate: SSID: 华硕
I (19775) chip[DL]: Disconnecting WiFi station interface
E (19781) chip[DL]: esp_wifi_disconnect() failed: UNKNOWN ERROR
W (19778) ledc: LEDC FADE TOO FAST
I (19788) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (19977) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (19977) CHIP[DL]: kCHIPoBLEServiceMode_Enabled:1, kAdvertisingEnabled:1, no connets:0
I (20263) EF-I: [main_task_wdt_feed_task:64]feed watch dog for main task
W (20293) EF-W: [memoryMonitoringTasks:43]current free heap:47.27K inter free heap:41.02K min free heap:26.74K

W (20803) ledc: LEDC FADE TOO FAST

dhrishi, 2023-04-26:
@sidhoo2 Please do not paste the logs, instead upload the complete log file. The logs indicate that the Wi-Fi network connectivity wasn't established. Can you please check the SSID and passphrase, if correct.

Also, the application seems to be different than the standard examples. Please provide any patch/diff on top of an existing example for us to recreate the issue.

sidhoo2, 2023-04-27:
@dhrishi  ok, i will pay attention next time.  you are right , but it can paired sucessful some time with this SSID.

dhrishi, 2023-04-29:
@sidhoo2 Please check if the AP is far away from the device. Also, if possible, please reset to factory the ESP32 and upload the commissioning logs from the new boot in case of a failure.

dhrishi, 2023-05-16:
@sidhoo2 Any updates here? Please close the issue if resolved

sidhoo2, 2023-05-21:
okay,  can't find the reason, and i hv never meet this issue these days.

closed 2023-05-21 by sidhoo2