espressif/esp-matter #353: How to properly create custom data model?

Gonta01, 2023-04-27:
Espressif's SDK for Matter [guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#adding-custom-data-model-fields) describes custom data model creation. 
Also there is an example of custom cluster creation in [esp_matter_rainmaker](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L216).
Interesting approach was described [here](https://github.com/espressif/esp-matter/issues/154#issuecomment-1378975113).

Based on the resources above I can make assumptions about how to properly create custom data model.
1) One way is to create a node which will contain endpoint 0 by default. Next step is to create custom cluster(s) with attributes and commands inside endpoint 0. As I understand this approach is implemented in esp_matter_rainmaker. In this case cluster will be recognized only by own controller. 
2) Another approach is to create additional endpoint (for example 1) with some standard device type. Then add to this endpoint a custom clusters. In this case the clusters of device type will be recognized by Google Home app or any other matter controller but the custom cluster will be recognized only by own controller.

If my conclusions correct the following questions occur:
- it isn't clear what for non-standard endpoint [creation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#id1)?
- is it a good idea to add some clusters to endpoint 0 (as it is described in the first approach)?

dhrishi, 2023-04-27:
@Gonta01 

>it isn't clear what for non-standard endpoint [creation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#id1)?

This is just an example. If someone wants to create an entirely new endpoint.
 
>is it a good idea to add some clusters to endpoint 0 (as it is described in the first approach)?

We've added RainMaker cluster to endpoint 0 as its configuration is applicable to the entire node. Root node or endpoint 0 is in general applicable to the entire node. If you want to add a custom functionality that is specific to a specific device type, then you can add it on that endpoint. It totally depends on your usecase

dhrishi, 2023-04-27:
@Gonta01 
I am working internally to get the RainMaker + Matter example out (in the [esp-rainmaker](https://github.com/espressif/esp-rainmaker) repository). Although, I do not have concrete timelines as of now

Gonta01, 2023-04-28:
> This is just an example. If someone wants to create an entirely new endpoint.

I conducted some experiments.
1) In the first case I created node with endpoint 0 and added temperature measurement humidity measurement clusters to it. Then commissioned device by using Home Assistant. Everything works fine.
2) In the second case I created  node with endpoint 0 then created endpoint 1 with temperature sensor device type and added humidity measurement cluster to it. The result is the same as previous. 
3) In the third case I created node with endpoint 1 then created bare endpoint 1 and added temperature measurement and humidity measurement clusters to it. In this case device connected to Home Assistant but there weren't any entities to monitor temperature and humidity.

I assume that in the third case a matter controller (Home Assistant) haven't recognized Device Type ID inside endpoint and this is why it haven't recognized clusters IDs in the endpoint 1.
So matter controllers first identifies endpoint with device type ID, then cluster with cluster's ID and then attributed, commands, events with its IDs. 
Do we need to assign some device type id to the bare endpoint?

dhrishi, 2023-04-29:
@Gonta01 The descriptor cluster on each endpoint should atleast have one device type (from the device type specification) for that endpoint. So, yes, you should assign atleast one device type id for your endpoint having standard clusters.

For a custom endpoint, anyway the ecosystem controllers won't understand the device type id and the clusters on that endpoint.

dhrishi, 2023-05-06:
Closing as the question is answered

closed 2023-05-06 by dhrishi

Tim4497, 2023-05-11:
> For a custom endpoint, anyway the ecosystem controllers won't understand the device type id and the clusters on that endpoint.

What's the difference between a custom endpoint and a bare endpoint? I want to create a custom endpoint called e.g refrigerator and give it a on/off and a temperature setting dial.