espressif/esp-matter #354: [BUG] [Fan Control] Fan Mode can set 7

mmt-10, 2023-04-28:
## Reproduction steps

```bash
$ cd ${WORKSPACE}
$ source ./esp-idf-v5.0.1/export.sh
$ source ./esp-matter/export.sh
$ cd esp-matter/examples/light
$ idf.py set-target esp32c3
$ nano main/app_main.c
```

I modified as below.
[app_main.cpp.txt](https://github.com/espressif/esp-matter/files/11349994/app_main.cpp.txt)

```c
uint16_t fan_endpoint_id = 0;

extern "C" void app_main()
{
...
    fan::config_t fan_config;
    fan_config.fan_control.fan_mode = 0;
    endpoint_t *endpoint_fan = fan::create(node, &fan_config, ENDPOINT_FLAG_NONE, NULL);
...

    if (!node || !endpoint_fan) {
        ESP_LOGE(TAG, "Matter node creation failed : fan");
    }
...
    fan_endpoint_id = endpoint::get_id(endpoint_fan);
    ESP_LOGI(TAG, "Fan created with fan_endpoint_id %d", fan_endpoint_id);
```

```bash
idf.py build
idf.py erase-flash flash
idf.py monitor
```

on other terminal,
```bash
$ ./chip-tool pairing ble-wifi 1234 ${SSID} ${PASSWORD} 20202021 3840
$ ./chip-tool fancontrol write fan-mode 10 1234 2
$ ./chip-tool fancontrol read fan-mode 1234 2
```

> [1682647394.537201][89674:89676] CHIP:TOO: fan mode: 7

fan-mode was set 7.

As my understanding, FanMode should support range: 0 to 6 in Matter-1.0-Application-Cluster-Specification.

I cannot specify root cause `Matter SDK` or `esp-matter`.

[https://github.com/project-chip/connectedhomeip/issues/26288](https://github.com/project-chip/connectedhomeip/issues/26288)

## Environment
Host OS : Ubuntu 22.04
ESP-IDF :  v5.0.1 ( commit a4afa44435ef4488d018399e1de50ad2ee964be8 (HEAD, tag: v5.0.1) )
ESP-MATTER : commit 0b1762e5119c6cd4703ebb2166441187bff3fd9c (HEAD -> main, origin/main, origin/HEAD)
Target Board : ESP32-C3-DevKitM-1 (ESP32-C3-MINI-1)

dhrishi, 2023-04-28:
@mmt-10 We will check this. But note that, the fan control cluster is provisional in the Matter specification, for now. Which means that it isn't certifiable at this point.

jadhavrohit924, 2023-04-28:
@mmt-10 This behaviour is expected. Since you set fan mode value to `10`, it is `out of range and unknown` to the sdk. So it internally got set to `7` which is` kUnknownEnumValue`.

mmt-10, 2023-04-28:
Hi,
Thank you for your reply.

you mean root cause is `Matter SDK` ?

Matter SDK rejects `out of range` value as below.
[https://github.com/project-chip/connectedhomeip/issues/26288#issuecomment-1526963176](https://github.com/project-chip/connectedhomeip/issues/26288#issuecomment-1526963176)

jadhavrohit924, 2023-04-28:
> you mean root cause is Matter SDK ?

Since esp-matter's connectedhomeip submodule is on `bd17b9f` and it is working as mentioned here https://github.com/project-chip/connectedhomeip/issues/26288#issuecomment-1526963176. So looks like Matter sdk is not root cause.

We will look into it.

bzbarsky-apple, 2023-04-28:
You need to be doing the XML-defined range checks, or rejecting kUnknownEnumValue or both.

And no matter what, kUnknownEnumValue absolutely must _NOT_ be sent on the wire.  This is a hard requirement for forward compat.

jadhavrohit924, 2023-04-28:
Correct. Thanks Boris!!

closed 2023-04-29 by espressif-bot