espressif/esp-matter #358: Received query image failure

SYESHASWINI, 2023-04-29:
I have announced the ota from the provider , but the ota is not successful the query image getting failed.
Attaching the chip-tool log.
[chip-tool query faild.txt](https://github.com/espressif/esp-matter/files/11359030/chip-tool.query.faild.txt)
[ota-announce.txt](https://github.com/espressif/esp-matter/files/11359032/ota-announce.txt)

jadhavrohit924, 2023-05-02:
@SYESHASWINI Could you please attach whole (Not the failed only) logs? If you provide us 3 log files of `ota-provider, ota-requestor and chip-tool` stating from the commissioning would be really helpful to resolve your issue.

SYESHASWINI, 2023-05-03:
@jadhavrohit924 the query image is successful now. 

The software version is showing 0 before and after ota.  In the log after ota, I came across **E (2742) chip[SWU]:** Failed to confirm image: 3, what does this mean.

When I commission through google home app the software version is 2, where the version number is increased. But in the basic-information cluster it is '0'.

While I try to commission through the console using **matter ota query** it gives Error as 47. Can we test ota without using chip-tool.

Attaching the logs
[basic_read_requestor.txt](https://github.com/espressif/esp-matter/files/11380085/basic_read_requestor.txt)
[after_ota.txt](https://github.com/espressif/esp-matter/files/11380111/after_ota.txt)

jadhavrohit924, 2023-05-03:
@SYESHASWINI  
`Failed to confirm image: 3` comes when there is mismatch between the versions ( i.e version provided by provider and version of ota-requestor after bootup/applyUpdate ) see [here](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/OTAImageProcessorImpl.cpp#L78-L81)

I'm not getting this error while performing ota. Since you have not provided me full logs, I am not sure why you are failing at this stage.

Please provide us full logs.

jadhavrohit924, 2023-05-04:
@SYESHASWINI 
> While I try to commission through the console using matter ota query it gives Error as 47. Can we test ota without using chip-tool.

Yes we can use chip shell for that.

Steps:

1. Commission the ota-provider and ota-requestor app
2. Write into acl of ota-provider
3. Write the default ota provider into the ota-requestor using command. ``` ./out/debug/chip-tool otasoftwareupdaterequestor write default-otaproviders '[{"fabricIndex": 1, "providerNodeID": 1234, "endpoint": 0}, {"fabricIndex": 1, "providerNodeID": 20, "endpoint": 0}]' <REQUESTOR_NODE_ID> 0 ```
4. Then use `matter ota query ` command

SYESHASWINI, 2023-05-04:
@jadhavrohit924 attaching the error logs of confirm ota image failure.
[chip-tool log.txt](https://github.com/espressif/esp-matter/files/11394184/chip-tool.log.txt)
[ota_requestor2.txt](https://github.com/espressif/esp-matter/files/11394185/ota_requestor2.txt)
[Ota_rquestor1.txt](https://github.com/espressif/esp-matter/files/11394192/Ota_rquestor1.txt)

SYESHASWINI, 2023-05-04:
@jadhavrohit924 followed the steps and getting the error of 47. attached the log.
[Matter_ota_query_cmd.txt](https://github.com/espressif/esp-matter/files/11394350/Matter_ota_query_cmd.txt)

[image]

SYESHASWINI, 2023-05-04:
@jadhavrohit924 from the basic information cluster, before and after ota the ota-requestor version is 0 and ota-provider version is 1. But when I commission the device through google home, the ota-requestor version is 1 before and 2 after the successful ota.

jadhavrohit924, 2023-05-05:
@SYESHASWINI 
> matter ota query 1 0x728 0
Error: 2f
Error: 47

This is not the actual command.

You should use only `matter ota query`

>matter ota query
Done
E (96386) chip[SWU]: No suitable OTA Provider candidate found
E (96386) chip[SWU]: No OTA Providers available for immediate query

This error is self explanatory you have not set the provider. You are missing 3rd step.

Write the default ota provider into the ota-requestor using command. `./out/debug/chip-tool otasoftwareupdaterequestor write default-otaproviders '[{"fabricIndex": 1, "providerNodeID": 1234, "endpoint": 0}, {"fabricIndex": 1, "providerNodeID": 20, "endpoint": 0}]' <REQUESTOR_NODE_ID> 0`

jadhavrohit924, 2023-05-05:
> But when I commission the device through google home, the ota-requestor version is 1 before and 2 after the successful ota.

This means you have successfully done the ota using Google.

dhrishi, 2023-05-06:
I see that this is being tracked here https://github.com/project-chip/connectedhomeip/issues/26346. I will close the issue in this repo

closed 2023-05-06 by dhrishi

Shahmeera, 2024-01-10:
I am having the same error: chip[SWU]: Failed to confirm image: 3.
and I have project config 
CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
and my 
cmake_minimum_required(VERSION 3.5)

set(PROJECT_VER "v1.0")
set(PROJECT_VER_NUMBER 1)

and device logs are as below
I (681) cpu_start: Project name:     chip-lighting-app
I (687) cpu_start: App version:      v1.0
but as soon as I update the firmware, version remains same and then this error appears.