espressif/esp-matter #360: type_traits: No such file or directory

arpitsahu00740, 2023-04-29:
I want to add matter light to my code, i am referring the light example of esp-matter sdk. On following i have made the changes to the CMakeLists(attached file) and copied the zap-generated folder, but i am getting following error while building the project -

Traceback (most recent call last):
esp-matter/connectedhomeip/connectedhomeip/src/app/CommandSender.h:27:10: fatal error: type_traits: No such file or directory
 #include <type_traits>
[CMakeLists.txt](https://github.com/espressif/esp-matter/files/11358650/CMakeLists.txt)

arpitsahu00740, 2023-05-03:
set(PRIV_REQUIRES_LIST device esp_matter esp_matter_console app_reset)

```
idf_component_register(SRCS "${COMPONENT_SRCS}"
					   INCLUDE_DIRS ${COMPONENT_ADD_INCLUDEDIRS}
					   PRIV_REQUIRES      ${PRIV_REQUIRES_LIST}  
					  )
```
Over adding this to main CMakeLists, the error changes to be -
```
fatal error: cJSON.h: No such file or directory
 #include "cJSON.h"
```

shubhamdp, 2023-05-04:
@arpitsahu00740 I assume below error was solved after adding `esp-matter` component to components list in `main/CMakeLists.txt`.
```
Traceback (most recent call last):
esp-matter/connectedhomeip/connectedhomeip/src/app/CommandSender.h:27:10: fatal error: type_traits: No such file or directory
 #include <type_traits>
```

esp-matter do not use component `cJSON`, by any chance it is being used by your project?
If yes, then you would have to add that as a component dependency in the component using `cJSON`.

arpitsahu00740, 2023-05-04:
Before adding esp-matter components it was working fine, is problem caused due to incorrect integration?

shubhamdp, 2023-05-05:
esp-matter do not use `cJSON` component, so this error is not inherited because of esp-matter.

Below error is because a C compiler is trying to compile a C++ code, and it is unaware about that include file.
```
Traceback (most recent call last):
esp-matter/connectedhomeip/connectedhomeip/src/app/CommandSender.h:27:10: fatal error: type_traits: No such file or directory
 #include <type_traits>
```

To fix this you will have to change the `.c` files to `.cpp` and `extern "C"` the pure `C` functions that are being used from your code.

I am able to successfully integrate the esp-matter to idf's hello_world example, take a look at those changes here: https://github.com/shubhamdp/esp-idf/commit/14f2ba0ae79a45be6df9ead5c0ccd5ba08cd07fe

arpitsahu00740, 2023-05-05:
To implement matter, do I have to change whole project .c file to .cpp or there is any other way out?

shubhamdp, 2023-05-08:
@arpitsahu00740 does your project uses `cJSON` component? Also, if possible please attach the whole logs, it would let us know who is including the `cJSON.h`.

arpitsahu00740, 2023-05-08:
Yes, a file use the cJSON component. I have attached the log - 
```
phoenix/main/ble.h:23:10: fatal error: cJSON.h: No such file or directory
 #include "cJSON.h"
          ^~~~~~~~~
compilation terminated.
```

shubhamdp, 2023-05-08:
Then, it should be fixed by adding `json` to the `PRIV_REQUIRES` or `REQUIRES` list to the component using it. Please read [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements) for where to add.

And it looks like your main component is using `cJSON` (guessing based on directory structure), in that case, it can be fixed by adding `json` to CMakeLists.txt in your https://github.com/espressif/esp-matter/issues/360#issuecomment-1536774113

arpitsahu00740, 2023-05-09:
After adding all the missing components by adding to the **PRIV_REQUIRES**, I am still getting the following error back.

```
/esp-matter/connectedhomeip/connectedhomeip/src/app/CommandSender.h:27:10: fatal error: type_traits: No such file or directory
 #include <type_traits>
          ^~~~~~~~~~~~~
compilation terminated.
```
Build is working fine, if i don't add any matter related code.
I have added the cpp file containing matter related code and still getting this error!

shubhamdp, 2023-05-10:
Okay, can you please share the complete build logs, because I can not figure out what's going on by just looking those three log lines.

arpitsahu00740, 2023-05-10:
```
[7/13] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj 
/Users/Arpit/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/build/config -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/main -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/demos/ota/common/include -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/portable/os -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/standard/backoffAlgorithm/source/include -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/include -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/portable -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/dependency/coreJSON/source/include -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/dependency/3rdparty/tinycbor/src -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/standard/coreMQTT/source/include -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/standard/coreMQTT/source/interface -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/port/logging -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/port/network_transport -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/port/posix-compat -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/platform/include -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/port/crypto -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/port/ota -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/standard/corePKCS11/source/dependency/3rdparty/pkcs11 -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/standard/corePKCS11/source/include -I/Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/components/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/standard/corePKCS11/source/dependency/3rdparty/mbedtls_utils -I/Users/Arpit/esp/esp-idf/components/newlib/platform_include -I/Users/Arpit/esp/esp-idf/components/freertos/include -I/Users/Arpit/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/Users/Arpit/esp/esp-idf/components/freertos/port/riscv/include -I/Users/Arpit/esp/esp-idf/components/freertos/include/esp_additions -I/Users/Arpit/esp/esp-idf/components/esp_hw_support/include -I/Users/Arpit/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/Arpit/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/Users/Arpit/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -I/Users/Arpit/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -I/Users/Arpit/esp/esp-idf/components/heap/include -I/Users/Arpit/esp/esp-idf/components/log/include -I/Users/Arpit/esp/esp-idf/components/lwip/include/apps -I/Users/Arpit/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/Arpit/esp/esp-idf/components/lwip/lwip/src/include -I/Users/Arpit/esp/esp-idf/components/lwip/port/esp32/include -I/Users/Arpit/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/Arpit/esp/esp-idf/components/soc/include -I/Users/Arpit/esp/esp-idf/components/soc/esp32c3/. -I/Users/Arpit/esp/esp-idf/components/soc/esp32c3/include -I/Users/Arpit/esp/esp-idf/components/hal/esp32c3/include -I/Users/Arpit/esp/esp-idf/components/hal/include -I/Users/Arpit/esp/esp-idf/components/hal/platform_port/include -I/Users/Arpit/esp/esp-idf/components/esp_rom/include -I/Users/Arpit/esp/esp-idf/components/esp_rom/include/esp32c3 -I/Users/Arpit/esp/esp-idf/components/esp_rom/esp32c3 -I/Users/Arpit/esp/esp-idf/components/esp_common/include -I/Users/Arpit/esp/esp-idf/components/esp_system/include -I/Users/Arpit/esp/esp-idf/components/esp_system/port/soc -I/Users/Arpit/esp/esp-idf/components/esp_system/port/include/riscv -I/Users/Arpit/esp/esp-idf/components/esp_system/port/public_compat -I/Users/Arpit/esp/esp-idf/components/riscv/include -I/Users/Arpit/esp/esp-idf/components/driver/include -I/Users/Arpit/esp/esp-idf/components/driver/esp32c3/include -I/Users/Arpit/esp/esp-idf/components/esp_pm/include -I/Users/Arpit/esp/esp-idf/components/esp_ringbuf/includ [... cut ...]
                 from /Users/Arpit/Documents/Atomberg/esp-matter/components/esp_matter/esp_matter.h:22,
                 from /Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/main/matter.cpp:4,
                 from /Users/Arpit/Documents/Atomberg/esp-matter/examples/phoenix/main/app_main.c:22:
/Users/Arpit/Documents/Atomberg/esp-matter/connectedhomeip/connectedhomeip/src/app/CommandSender.h:27:10: fatal error: type_traits: No such file or directory
 #include <type_traits>
          ^~~~~~~~~~~~~
compilation terminated.
```

shubhamdp, 2023-05-10:
Are you including `matter.cpp` in your `app_main.c`? If so, can you create additional header file(or use existing) and add the API that you want to call in `app_main.c`.

Also, check [these](https://github.com/shubhamdp/esp-idf/commit/04075d363539a19a33bf9feea0d8fcd473361240) changes, I tried doing the same thing.

I have `matter_main.cpp` and `matter_main.h` and I am calling `matter_main()` API from `hello_world_main.c` and I could compile fine.

Also, try comparing the changes, that I have done in above commit with what you have done, could be helpful in your transition from non-matter to matter transition.

Also, keep the includes minimal, include only those headers that are really required.

arpitsahu00740, 2023-05-10:
```
Hello world!
E (819) esp_matter_core: Failed to initialize CHIP stack
E (819) esp_matter_core: Error initializing matter

assert failed: _prvTaskExitError port.c:126 (uxCriticalNesting == ~0UL)
Stack dump detected
Core  0 register dump:
MEPC    : 0x40380850  RA      : 0x4038aa06  SP      : 0x3fcb84d0  GP      : 0x3fc93e00  
0x40380850: panic_abort at /Users/Arpit/esp/esp-idf/components/esp_system/panic.c:408

0x4038aa06: __ubsan_include at /Users/Arpit/esp/esp-idf/components/esp_system/ubsan.c:294

TP      : 0x3fc753f4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fcb8637  S1      : 0x00000080  A0      : 0x3fcb8528  A1      : 0x3fc96241  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fca9000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcb8528  S3      : 0x00000001  
S4      : 0x3fcb8528  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

/Library/Developer/CommandLineTools/usr/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/Arpit/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb-3.10.xX5bWmo
panic_abort (details=details@entry=0x3fcb8528 "assert failed: _prvTaskExitError port.c:126 (uxCriticalNesting == ~0UL)") at /Users/Arpit/esp/esp-idf/components/esp_system/panic.c:408
408	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x3fcb8528 "assert failed: _prvTaskExitError port.c:126 (uxCriticalNesting == ~0UL)") at /Users/Arpit/esp/esp-idf/components/esp_system/panic.c:408
#1  0x4038aa06 in esp_system_abort (details=details@entry=0x3fcb8528 "assert failed: _prvTaskExitError port.c:126 (uxCriticalNesting == ~0UL)") at /Users/Arpit/esp/esp-idf/components/esp_system/esp_system.c:137
#2  0x40391010 in __assert_func (file=<optimized out>, file@entry=0x3c138734 "", line=line@entry=126, func=<optimized out>, func@entry=0x3c138674 <__func__.5360> "", expr=<optimized out>, expr@entry=0x3c1386cc "") at /Users/Arpit/esp/esp-idf/components/newlib/assert.c:85
#3  0x4038dc32 in _prvTaskExitError () at /Users/Arpit/esp/esp-idf/components/freertos/port/riscv/port.c:126
#4  prvTaskExitError () at /Users/Arpit/esp/esp-idf/components/freertos/port/riscv/port.c:140
Backtrace stopped: frame did not save the PC
ELF file SHA256: d447b70160d6cfa0

```

shubhamdp, 2023-05-11:
@arpitsahu00740 I had tested hello_world changes before pushing.

I used the below revisions
- esp-idf: v5.0.1
- esp-matter: 731cc4acb45b82fb05307454f7863f3c6afe0bf0 and 35242dd43e0b0dd31086bbb756c27c68df368f9d

You can try that, If you are using any other revisions share them here.

arpitsahu00740, 2023-05-16:
Any fixed for it?

shubhamdp, 2023-05-25:
@arpitsahu00740 I am unable to reproduce the problem. If you have a minimal reproducible patch/code, please attach it

dhrishi, 2023-05-31:
@arpitsahu00740 Closing this issue as this error is resolved

closed 2023-05-31 by dhrishi

crealodi, 2023-06-12:
Hi there, I had the same problems but could solve them with the changes provided by @shubhamdp 
Unfortunately, with this configuration, no components in the projects components directory are found. 
What do I have to do, so that the build system finds these components once again?
The build process fails with 
`undefined reference to 'function_that_works()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.`
The function_that_works() is in a bare minimum component created with the "ESP-IDF Create new ESP-IDF Component" of VS Code - and, as the name implies - works, before adding the code that makes matter run. 

Thank you for your inputs and all the best,

crealodi