espressif/esp-matter #361: Blank project

glennmullerar, 2023-05-01:
Hello, 
I am looking for a blank esp-matter project. Have you created one?

shubhamdp, 2023-05-02:
@glennmullerar What do you mean by blank project?

You can take a look at any example and specifically below lines, (these are from [generic_switch](https://github.com/espressif/esp-matter/blob/1b95fcf66793c2ead7df933d8d80eefd7f8f6b95/examples/generic_switch)) https://github.com/espressif/esp-matter/blob/1b95fcf66793c2ead7df933d8d80eefd7f8f6b95/examples/generic_switch/main/app_main.cpp#L96-L100

First two lines creates a node and endpoint 0 i.e. root endpoint and last two lines creates another endpoint creating generic switch device type.

Also, please check the [Developing your Product](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#developing-your-product) section in programming guide, it's a step by step guide for writing an example.

glennmullerar, 2023-05-03:
Okay, thanks, 

I have another question if I want to rename the example (generic_switch -> temperature_sensor) do I have to modify some files?

shubhamdp, 2023-05-03:
1. Modify data model in `main/app_main.cpp` as per your requirement and driver code in `main/app_driver.cpp`.
2. Modify `CMakeLists.txt` to change the project name. [ref](https://github.com/espressif/esp-matter/blob/1b95fcf66793c2ead7df933d8d80eefd7f8f6b95/examples/generic_switch/CMakeLists.txt#L45)

dhrishi, 2023-05-06:
Closing this as the original question is answered

closed 2023-05-06 by dhrishi