espressif/esp-matter #362: Can't pair matter device based on ESP32-S3 with W5500 Ethernet IC

mgoeller, 2023-05-01:
As the title suggests, I have developed a matter device based on ESP32-S3 with an W5500 Ethernet IC. Using ESP-IDF 5.0.1  I can get an Ethernet connection up and running however pairing with Apple and Google Home App is not working (either the pairing process fails or I get no response right after). When I try the same board with WiFi, everything is working as expected.

Please advise?

PSONALl, 2023-05-02:
@mgoeller can you share device side logs for both the cases you mentioned when pairing fails or the device sends no response? 
currently, there are issues in ethernet commissioning with v1.1.0.0 which are fixed in the [upstream PR](https://github.com/project-chip/connectedhomeip/pull/25876/files). and there is also a corresponding fix required in esp-matter side as mentioned in (https://github.com/espressif/esp-matter/issues/271#issuecomment-1506454360), 
We will update the connectedhomeip submodule of esp-matter soon with this change

mgoeller, 2023-05-17:
@PSONALl you can find the device logs [here](https://justpaste.it/6ngag). Right after boot the W5500 Ethernet driver is successfully initialized. After an IP address is obtained I start matter. Therefore, I had to comment out `esp_event_loop_create_default()` in `esp_matter_core.cpp`. I also adapted `ConnectivityManagerImpl_Ethernet.cpp` as follows:

```
CHIP_ERROR ConnectivityManagerImpl::InitEthernet()
{
    return CHIP_NO_ERROR;
}

void ConnectivityManagerImpl::OnEthernetPlatformEvent(const ChipDeviceEvent * event)
{
}
```

A bridge with an onoff light is initialized. In the device log you can see that I manage to successfully pair with the Google Home app. But right after that the lightbulb is offline. When I open the bridge settings in the Google Home app, I am able to pull some device information about it. 

And, of course, I applied the two patches as you did. I am thankful for any ideas!

_Originally posted by @jzwyssig in https://github.com/espressif/esp-matter/issues/271#issuecomment-1531758540_

I applied the patches of this thread, and am running into the same problem as @jzwyssig. Where exactly does that problem come from?

PSONALl, 2023-05-29:
@mgoeller The fix for ethernet commissioning is merged now can pull the latest code and try again?

mgoeller, 2023-06-06:
Thank you @PSONALl, I can successfully commission now via Ethernet and / or WiFi. However the bridged Matter endpoints immediately go offline after successful pairing with Google Home when connected via Ethernet.

As stated in this thread: https://github.com/espressif/esp-matter/issues/271

So there must be still something wrong on your side of implementing Matter?

mgoeller, 2023-06-13:
I tried with a single (non-bridged) accessory and there it works. So it must be an error on your side. Please check and fix asap. We want to go into production.

PSONALl, 2023-06-14:
@mgoeller Can you specify which bridge application you are using?
If possible please share a patch to reproduce this issue

mgoeller, 2023-06-14:
I just compiled a matter firmware with esp-idf 5.0.2 and the latest esp-matter for a single lightbulb, no bridge or any accessory. Using ESP32-S3 as hardware and a W5500 ethernet chip as indicated before.

I can pair with Google Home, however the device goes immediately offline after successful pairing. I can from there still share to Apple Home (as another fabric) and everything works fine there.

We really need to get our device into production and certification. Please fix this issue asap.

PSONALl, 2023-06-15:
@mgoeller I think this is the issue with Google Home, I have reported it to Google's Issue Tracker
I am also looking into it 
I will notify you whenever the next update is available

dhrishi, 2023-07-08:
@mgoeller This should ideally fix the issue https://github.com/project-chip/connectedhomeip/pull/27808/files

We will soon update the connectedhomeip submodule in esp-matter with this. If you want, you can verify this fix by adding the changes locally

jzwyssig, 2023-07-10:
@dhrishi @PSONALl I can confirm Ethernet is working now with google home. As I use the W5500 chipset I already provided the necessary callbacks. Therefore, I cannot verify all of your changes. But the magic line that makes everything work is `ESP_ERROR_CHECK(esp_netif_create_ip6_linklocal(eth_netif));` in the `ETHERNET_EVENT_CONNECTED` event.
Thanks!

dhrishi, 2023-07-11:
@jzwyssig Nice. That is good to know. I will keep this issue open until the connectedhomeip submodule (with the fix) is updated

dhrishi, 2023-08-28:
Closing this as the fix is present in the submodule on the main branch

closed 2023-08-28 by dhrishi

JoseAntonioMG, 2024-08-16:
Hi @jzwyssig , could you share the full code for the W5500 ethernet implementation in matter?

Thanks.

DanieleB7, 2024-10-17:
> Hi @jzwyssig , could you share the full code for the W5500 ethernet implementation in matter?
> 
> Thanks.

Would you kindly share code for this @jzwyssig if you are allowed to. thanks