espressif/esp-matter #363: Reducing footprint of ESP-Matter

marcelgoh, 2023-05-02:
Hi,

I'm trying to gauge the size/memory consumption of Matter in order to see if we can incorporate it into our project. I did `idf.py size-components` on the light example and got this:
[image]
As I understand, the `libbtdm_app.a` and `libnet80211.a` components are related to bluetooth and thread respectively. If I do not need this for my use-case, is it possible to remove these components by disabling the respective features? I tried `CONFIG_BT_ENABLE=n` in `sdkconfig.defaults` but that didn't change the binary size or the presence of that `.a` file.

It is worrying to me that even an app as simple as the light example already uses a significant portion of DRAM/IRAM. If you had any other suggestions as to how one might reduce the size / RAM usage of ESP-Matter, I would appreciate hearing them as well! Thanks in advance.

jonsmirl, 2023-05-02:
You likely need to use a module with PSRAM on it.  I can run a single button device on a C3 without PSRAM. But the eight button, keypad version of the same device needs PSRAM. Once you add PSRAM you should have plenty of memory. Just make sure to set allocation strategy to use PSRAM in the various components.

igrr, 2023-05-02:
> I tried CONFIG_BT_ENABLE=n in sdkconfig.defaults but that didn't change the binary size or the presence of that .a file.

Please note that IDF build system only uses `sdkconfig.defaults` file when no `sdkconfig` file exists. If you want to make some configuration changes, you can either a) edit `sdkconfig` or b) edit `sdkconfig.defaults` and delete `sdkconfig`, so that it gets re-generated on next build.

However, given that network provisioning functionality depends on Bluetooth, I doubt that disabling Bluetooth will work without further changes.

marcelgoh, 2023-05-02:
Thanks for the super quick replies! @jonsmirl sorry for the amateur question, but PSRAM this available for my ESP32-WROVER-E and what is the best way to enable it/ensure that it's enabled? I searched `PSRAM` in sdkconfig and only found `CONFIG_SPIRAM`, which I'm not sure is the right setting to set.

@igrr Ah, sorry for the dumb mistake! I deleted `sdkconfig` before rebuilding and that whole `.a` file is now gone, reducing the binary size. And I did some basic tests and everything seems to still work the same as before over WiFi? Would you know the corresponding setting to get rid of `libnet80211.a`, so I can test if that still works too?

marcelgoh, 2023-05-02:
Ah, thanks, it appears mine is already enabled, though perhaps not completely set up right:

```
I (585) quad_psram: This chip is ESP32-D0WD
I (585) esp_psram: Found 8MB PSRAM device
I (585) esp_psram: Speed: 40MHz
I (589) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (596) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
```

jonsmirl, 2023-05-02:
Now you need to turn it on in the various components.

[image]

dhrishi, 2023-05-07:
@marcelgoh Please close this issue if you are able to use the PSRAM and move things into it

marcelgoh, 2023-05-08:
Hi, sorry but I'm still unable to use all 8MB of my PSRAM, even after setting Flash size to 8MB in the Serial flasher options of `idf.py menuconfig`. Is the 4MB the best that is possible, despite the fact that it says "Found 8MB PSRAM device"?

PSONALl, 2023-05-09:
@marcelgoh Have you tried setting Flash size to 4MB and setting Octal SPIRAM, It worked for me in case of esp32s3
Which kit are you using?

marcelgoh, 2023-05-09:
@PSONALl where is the setting for Octal SPIRAM in `menuconfig`? I'm not seeing it in Serial flasher config

jonsmirl, 2023-05-09:
Using 8MB is not so simple on an older ESP32. Here is how to do it...
https://github.com/espressif/esp-idf/tree/master/examples/system/himem

Much easier to just switch onto a new ESP32S3 module where using the 8MB is straight forward.

marcelgoh, 2023-05-09:
Ah, okay yes I'm using the older ESP32. Thanks for linking that example though!

closed 2023-05-09 by marcelgoh