espressif/esp-matter #364: mDNS problem

jonsmirl, 2023-05-02:
What is going on in this attached log?
[mDNS.log](https://github.com/espressif/esp-matter/files/11376900/mDNS.log)

There are two commissions going on. I am using Android.
1) First commission is being done by Google. They are doing it so that they can set the wifi ssid/pw into the device.
2) Second commission is my app. This second commission is what triggered the first one since the Google lib knew it had to insert the SSID/pw into the device.

I am making my own certificates so that could be part of the problem.

It got to this point and it died:
~~~
I (64213) chip[FP]: NOC chain validation successful
I (64213) chip[FP]: Added new fabric at index: 0x2I (64213) chip[FP]: Assigned compressed fabric ID: 0x2A3116CDB0CC76FE, node ID: 0x0000000000000002
I (64213) chip[TS]: Last Known Good Time: 2023-05-02T10:15:03
I (64223) chip[TS]: New proposed Last Known Good Time: 2021-06-10T00:00:00
I (64223) chip[TS]: Retaining current Last Known Good TimeI (64243) chip[ZCL]: OpCreds: ACL entry created for Fabric index 0x2 CASE Admin Subject 0xDEDEDEDE0000001D
I (64243) chip[DIS]: Advertise operational node 4284C2F5A80B5F7C-8E11AC440B94CF84
I (64243) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 4284C2F5A80B5F7C-8E11AC440B94CF84.
I (64253) chip[DIS]: mDNS service published: _matter._tcp
I (64253) chip[DIS]: Advertise operational node 2A3116CDB0CC76FE-0000000000000002
I (64253) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 2A3116CDB0CC76FE-0000000000000002.
I (64273) chip[DIS]: mDNS service published: _matter._tcp
I (64273) chip[ZCL]: OpCreds: successfully created fabric index 0x2 via AddNOC
I (64273) chip[EM]: <<< [E:29773r S:3097 M:104859328 (Ack:187132537)] (S) Msg TX to 2:FFFFFFFB00000000 [76FE] --- Type 0001:09 (IM:InvokeCommandResponse)
I (64273) chip[IN]: (S) Sending msg 104859328 on secure session with LSID: 3097
E (64283) chip[DL]: Long dispatch time: 325 ms, for event type 3
I (64283) chip[EM]: >>> [E:29772r S:3097 M:187132538 (Ack:104859326)] (S) Msg RX from 2:FFFFFFFB00000000 [76FE] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (64333) chip[EM]: >>> [E:29773r S:3097 M:187132539 (Ack:104859328)] (S) Msg RX from 2:FFFFFFFB00000000 [76FE] --- Type 0000:10 (SecureChannel:StandaloneAck)

>>> Why is only one mDNS visible? 0002 is missing
~~~
It died because the second mDNS did not become visible
~~~
2023-05-02 14:36:04.496 10675-11601 EM                      com.lowpan.m2                        D  Flushed pending ack for MessageCounter:104859328 on exchange 29773i
2023-05-02 14:36:04.676 10675-11601 DIS                     com.lowpan.m2                        D  Checking node lookup status after 200 ms
2023-05-02 14:36:04.725 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  Service discovery started. regType: _matter._tcp
2023-05-02 14:36:04.727 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  onServiceFound: found service not a target for resolution, ignoring name: 4284C2F5A80B5F7C-8E11AC440B94CF84, type: _matter._tcp., host: null, port: 0, network: 108, txtRecord: 
2023-05-02 14:36:04.727 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  onServiceFound: found service not a target for resolution, ignoring name: 2A3116CDB0CC76FE-0000000000000002, type: _matter._tcp., host: null, port: 0, network: 108, txtRecord: 
2023-05-02 14:36:04.821 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  onServiceFound: found service not a target for resolution, ignoring name: C7944E937AA07DD5-0083161AE5399201, type: _matter._tcp., host: null, port: 0, network: 108, txtRecord: 
2023-05-02 14:36:05.028 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  onServiceFound: found service not a target for resolution, ignoring name: 950545243CFE2D15-00000000F1077CC7, type: _matter._tcp., host: null, port: 0, network: 108, txtRecord: 
2023-05-02 14:36:05.029 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  onServiceFound: found service not a target for resolution, ignoring name: C7944E937AA07DD5-00867AF748BEBA01, type: _matter._tcp., host: null, port: 0, network: 108, txtRecord: 
2023-05-02 14:36:05.033 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  onServiceFound: found service not a target for resolution, ignoring name: C7944E937AA07DD5-008462EE7296AD71, type: _matter._tcp., host: null, port: 0, network: 108, txtRecord: 
2023-05-02 14:36:05.034 10675-11564 NsdService...ndResolver com.lowpan.m2                        D  onServiceFound: found service not a target for resolution, ignoring name: 9CC0373667C4F408-204B50A898A38F65, type: _matter._tcp., host: null, port: 0, network: 108, txtRecord: 
2023-05-02 14:36:09.494 10675-11607 NsdService...ndResolver com.lowpan.m2                        D  Service discovery timed out after 5000 ms
2023-05-02 14:36:09.497 10675-11564 NsdService...ndResolver com.lowpan.m2                        I  Discovery stopped: _matter._tcp
2023-05-02 14:36:34.523 10675-10675 NsdManager...ceResolver com.lowpan.m2                        D  resolve: Timing out
2023-05-02 14:36:49.518 10675-11601 DIS                     com.lowpan.m2                        D  Checking node lookup status after 45042 ms
2023-05-02 14:36:49.518 10675-11601 DIS                     com.lowpan.m2                        E  OperationalSessionSetup[1:0000000000000002]: operational discovery failed: ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
2023-05-02 14:36:49.518 10675-11601 CTL                     com.lowpan.m2                        D  Device connection failed. Error ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
2023-05-02 14:36:49.520 10675-11601 BaseCompletionListener  com.lowpan.m2                        D  onCommissioningStatusUpdate nodeId [2]  stage [FindOperational]  errorCode [50]
2023-05-02 14:36:49.520 10675-11601 CTL                     com.lowpan.m2                        D  Error on commissioning step 'FindOperational': '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout'
2023-05-02 14:36:49.520 10675-11601 CTL                     com.lowpan.m2                        E  Failed to perform commissioning step 18
2023-05-02 14:36:49.520 10675-11601 CTL                     com.lowpan.m2                        D  Going from commissioning step 'FindOperational' with lastErr = '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
2023-05-02 14:36:49.520 10675-11601 CTL                     com.lowpan.m2                        D  Performing next commissioning step 'Cleanup' with completion status = '../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout'
2023-05-02 14:36:49.522 10675-11601 BaseCompletionListener  com.lowpan.m2                        D  onCommissioningStatusUpdate nodeId [2]  stage [Cleanup]  errorCode [0]
2023-05-02 14:36:49.522 10675-11601 CTL                     com.lowpan.m2                        D  Successfully finished commissioning step 'Cleanup'
2023-05-02 14:36:49.523 10675-11601 BaseCompletionListener  com.lowpan.m2                        D  onCommissioningComplete(): nodeId [2] errorCode [50]
2023-05-02 14:36:49.527 10675-10675 AppCommiss...gRequested com.lowpan.m2                        E  onCommissioningRequested() failed
                                                                                                    java.lang.IllegalStateException: Commissioning failed with error code [50]
                                                                                                    	at com.lowpan.m2.chip.ChipClient$awaitCommissionDevice$2$1.onCommissioningComplete(ChipClient.kt:293)
                                                                                                    	at chip.devicecontroller.ChipDeviceController.onCommissioningComplete(ChipDeviceController.java:335)
2023-05-02 14:36:49.535 10675-10675 AppCommiss...gRequested com.lowpan.m2                        D  Commissioning: commissioningServiceDelegate.sendCommissioningError() succeeded
2023-05-02 15:03:52.277 10675-14306 TrafficStats            com.lowpan.m2                        D  tagSocket(130) with statsTag=0xffffffff, statsUid=-1
~~~

jonsmirl, 2023-05-02:
I added the CAT tag and it made no difference.

jonsmirl, 2023-05-02:
The problem is that Android and ESP32 are disagreeing on the compressed fabric ID.

jonsmirl, 2023-05-02:
I've figured out that I need to build a NOCChainIssuer class.

dhrishi, 2023-05-07:
Closing this here

closed 2023-05-07 by dhrishi