espressif/esp-matter #366: creating an Sleepy End Device (SED)

glennmullerar, 2023-05-03:
Hello,

Hello,

I would like to create a SED. In the Matter specification it is indicated that we have to change the value of the field 
```SLEEPY_IDLE_INTERVAL```  or ```SLEEPY_ACTIVE_INTERVAL``` depending on the mode we want to use. However, I couldn't find anything in the ```sdkconfig``` and in the esp-matter files (```esp-matter/components```). Is it possible to do this or is it a feature that will come later. 

Most probably I did not look in the right place. 

Thanks in advance for your answer.

chshu, 2023-05-04:
@glennmullerar It's configurable by `CHIP_CONFIG_MRP_LOCAL_ACTIVE_RETRY_INTERVAL` and `CHIP_CONFIG_MRP_LOCAL_IDLE_RETRY_INTERVAL`, the detailed description about these two parameters are here:

https://github.com/project-chip/connectedhomeip/blob/master/src/messaging/ReliableMessageProtocolConfig.h#L52

Could you modify the default values and have a test, let us know if it works for you. We will make it configurable via IDF menuconfig.

glennmullerar, 2023-05-05:
Thank you for your answer. 

I see that by changing the value the number of steps at the time of the build changes. 

However, is there a chip-tool or other command that would allow to know if it is well known as SED by the other devices in the network?

chshu, 2023-05-26:
It's could be achieved by the new ICD (Intermittently Connected Devices) feature, and there is a `clientmonitoring` cluster cmd can be used by chip-tool.

But the feature is currently not supported in esp-matter yet, since the feature is still under developing, we will add the support in short future. Could you let us know which SoC are you using, Wi-Fi or Thread?

marciogranzotto, 2023-05-30:
I'm trying to make a SED with ESP32-H2 using thread. Is that not supported yet as well? @chshu

chshu, 2023-05-30:
@marciogranzotto The H2 sleep mode is not supported yet (IDF and esp-matter), will keep you posted.

Gonta01, 2023-06-01:
What about ESP32C6? Is it possible to create Thread Sleep End Device on ESP32C6 basis?

marciogranzotto, 2023-06-16:
@chshu is it posible with the C6?

chshu, 2023-06-25:
> The aim is to reduce the power consumption of a WiFi device.
@glennmullerar If it's a Wi-Fi device, you don't need to wait for the ICD feature, you can just enable the Wi-Fi power save mode ([example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/power_save)).

chshu, 2023-06-25:
@Gonta01 @marciogranzotto To enable the Thread Sleepy End Device on H2 and C6 for Matter application, the following two tasks need to be done:
- Light Sleep support for Thread in esp-idf
- ICD feature support in esp-matter

We target to have the two features ready by the end of July.

caipiblack, 2023-07-23:
I’m’also looking for this feature in WiFi and Thread.

I want to see the average power consumption we can expect with esp32-c6 in wifi and thread as sleepy end device.

The objective is to compare nRF52 vs ESP32-C6.

I prefer the esp ecosystem but high power consumption will not be acceptable for my application.

marciogranzotto, 2023-07-28:
@chshu I see that the Light Sleep support for Thread in esp-idf is already implemented. Is there any news about the ICD/SED feature in esp-matter?

Looking forward for it so I can get my product working!

chshu, 2023-07-31:
@LinJiaDeng @caipiblack @marciogranzotto The Thread light sleep on H2 and C6 are all supported in ESP-IDF now, you may first refer to the [H2 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-h2_datasheet_en.pdf) for some power consumption data, and evaluate the H2 / C6 power consumption using this example: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_sleepy_device.

We are actively adding the ICD support to esp-matter, will posted here once it's ready.

marciogranzotto, 2023-09-21:
@chshu any news?
If there's anything I can help with on the implementation, I would be happy to

chshu, 2023-09-22:
@marciogranzotto The ICD feature itself has been supported in Matter ESP platform (https://github.com/project-chip/connectedhomeip/pull/28359), some sleep tunning work is still under progress, will update here once it's done (target to mid Oct).

pogadlin, 2023-10-17:
> > The aim is to reduce the power consumption of a WiFi device.
> > @glennmullerar If it's a Wi-Fi device, you don't need to wait for the ICD feature, you can just enable the Wi-Fi power save mode ([example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/power_save)).

Hi @chshu,
I want to check power consumption on ESP32-C6 for Wi-Fi, as you mentioned how to enable Wi-Fi power save mode?

chshu, 2023-10-19:
@glennmullerar @marciogranzotto The Matter ICD support on ESP32-H2 is available in esp-matter now, please check the example: https://github.com/espressif/esp-matter/tree/main/examples/sleepy_device

marciogranzotto, 2023-10-19:
@chshu just for the H2? what about C6?

flashspys, 2023-10-19:
The example does not compile for the C6 out of the box. I was able to make it compiling after I've removed the espressif/mdns and /insights packages and a few other minor h2->c6 renaming. For HomeKit commissioning I had to comment out the BLE Sleep changes.
Even though, I'm not 100% sure if the example works as expected. I have a cheap USB power meter that says that my C6 still needs 40mA / 200mW which should be way too high. Light sleep on ESP should take 0.8 mA / 4mW.

Can you tell me how to check which mode the ESP32 is currently in and if this behavior is expected?

chshu, 2023-10-20:
@marciogranzotto @flashspys Currently the example only supports ESP32-H2 Thread Sleepy device.

C6 Thread is pending on a sleep issue, will be supported once the issue is fixed in IDF.
C6 Wi-Fi support will be added later.

marciogranzotto, 2023-11-23:
@chshu any news on the C6 sleep issue? Btw, will this be on which version on the idf? I'm currently developing against `release/v5.1`

marciogranzotto, 2023-12-04:
@chshu any news on the C6 sleep issue? Btw, will this be on which version of the IDF? I'm currently developing against release/v5.1

chshu, 2023-12-05:
> @chshu any news on the C6 sleep issue? Btw, will this be on which version of the IDF? I'm currently developing against release/v5.1

The C6 sleep issue has been fixed in IDF, but some compiling issues (as mentioned in https://github.com/espressif/esp-matter/issues/366#issuecomment-1771792028) depend on some changes in other repos, the fixes are under merging.

We will prepare a patch for your testing soon, you may also evaluate the Matter sleep feature on H2 first.

itatboy, 2023-12-11:
Hi, When can C2/C3 support deep sleep?

bskimaaa, 2023-12-15:
[Questions about features added to ESP-IDF v5.1]
Does this mean that the wifi sleep (SED, ICD) function is supported in the ESP32C6 Matter project?
Does this mean that the thread sleep function is supported in the ESP32C6 Matter project?
Does the ESP32C3 Matter project also support wifi sleep (SED, ICD) function?

Pips801, 2023-12-17:
> [Questions about features added to ESP-IDF v5.1]
> 
> Does this mean that the WIFI sleep (SED, ICD) function is supported in the ESP32C6 Matter project?
> 
I can not confirm WiFi sleepy, I will have to try later, but I can confirm that Power Management (PM) and ICD server can be enabled on the c6.
> Does this mean that the Thread sleep function is supported in the ESP32C6 Matter project?
> 
Yes. After following the batches advised from Espressif, modifying the upstream project chip cmake and idf component files, plus a little bit of tweaking for my project to reenable secure-cert-manager, I was able to make a Matter over Thread Sleepy generic_switch endpoint on the C6 - https://github.com/espressif/esp-matter/assets/5651603/8d2431ba-2cc5-4e57-b3b3-dabd95e6a89f

> Does the ESP32C3 Matter project also support WIFI sleep (SED, ICD) function?

I cannot confirm, I don't have a C3, but if WiFi sleep works for the C6 I bet it would work for the C3.

marciogranzotto, 2023-12-19:
I also applied the patches, using esp-idf commit `a36d8bc742` (release/v5.2) and esp-matter commit `96dc30b` (release/v1.2).

However my ESP32-C6-DevKitC-1 keeps crashing when running the `sleepy_device` example:

<details>
  <summary>Logs (click to expand)</summary>
  
  ```
  Rebooting...
  ESP-ROM:esp32c6-20220919
  Build:Sep 19 2022
  rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
  SPIWP:0xee
  mode:DIO, clock div:2
  load:0x40875720,len:0x1804
  load:0x4086c410,len:0xde0
  load:0x4086e610,len:0x2df8
  entry 0x4086c41a
  I (25) boot: ESP-IDF v5.2-beta1-226-ga36d8bc742 2nd stage bootloader
  I (26) boot: compile time Dec 19 2023 19:37:56
  I (27) boot: chip revision: v0.0
  I (30) boot.esp32c6: SPI Speed      : 80MHz
  I (34) boot.esp32c6: SPI Mode       : DIO
  I (39) boot.esp32c6: SPI Flash Size : 4MB
  I (44) boot: Enabling RNG early entropy source...
  I (49) boot: Partition Table:
  I (53) boot: ## Label            Usage          Type ST Offset   Length
  I (60) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
  I (68) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
  I (75) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000
  I (82) boot:  3 otadata          OTA data         01 00 0001d000 00002000
  I (90) boot:  4 phy_init         RF data          01 01 0001f000 00001000
  I (97) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
  I (105) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
  I (112) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
  I (120) boot: End of partition table
  I (125) esp_image: segment 0: paddr=00020020 vaddr=42110020 size=38c68h (232552) map
  I (226) esp_image: segment 1: paddr=00058c90 vaddr=40800000 size=07388h ( 29576) load
  I (241) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=10399ch (1063324) map
  I (668) esp_image: segment 3: paddr=001639c4 vaddr=40807388 size=0f284h ( 62084) load
  I (698) esp_image: segment 4: paddr=00172c50 vaddr=50000000 size=00048h (    72) load
  I (708) boot: Loaded app from partition at offset 0x20000
  I (709) boot: Disabling RNG early entropy source...
  I (720) cpu_start: Unicore app
  W (730) clk: esp_perip_clk_init() has not been implemented yet
  I (736) cpu_start: Pro cpu start user code
  I (736) cpu_start: cpu freq: 160000000 Hz
  I (737) cpu_start: Application information:
  I (739) cpu_start: Project name:     sleepy_device
  I (745) cpu_start: App version:      1.0
  I (749) cpu_start: Compile time:     Dec 19 2023 19:50:01
  I (755) cpu_start: ELF file SHA256:  66248d460...
  I (761) cpu_start: ESP-IDF:          v5.2-beta1-226-ga36d8bc742
  I (767) cpu_start: Min chip rev:     v0.0
  I (772) cpu_start: Max chip rev:     v0.99 
  I (777) cpu_start: Chip rev:         v0.0
  I (782) heap_init: Initializing. RAM available for dynamic allocation:
  I (789) heap_init: At 40828F40 len 000536D0 (333 KiB): RAM
  I (795) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
  I (801) heap_init: At 50000048 len 00003FA0 (15 KiB): RTCRAM
  I (808) spi_flash: detected chip: generic
  I (812) spi_flash: flash io: dio
  W (816) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
  I (830) sleep: Enable automatic switching of GPIO sleep configuration
  I (837) sleep_clock: System Power, Clock and Reset sleep retention initialization
  I (845) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
  I (853) sleep_sys_periph: Interrupt Matrix sleep retention initialization
  I (860) sleep_sys_periph: HP System sleep retention initialization
  I (867) sleep_sys_periph: TEE/APM sleep retention initialization
  I (874) sleep_sys_periph: UART sleep retention initialization
[... 54 lines cut ...]
  A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000099  S3      : 0x40816ea8  
  S4      : 0x40816ea8  S5      : 0x00000001  S6      : 0x4083dc6c  S7      : 0x00000000  
  S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
  T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
  MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
  0x40800001: _vector_table at ??:?

  MHARTID : 0x00000000  

  Backtrace:

  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/marciorodrigues/Olibra/code/matter/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb-3.9.Awd8AYd
  panic_abort (details=details@entry=0x40816ea8 <xIsrStack+1256> "assert failed: osi_assert_wrapper bt.c:263 (0)") at /Users/marciorodrigues/Olibra/code/matter/esp-idf/components/esp_system/panic.c:472
  472         *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
  #0  panic_abort (details=details@entry=0x40816ea8 <xIsrStack+1256> "assert failed: osi_assert_wrapper bt.c:263 (0)") at /Users/marciorodrigues/Olibra/code/matter/esp-idf/components/esp_system/panic.c:472
  #1  0x4080d4e2 in esp_system_abort (details=details@entry=0x40816ea8 <xIsrStack+1256> "assert failed: osi_assert_wrapper bt.c:263 (0)") at /Users/marciorodrigues/Olibra/code/matter/esp-idf/components/esp_system/port/esp_system_chip.c:93
[... cut ...] y=0x421180dc "") at /Users/marciorodrigues/Olibra/code/matter/esp-idf/components/newlib/assert.c:81
  #3  0x40802b8a in osi_assert_wrapper (ln=<optimized out>, fn=<optimized out>, param1=<optimized out>, param2=<optimized out>) at /Users/marciorodrigues/Olibra/code/matter/esp-idf/components/bt/controller/esp32c6/bt.c:263
  #4  0x40804cde in ble_lll_sched_check_remaining_entries ()
  Backtrace stopped: frame did not save the PC
  ELF file SHA256: 66248d460
  ```
</details>

Can you please advise?

marciogranzotto, 2023-12-20:
@Pips801 could you tell me which commits of the esp-idf and esp-matter you use to achieve C6 Thread sleepy device? I would greatly appreciate it!

Pips801, 2023-12-20:
> @Pips801 could you tell me which commits of the esp-idf and esp-matter you use to achieve C6 Thread sleepy device? I would greatly appreciate it!

I will have to go check the build environment, but it is IDF v5.1.1 with the latest matter 1.2 repository for esp-matter.

Can you post your sdkconfig for the sleepy c6? I ended up copying most everything from the Sleepy C2 example code and the C2 sdkcondig. 

I did also encounter some sporadic crashes like yours with the Bluetooth stack. One thing that I changed, that may help, was changing `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` from `y` to `n` in the sdkconfig for the C6. since the panic happens right at a ble-related function in the ble stack.

marciogranzotto, 2023-12-24:
I've tried with both `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` set to `y` and to `n`. I also tested the `sleepy_device` example, and the `light` example with the sleepy_device changes applied to it.

I keep getting the same crash on `ble_lll_sched_check_remaining_entries`, which appears to be on the lower lever bluetooth driver of the ESP32-C6.

I'll try to zip my project later and post it here.

wqx6, 2023-12-28:
> Does this mean that the WIFI sleep (SED, ICD) function is supported in the ESP32C6 Matter project?

No, Wi-Fi ICD is still not supported for our platform. In fact, there is no `SED` for Wi-Fi devices. The Wi-Fi should wake up to keep the connection per 100ms~500ms

> Does this mean that the Thread sleep function is supported in the ESP32C6 Matter project?

Yes, the Thread sleep is supported for ESP32C6 for IDF tag v5.1.2, and you can build the sleepy-device examples for C6 if you have applied the two patches

> Does the ESP32C3 Matter project also support WIFI sleep (SED, ICD) function?

No.

wqx6, 2023-12-28:
> Can you please advise?

@marciogranzotto @Pips801 Sorry I forgot to add the phy configuration CONFIG_ESP_PHY_MAC_BB_PD=y to the C6's sdkconfig file in the patch file, please add that configuration option and build the example based on IDF release/v5.1(commit id 47c89441c16e0933474c8dc7d8ca9b3f75d2a1da).

wqx6, 2024-01-19:
Here is the current picture of C6 with the same default ICD parameters as H2. 
[image]

chshu, 2024-02-22:
Hi All, Both H2 and C6 ICD support is ready, please try out this example: https://github.com/espressif/esp-matter/tree/main/examples/icd_app.

Note that the IDF release/v5.1 branch with the commit id [931eaf7320](https://github.com/espressif/esp-idf/tree/931eaf7320b6c0b9acc9711ba4774f4f1bd3dae7) or later, should be used for C6 sleep.

closed 2024-02-22 by chshu

Kristian8606, 2024-06-22:
@marciogranzotto  Did you manage to reduce esp32 consumption, and what values ​​did you measure?
My measurements are esp32h2 3mA and esp32c6 9mA
this is too much consumption.

caipiblack, 2024-08-13:
Hello,

In icd_app readme (https://github.com/espressif/esp-matter/tree/main/examples/icd_app) it say's:

**Note:** For ESP32-C6, please use ESP-IDF on branch release/v5.1 with the commit id [931eaf7320](https://github.com/espressif/esp-idf/tree/931eaf7320b6c0b9acc9711ba4774f4f1bd3dae7).

So:

1. Why do we have to use this specific commit ?
2. Is it possible to use newer commits ?
3. What is the real status of icd_app ? Working ?

Siddhiq74, 2024-12-18:
How to adjust the 802.15.4 TX Power in  lit-icd-app  ( Matter examples).  I am using a ESP32h2 device as portable switch ( battery backed).  I need to adjust the TX Power to reduce power consumption from the default +20bm to +9dm.

dmartauz, 2025-02-27:
> How to adjust the 802.15.4 TX Power in lit-icd-app ( Matter examples). I am using a ESP32h2 device as portable switch ( battery backed). I need to adjust the TX Power to reduce power consumption from the default +20bm to +9dm.

@Siddhiq74 have you found a solution? Is the TX power fixed or is it regulated dynamically?

Siddhiq74, 2025-02-27:
> [@Siddhiq74](https://github.com/Siddhiq74) have you found a solution? Is the TX power fixed or is it regulated dynamically?

No,  Still trying to get the right solution

dmartauz, 2025-02-27:
@Siddhiq74 I only found this resource:

If developing based on Thread, it is recommended to adjust the transmit power of Thread through the openthread API otPlatRadioSetTransmitPower.

If developing directly based on the 802.15.4 MAC driver, it is recommended to adjust the transmission power of Thread through esp_ieee802154_set_txpower.

https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/performance/modify-tx-power.html

I did not try it yet. What power consumption are you able to achieve with H2?

laptopfreek0, 2025-04-04:
@Siddhiq74  @dmartauz Have either of you been able to reduce the TX power. I have attempted both `esp_ieee802154_set_txpower(-9);`
and
`otPlatRadioSetTransmitPower(NULL, 9);`

Neither function has resulted in any less power consumption. FYI I was able to achieve as low as 100 uA while sleeping. Any further attempt to reduce power has been thwarted.