espressif/esp-matter #367: Commissioning hangs without error

alex-moran-deako, 2023-05-04:
I've copied the reference example light in my own project.  Matter starts up just fine, and my commissioning device shows up as a "Matter Accessory". After establishing a secured channel, commissioning halts without warning and eventually both devices times out.

Attached are the commissioning logs and the sdkconfig used in the project.

[sdkconfig.txt](https://github.com/espressif/esp-matter/files/11393056/sdkconfig.txt)
[commission_fail_log.txt](https://github.com/espressif/esp-matter/files/11393057/commission_fail_log.txt)

I am able to repro this 100% of the time. I am using commit bafc378e68f1c32f72329ebcc986f2cfa88d67ed of esp-matter.

shubhamdp, 2023-05-04:
Do you see the same failure with `light`?

shubhamdp, 2023-05-04:
Also, can you please share the IDF version/commit being used here and how are you commissioning the device as well. If you are using the chip-tool for commissioning, can you please share those logs as well.

alex-moran-deako, 2023-05-04:
I unfortunately do not see the same failure with examples/light with the same commissioning device. Comparing the two sdkconfigs, I'm having trouble seeing what the difference would be. 
[light_example_sdkconfig.txt](https://github.com/espressif/esp-matter/files/11399938/light_example_sdkconfig.txt)

I'd be one thing if my application was failing with errors, but the fact the commissioner stops communicating all together makes me think that my application is somehow not responding correctly during the initial handshaking. 

I am commissioning with an iOS device / HomePod.

I am using v4.4.3 of IDF at 6407ecb3f.

I am working on using the chip-tool for commissioning to hopefully get more insight on what the commissioner is doing.

alex-moran-deako, 2023-05-04:
I tried to get the chip tool working today without luck.

I am issues even with the light example finding the device. I see the BLE scan showing lots of MACs with "this is not a matter device" and eventually 

```
[1683240400.577652][78417:78419] CHIP:BLE: BLE scan error: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/bluez/ChipDeviceScanner.cpp:154: CHIP Error 0x00000032: Timeout
```

The device is discoverable by my apple device, and when its running the sample application commissions with it.

PSONALl, 2023-05-05:
@alex-moran-deako From the device logs you shared your device is advertising these parameters
```
Advertise commission parameter vendorID=65521 productID=32768 discriminator=3841/15 cm=1
```
Can you check If the QR Code you are scanning for has a discriminator 3841 ?
And also command you are using from commissioning has the correct discriminator?

PSONALl, 2023-05-05:
You can scan this QRCode for commissioning using Apple Home:

[image]

sisodiakaran, 2023-05-06:
@PSONALl How you generated this code? I mean the content of QR?

dhrishi, 2023-05-07:
@sisodiakaran See this as an example https://github.com/espressif/esp-matter/issues/372#issuecomment-1536261601 

Please close this if your issue is resolved

alex-moran-deako, 2023-05-08:
I think the issues lies with the idea that the commissioner is not talking to cluster 0x30. In the successful light example, I can see the commissioner interacting with 0x30. Whereas with my application with the ported sample code, the commissioner is talking to cluster 0x31. Why is this?

PSONALl, 2023-05-09:
> When I use the QR code you generated, the commissioning does not get past the PASE_Pake Key exchange when establishing the secure channel.

This is because while generating a QRcode, you need to replace the values you are using, in this case, you need to use the passcode 20202020 like this:

```
./out/debug/chip-tool payload generate-qrcode --existing-payload MT:Y.K9042C00KA0648G00 --discriminator 3841 --setup-pin-code 20202020 --vendor-id 65521 --product-id 32768 --commissioning-mode 0 --rendezvous 2
```

You will get a QR Code payload and manual pairing code as a output of this command and then QRCode can be generated here from the payload: https://project-chip.github.io/connectedhomeip/qrcode.html

shubhamdp, 2023-05-09:
> When I use my iOS device to commission the device, i got to "More options" when adding the accessory and I use the passcode "20202020". If I see a different passcode, say 12345678, I see that my commissioning fails during the secure channel establishment. The fact that it gets past this phase has alluded to me that the authentication has been successful.

Setup Code and passcode is different. You have entered the passcode here and it expects the "Setup Code".

To generate the setup code please try the following command and enter the generated setup code. 
```
chip-tool payload generate-manualcode --discriminator 3841 --setup-pin-code 20202020 --commissioning-mode 0
```

alex-moran-deako, 2023-05-09:
Doesn't the fact that the secure channel gets established and is past the key exchange mean that it has nothing to do with the key? Using the same mfg binary and method of commissioning with the esp-matter light example, using the passcode 20202020 in the Home application successfully commissions the device. All that being said, I have generated the QR code with the command recommended by @PSONALl  and will try this tomorrow morning.

As a side note. I have opened a thread with the CHIP repo as well, and we have came to a different conclusion. We suspect that the network commissioning cluster is not configured correctly, and is responding with a feature map of 0, which is not supported in the matter spec. The smoking gun being:

` I (109871) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0031's Attribute 0xFFFC is 0 **********`

Following the [light example code](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L154), it does not configure the network cluster which means it must be being done under the hood somewhere.

[Reference thread here](https://github.com/project-chip/connectedhomeip/issues/26363#issuecomment-1539402905).

shubhamdp, 2023-05-10:
> The smoking gun being:
 I (109871) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0031's Attribute 0xFFFC is 0 **********

This is not the smoking gun... If you see [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp#L422-L431), `feature map` is not managed by esp-matter and this sadly this print is a bug [fixing the prints on us and tracking it internally].

For network commissioning cluster the feature map is set dynamically based on the commissioning driver being used. (please see [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/network-commissioning/network-commissioning.h#L121-L140)).

If possible can you please commission your app with the chip-tool and attach the logs here? It reads all the required attributes after PASE (Password authenticated session establishment) and we can check there what does device is sending.

alex-moran-deako, 2023-05-10:
I have used the qr code generated by 
`./out/debug/chip-tool payload generate-qrcode --discriminator 3841 --setup-pin-code 20202020 --vendor-id 65521 --product-id 32768 --commissioning-mode 0 --rendezvous 2`
Which yields MT:Y.K90CEK012Z.548G00. 

The result is still the same as using the passocde 20202020; Key exchange occurs and then it hangs while the commissioning device is querying information.

I have modified the [code here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp#L431)
- 
from  
`global::attribute::create_feature_map(cluster, 0);`
to 
`global::attribute::create_feature_map(cluster, 1);`

And now commissioning at least gets _further_, although still does not complete. It gets past the operation CSR exchange and then hangs. As I have mentioned in my previous comment, when I opened the ticket to the CHIP github, I was working and an apple engineer who confirmed responding the query of the network cluster feature map with 0 would cause the commissioner to be in an undefined state. This lines up with exactly what I was seeing, and our suspected fix resolved the issue.

On the other hand, I was able to get the chip-tool finally working, and commissioning there succeeds regardless of the feature-map of cluster 0x31. 

[chip-tool-commissioning-success.txt](https://github.com/espressif/esp-matter/files/11445337/chip-tool-commissioning-success.txt)
[device-commissioning-success.txt](https://github.com/espressif/esp-matter/files/11445333/device-commissioning-success.txt)

So it looks like my commissioning issues lie with an iOS device only.

bzbarsky-apple, 2023-05-10:
> For network commissioning cluster the feature map is set dynamically based on the commissioning driver being used. 

That's if the `Instance::Read` is reached for that attribute.  The driver registers an attribute access interface, but if the driver is never created, or if something else overrides that attribute access interface, it might not get a chance to return the value.

dhrishi, 2023-05-24:
@alex-moran-deako Please close this if your issue is resolved

alex-moran-deako, 2023-05-24:
Resolved the issue after making the change to esp-idf described here: https://github.com/espressif/esp-matter/issues/367#issuecomment-1542697805