espressif/esp-matter #369: ZAP error

va1labh, 2023-05-04:
The light-switch in the example fails to build every time due to the zap-cli. I don't seem to find any solutions. 
Error Code : 
Searching for zcl file from /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/light-switch-app/light-switch-common/light-switch-app.zap
FAILED TO EXECUTE ZAP GENERATION: No such file or directory - "/home/vallabh/MatterPrj/esp-matter/.zap/zap-cli"
********************************************************************************
* You may need to install zap. Please ensure one of these applies:
* - `zap-cli` is in $PATH. Install from https://github.com/project-chip/zap/releases
*   see docs/guides/BUILDING.md for details
* - `zap-cli` is in $ZAP_INSTALL_PATH. Use this option if you
*   installed zap but do not want to update $PATH
* - Point $ZAP_DEVELOPMENT_PATH to your local copy of zap that you
*   develop on (to use a developer build of zap)
********************************************************************************

shubhamdp, 2023-05-04:
zap-cli is needed to build the matter project.

Running [install.sh](https://github.com/espressif/esp-matter/blob/main/install.sh#L16) installs `zap-cli` and sourcing [export.sh](https://github.com/espressif/esp-matter/blob/main/export.sh#L56) exports the `zap-cli` path.

Can you check if `.zap` directory contains the zap-cli utility? If not re run the install.sh and then source the export.sh and it will fix the error.

dhrishi, 2023-05-07:
@va1labh Please see if the above fixes your issue. It should ideally work

dhrishi, 2023-05-16:
I am closing this. Please re-open if you still see the issue.

closed 2023-05-16 by dhrishi

topraveenpaul, 2023-06-27:
I get the same error. Tried running install.sh and sourcing export.sh. Didnt work.

Do we have any working branch for esp-idf and esp-matter with clear instructions that can be followed so that one can blindly follow the same to make atleast the all-cluster-app work? 

Please see the error below:

Successfully created esp32 image.
Generated /home/pi/esp-matter/examples/light/build/bootloader/bootloader.bin
[103/103] cd /home/pi/esp-matter/examples/light/build/bootloader/esp-idf/esptool_py && /home/pi/.espr... --offset 0xc000 bootloader 0x1000 /home/pi/esp-matter/examples/light/build/bootloader/bootloader.binBootloader binary size 0x6790 bytes. 0x4870 bytes (41%) free.
[638/1151] Performing configure step for 'chip_gn'Done. Made 298 targets from 236 files in 3239ms
[700/1151] Performing build step for 'chip_gn'FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a
cd /home/pi/esp-matter/examples/light/build/esp-idf/chip && ninja esp32
[40/755] ACTION //third_party/connectedhomeip/src/controller/data_model:data_model_zapgen_zap_pregen(//third_party/connectedhomeip/config/esp32/toolchain:esp32)FAILED: gen/third_party/connectedhomeip/src/controller/data_model/zap_pregen/zap-generated/access.h gen/third_party/connectedhomeip/src/controller/data_model/zap_pregen/zap-generated/gen_config.h gen/third_party/connectedhomeip/src/controller/data_model/zap_pregen/zap-generated/endpoint_config.h gen/third_party/connectedhomeip/src/controller/data_model/zap_pregen/zap-generated/CHIPClientCallbacks.h gen/third_party/connectedhomeip/src/controller/data_model/zap_pregen/zap-generated/CHIPClusters.h gen/third_party/connectedhomeip/src/controller/data_model/zap_pregen/zap-generated/IMClusterCommandHandler.cpp
python ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../../../../connectedhomeip/connectedhomeip/config/esp32/ --current-path ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model --default-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --current-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --capture-output --python-dep-list-files gen/third_party/connectedhomeip/src/controller/data_model/data_model_zapgen_zap_pregen_metadata_path_list.txt -- ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/tools/zap/generate.py --no-prettify-output --templates /home/pi/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/zap-templates/app-templates.json --output-dir /home/pi/esp-matter/examples/light/build/esp-idf/chip/gen/third_party/connectedhomeip/src/controller/data_model/zap_pregen/zap-generated --lock-file /home/pi/esp-matter/examples/light/build/esp-idf/chip/zap_gen.lock --parallel /home/pi/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.zap
Searching for zcl file from /home/pi/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.zap
FAILED TO EXECUTE ZAP GENERATION: No such file or directory - "/home/pi/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/zap/zap-cli"
********************************************************************************
* You may need to install zap. Please ensure one of these applies:
* - `zap-cli` is in $PATH. Install from https://github.com/project-chip/zap/releases
*   see docs/guides/BUILDING.md for details
* - `zap-cli` is in $ZAP_INSTALL_PATH. Use this option if you
*   installed zap but do not want to update $PATH
* - Point $ZAP_DEVELOPMENT_PATH to your local copy of zap that you
*   develop on (to use a developer build of zap)
********************************************************************************
[41/755] ACTION //third_party/connectedhomeip/src/controller/data_model:data_model_codegen_codegen(//third_party/connectedhomeip/config/esp32/toolchain:esp32)FAILED: gen/third_party/connectedhomeip/src/controller/data_model/app/PluginApplicationCallbacks.h gen/third_party/connectedhomeip/src/controller/data_model/app/callback-stub.cpp
python ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../../../../connectedhomeip/connectedhomeip/config/esp32/ --current-path ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model --default-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --current-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --capture-output --python-dep-list-files gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen_codegen_metadata_path_list.txt -- ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir gen/third_party/connectedhomeip/src/controller/data_model --expected-outputs gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen.expected.outputs ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.matter
[... cut ...]
To reactivate this environment in the future, run this in your
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Building host tools

Done. Made 6568 targets from 325 files in 4567ms
ninja: Entering directory `/home/pi/esp-matter/connectedhomeip/connectedhomeip/out/host'
ninja: error: '../../third_party/libwebsockets/repo/lib/core-net/adopt.c', needed by 'obj/third_party/libwebsockets/repo/lib/core-net/libwebsockets.adopt.c.o', missing and no known rule to make it
pi@matter-controller1:~/esp-matter$

shubhamdp, 2023-06-27:
@topraveenpaul Have you installed the prerequisites as per the [documentation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup)?

stage-01, 2023-06-28:
try adding these requirements before installing:

cd ~/.espressif/esp-matter
python3 -m pip install -r requirements.txt

shubhamdp, 2023-06-28:
As suggested by @stage-01 and the documentation [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories), you have to source the IDF environment before executing `esp-matter/install.sh`

topraveenpaul, 2023-06-28:
Thank you. Apparently my Rpi board may have issues as I see the solid green light is on for extended period of time. So i need to try this out on a new board and will update if there are still problems.