espressif/esp-matter #370: Send Notification to Home App(s)

Tim4497, 2023-05-04:
Hey there, is it possible to send notifications to the users phones Home apps? 

For example: I have a light connected to my esp32 and when I turn the light on I want to wait 1 minute and the notify the user on his phone that the light was on for a minute.

Can I send custom messages and if not what types of notifications are possible and how to add them?

shubhamdp, 2023-05-04:
@Tim4497 That is possible, but, this is not something that is possible in firmware. In your home app that you are building, subscribe to the attributes that you are interested in and track the changes and send notifications as you wish.

Tim4497, 2023-05-04:
> @Tim4497 That is possible, but, this is not something that is possible in firmware. In your home app that you are building, subscribe to the attributes that you are interested in and track the changes and send notifications as you wish.

Thanks for you reply. To clarify what you mean with firmware. Do you mean my esp32 can't trigger (custom) notifications in e.g. Apple Home? Or do you mean its not possible with esp-matter at the moment, but I can do it with CHIP?

shubhamdp, 2023-05-11:
Hi @Tim4497 and sorry for the delayed response.

What do you exactly mean by notification? 1. Something that shows up in notification panel? OR 2. When the garage door opens it updates its state from `CLOSE` to `OPEN` in Home App (just an eg)?

(2) is already implemented in Apple Home, when Apple Home commissions the device it subscribes to all attributes on the device. So whenever some attributes changes on the device it gets updated in Apple Home.

shubhamdp, 2023-05-11:
For that, there is no special change required in firmware.
In Home App, In `Home Settings`, you may have to turn on the activity notifications there, please check the screenshots.

[image]
[image]

Tim4497, 2023-05-11:
Thank, but I cannot do custom notifications on a light switch. So Notifications are only available for specific device types?

shubhamdp, 2023-05-11:
I have no idea about that, Apple folks will be more suitable to answer this. Can you please ask this on https://github.com/project-chip/connectedhomeip.